raspberrypi/firmware #793: MMAL will freeze after enabling tunneled connection when QPU is enabled

Terminus-IMRC, 2017-04-06:
I found that MMAL will freeze after enabling tunneled connection when QPU is enabled.
Here is the code: https://github.com/Terminus-IMRC/mmal-fails-to-initialize-tunneled-connection-with-qpu

This program captures frames using `vc.ril.camera` and renders it using `vc.ril.video_render`, with these components connected by using `mmal_connection_*()` stuff.

When raspi executes the program after reboot with `QPU_ENABLE=1` and `CONNECTION_FLAG=MMAL_CONNECTION_FLAG_TUNNELLING`, MMAL will freeze.

However, if `QPU_ENABLE=0`, MMAL will not freeze.

In addition, when raspi executes the program after reboot with `QPU_ENABLE=0`, execution with `QPU_ENABLE=1` and `CONNECTION_FLAG=MMAL_CONNECTION_FLAG_TUNNELLING` will not freeze MMAL.

After MMAL is freezed, the led on OV5647 won't be off even after killing the program. And other MMAL application such as raspistill will also become not working. And `vcmsg log msg` says:
```
046795.558: mmal: fail_destroy: Timed out waiting to destroy ril.camera
```

Is this issue reproducible?

Thank you for reading this.

Terminus-IMRC, 2017-04-06:
If you replace the camera with `vc.ril.source`, it won't freeze.

Terminus-IMRC, 2017-04-07:
I found this thread: [MMAL camera and VC4 QPU](https://www.raspberrypi.org/forums/viewtopic.php?f=67&t=100317). In this thread, it is said that MMAL uses QPU for AWB processing. I'll have a try to disable AWB.

Terminus-IMRC, 2017-04-07:
Is AWB the only process that uses QPU even for now? And do you think the cause of the problem is this?

Terminus-IMRC, 2017-04-07:
There is a non-QPU scheme for that: https://github.com/raspberrypi/firmware/issues/760#issuecomment-287391025

6by9, 2017-04-07:
The QPU framework under the firmware does not support multiple simultaneous jobs from different sources - you need to acquire and release it as you have work to do.
```mailbox_qpu_enable(fd_mb, QPU_ENABLE);``` acquires the QPUs, so AWB will block waiting for it to become available. Likewise if you tried any OpenGLES operations at that point it would stall.

Yes, AWB, OpenGLES, and the mailbox service are the only users of the QPU that I am aware of from the firmware.

Terminus-IMRC, 2017-04-07:
Thanks. So switching the AWB process to non-QPU-based scheme will make vc.ril.camera and QPU work at the same time?

Terminus-IMRC, 2017-04-07:
Setting `~ ((1 << 2) | (1 << 3) | (1 << 5))` to `MMAL_PARAMETER_CAMERA_ISP_BLOCK_OVERRIDE` won't work, it seems.

6by9, 2017-04-07:
> So switching the AWB process to non-QPU-based scheme will make vc.ril.camera and QPU work at the same time?

Correct,  but disables ALL access to the QPUs from the VPU, so the mailbox QPU service is also not available. It is intended for handing the QPUs and V3D to the ARM for use by the vc4-kms-v3d driver, not so that you can hog the QPU through bad programming.

> Setting ~ ((1 << 2) | (1 << 3) | (1 << 5)) to MMAL_PARAMETER_CAMERA_ISP_BLOCK_OVERRIDE won't work, it seems.

No, that is disabling the hardware blocks. The control algorithms are running, but the hardware block is in bypass.

Call ```mailbox_qpu_enable(fd_mb, QPU_ENABLE);``` when you have some work to do on the QPUs, and ```mailbox_qpu_enable(fd_mb, 0);``` when you have completed that job.

Terminus-IMRC, 2017-04-07:
Thank you! I'll have a try.

Terminus-IMRC, 2017-04-07:
Is there any reason why the control algorithms on QPU cannot be disabled?

6by9, 2017-04-07:
The alternative algorithm is a significantly simpler grey-world based implementation, so compromises the white balance performance. We are not wishing to go down that route generally.
Switching to it is an unfortunate necessity in the case of vc4-kms-v3d as otherwise the VPU code stalls due to interrupt handling being disabled, but the ARM has also had any QPU state it had set up totally trampled. We will revisit whether there is a VPU based compromise when there is the resource available, but that isn't likely to be for a while.

AFAICT It is not a necessity in your case as you don't have a long running QPU job, you're just hogging the resource. IIRC AWB only requires about 40ms every 10 to 15 frames, so it is not a huge loading and should be easily schedulable between other jobs.

Terminus-IMRC, 2017-04-07:
I see. Thank you very much for doing hard work for Raspberry Pi and answering my insignificant questions.

Terminus-IMRC, 2017-04-07:
It worked! Thanks!

closed 2017-04-07 by Terminus-IMRC