raspberrypi/firmware #794: bootcode:serialnumber

radicspeter, 2017-04-07:
According to https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net.md
i would like to use serial_number in the network booting on Raspberry Pi 2.

But if i make a serialnumber folder in the tftpboot root folder, my raspberry is not try to search its tftpboot\serialnumber\start.elf it's on searching on tftpboot\start.elf.

[image]

popcornmix, 2017-04-07:
Ping @ghollingworth

pelwell, 2017-04-07:
Do you have any more history than that? I'd want to see from the first DHCP request onwards, and your trace starts much later than that.

radicspeter, 2017-04-07:
[image]

ghollingworth, 2017-04-08:
It's on my list, might have to ask someone else to look at it though!

pelwell, 2017-04-08:
From what I can see, bootcode.bin always tries to find "<serial>/start.elf" after DHCP and ARP. The only reason it would not is if your bootcode.bin predates that code, which was committed in July last year (I'd have to hunt around to find out which firmware release it first appeared in.

Which firmware are you running? Assuming you haven't tried to mix & match bootcode.bin and start.elf, `strings /boot/start.elf | grep VC_BUILD` should give a sensible answer.

pelwell, 2017-04-08:
Update: So I hunted around, and the bootcode with network booting support hit the next branch Jul 26 2016, then was merged into master Jan 10 2017.

amtssp, 2017-04-10:
Is it not the same issue as described here: https://github.com/raspberrypi/firmware/issues/754

pelwell, 2017-04-11:
Thanks for trying. I'm going to be limited in what I can do until next week, but between us we'll make it work.

radicspeter, 2017-04-11:
Thanks pelwell, i can't wait ;)

pelwell, 2017-04-18:
Can you download and test [this bootcode.bin](https://drive.google.com/open?id=0B_P-i4u-SLBXcEpRbDhSaVNJQ3c)? It should only alter behaviour in the case where bootcode.bin is present on the SD card and the failover to net booting.

radicspeter, 2017-04-19:
Thanks, i am going to test today, i have just seen in the #754 that it is working well now... But i will reply ;)

popcornmix, 2017-04-19:
The fix should be in latest rpi-update firmware

radicspeter, 2017-04-19:
Okay. now i tested and working good here! Thanks to solved the issue. Ticket closed...
[image]

closed 2017-04-19 by radicspeter