raspberrypi/firmware #795: PAL-B/G Colorcarrier (4.43 MHz) randomly lost since 4.9.14

Quantomax, 2017-04-08:
Since commit b30461d2621e71eb9b8845b38738af177b9181ed (bump to Kernel 4.9.14) I mostly get b/w video on a PAL-B/G TV (Germay, 4.43 MHz) via Video-Out, interchanged with only some seconds "color" every now and than. Going back to d8803bc3b369b087450e636beb0cb30857e3215e resolves the issue.

Config.txt:
disable_overscan=0
overscan_scale=1
hdmi_drive=2
sdtv_mode=2
sdtv_aspect=1
sdtv_disable_colourburst=0
gpu_mem_1024=256

popcornmix, 2017-04-08:
There were a couple of changes in that firmware. I'll try reverting one. Can you test:
https://drive.google.com/uc?id=0B-6zmEDJwxZEbnljcWdZUUpWZTA&export=download
and let me know if it is good/bad?

Quantomax, 2017-04-08:
Updated Firmware, copied files above to /boot, rebooted and seems to work ... 
Thanks. ;-)

popcornmix, 2017-04-08:
Thanks.

@ghollingworth This is due to "Set up HDMI VCO same for VEC as for HDMI".

popcornmix, 2017-04-19:
Can you test again with latest rpi-update firmware?

Quantomax, 2017-04-19:
Firmware / Kernel 4.9.23 works colourized as expected on PAL-B/G TV. Thanks for the fix.

closed 2017-04-19 by Quantomax