raspberrypi/firmware #797: #758 - How to confirm same issue?

nicksoph, 2017-04-13:
Hi, purchased a PiZero W a few weeks ago and only just started trying to use it. Using the latest full standard image, (full download twice onto 2 SD cards - same symptoms which are; The bluetooth Icon shows but there are only two grey bars with 2 red crosses where the wifi icon should be, clicking that produces an error message that 'No wireless interfaces found'. What do I need to do to test this is a fault with the board and not something that can be fixed by reprogramming at this end?

pelwell, 2017-04-13:
Open a terminal window and type:
```
grep Revision /proc/cpuinfo
```
What does it say?

nicksoph, 2017-04-13:
Hi - see edit

nicksoph, 2017-04-13:
Copied from terminal window 
Might be harder to find as plain text but error msgs in red on screen  
re brcmfmac at 
8.59... 
8.9.... 
11.42 ...
11.642...
14.8...

.

[dmesg.txt](https://github.com/raspberrypi/firmware/files/919096/dmesg.txt)

nicksoph, 2017-04-13:
Hi Phil, I think this is fixable from this end -  just managed to get the wifi adaptor to show.

pelwell, 2017-04-13:
Cool - I was failing to see anything wrong. Can I ask what you've done?

nicksoph, 2017-04-13:
Installed 10.4.17 Jessie full from rp org on sd card and noticed fault as described above. (rinsed and repeated) same issue. Reported fault here and posted info as shown. Followed advice on Rpi org forum suggesting the following - Removed SD card from Zero W and into RPi 3. Ran - sudo apt-get update & sudo apt-get upgrade. Didnt take a proper note of what was updated but 'update' seemed to install only some language files and 'upgrade' one small file (19kb?). Shutdown Rpi3 and rebooted the zero W with the updated sd card. Wireless network adaptor was now there and scanning. 

Went through the process again - freshly installed image onto card placed into zero W - network adaptor there! Nothing else physically changed - same kb/mouse/power supplies/cables.

Compared old and new vcgencmd & /proc/cpuinfo - exactly the same.
Attached new dmesg output- which to my untrained eye seems the same.
[dmesg2.txt](https://github.com/raspberrypi/firmware/files/919477/dmesg2.txt)

Thank you for your help.

closed 2017-04-13 by nicksoph