raspberrypi/firmware #798: Patched kernel panics with enable_uart=1 on a RPi3

anyc, 2017-04-14:
On my RPis, I use uboot to boot the RPi kernel with applied patches from the Ubuntu kernel PPA. Today, I tried a 4.9 kernel for the first time and while it works on a RPi2, the kernel only boots on a RPi3 if I disable enable_uart. With enable_uart=1, I can see the uboot console but I don't get any kernel output. If I attach an HDMI monitor I can see a kernel panic but unfortunately, the interesting messages scroll out immediately and I only see messages like "CPU $x stopping", "bad PC value" and "attempting to kill init". 

In my config.txt, I only have a kernel= statement to load uboot. Except for the kernel and uboot, the other files in /boot are up-to-date copies from this repository. In order to get a serial console, I also tried to boot without enable_uart and with pi3-disable-bt but still I don't get any output on the serial console. 

Hence, I would like to ask, if someone has an idea what I might be doing wrong here? The same setup worked with the 4.4 kernel (although the last one I tried was 4.4.35). Thank you!

pelwell, 2017-04-14:
1. Is uboot passing the firmware-generated dtb to the kernel, or is it creating/loading another?
2. What is in the command-line passed to the kernel?

anyc, 2017-04-14:
It should pass it through. I use a recent uboot version that includes the corresponding commit.

With enable_uart=0, I get the following commandline:
```8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait root=/dev/mmcblk0p2 rw rootwait rootfstype=btrfs verbose logo.nologo```. The only significant change should be the rootfstype=btrfs.

The last lines of vcdbg log msg:
```001377.805: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001380.840: *** Restart logging
001383.027: Failed to open command line file 'cmdline.txt'
001415.075: Loading 'u-boot-rpi3_32.bin' to 0x8000 size 0x5801c
001418.727: Kernel trailer DTOK property says yes
001418.743: Kernel trailer DDTK property says no
001424.949: Loading 'bcm2710-rpi-3-b.dtb' to 0x6001c size 0x449c
001555.215: Failed to load overlay 'pi3-disable-bt'
002614.188: Device tree loaded to 0x2effb700 (size 0x4877)
002619.768: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
014332.000: vchiq_core: vchiq_init_state: slot_zero = 0xfa980000, is_master = 1
014336.274: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
```
I just noticed the failed loading of the overlay. The dtbo is present, I will do some further research.

pelwell, 2017-04-14:
The DDTK=no flag in the "kernel" trailer is causing the firmware to look for the old form of overlays, i.e. *-overlay.dtb. Note the quotes around kernel because in this instance the firmware is loading uboot as the kernel, so the trailer is on uboot.

We no longer recommend using the mkknlimg utility - the default behaviour without the trailer is now suitable in almost all circumstances.

pelwell, 2017-04-14:
Another thought: take out the kgbdoc reference - I've seen it cause similar-sounding problems with ttyS0.

anyc, 2017-04-16:
Ok, I will skip the tool then. However, looks like I misconfigured something. :/ After rebuilding from scratch, everything works. Thank you for your help!

closed 2017-04-16 by anyc

anyc, 2017-04-16:
Ah, and you were right, kgdboc was the culprit. I forgot that I also changed this before rebuilding from scratch. If I add it again, booting fails again.