raspberrypi/firmware #800: RPi3 does not handle slow bitrates anymore

0xygen8, 2017-04-18:
I'm using a Pi2 to communicate with a device running at 300-e-7-1 (city-heating system that cannot be altered) I found that when I switched to a Pi3  a bitrate of 300bps is no longer a possible rate (only garbled data) I tried switching the UART (pi3-disable-bt) but that has no effect. Is there a way to solve this? (I'm using Raspbian Jessie lite dd 2017-04-10)

pelwell, 2017-04-18:
You are correct that, as it comes, the available (non-Bluetooth) Pi3 UART cannot communicate at 300 baud. The BCM2835 ARM Peripherals spec says:

    The mini UART uses 8-times oversampling. The Baudrate can be calculated from:
        baudrate = system_clock_freq/(8*(baudrate_reg + 1))

    If the system clock is 250 MHz and the baud register is zero the baudrate is 31.25 Mega
    baud. (25 Mbits/sec or 3.125 Mbytes/sec). The lowest baudrate with a 250 MHz system
    clock is 476 Baud.

In other words, 300 baud is too low.

However, with the pi3-disable-bt (or pi3-miniuart-bt) overlay, ttyAMA0 should become available again. Please make sure that your config.txt contains:
```
dtoverlay=pi3-disable-bt
```
and try again.

0xygen8, 2017-04-19:
Phil, I have added dtoverlay=pi3-disable-bt to the cmdline, rebooted (checked with dmesg) and tried but it still did not work. The reason I ended up asking the question here is because reading other peoples postings (to increase the baud rate) I have the notion that the uart clock of the Pi3 is adjusted to a higher setting...

pelwell, 2017-04-19:
I think there ought to be enough range in that divider for 300 baud from a 48MHz clock, but you can try adding `init_uart_clock=3000000` to config.txt to restore the old clock rate.

0xygen8, 2017-04-19:
hmm must be doing something wrong

```
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 dma.dmachans=0x7f35 bcm2709.boardrev=0xa02082 bcm2709.serial=0x60524289 bcm2709.uart_clock=48000000 smsc95xx.macaddr=B8:27:EB:52:42:89 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=860e97bf-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait init_uart_clock=3000000 dtoverlay=pi3-miniuart-bt
[    0.081168] uart-pl011 3f201000.uart: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.281801] 3f215040.uart: ttyS0 at MMIO 0x3f215040 (irq = 59, base_baud = 31250000) is a 16550
[    0.961452] 3f201000.uart: ttyAMA0 at MMIO 0x3f201000 (irq = 87, base_baud = 0) is a PL011 rev2
[  147.851767] uart-pl011 3f201000.uart: no DMA platform data
```
Strange...
```
ls -al /dev/serial*
lrwxrwxrwx 1 root root 5 Apr 19 06:29 /dev/serial0 -> ttyS0
lrwxrwxrwx 1 root root 7 Apr 19 06:29 /dev/serial1 -> ttyAMA0
```
and ttyAMA still does not work

pelwell, 2017-04-19:
It looks like you've added the additional settings to cmdline.txt - they should go in config.txt. Remove "init_uart_clock=3000000 dtoverlay=pi3-miniuart-bt" from cmdline.txt and add "dtoverlay=pi3-miniuart-bt" to config.txt, then you should find it works.

pelwell, 2017-04-19:
cmdline.txt is just for the kernel command line - all other firmware and device tree settings should go in config.txt.

0xygen8, 2017-04-19:
Phill -- You are briljant! 👍  thank you. as you expected the init_uart is not needed. 
My error was that I put the dtoverlay in the wrong file :(
please close this user-error

closed 2017-04-19 by pelwell