raspberrypi/firmware #801: PXE boot does not honor `siaddr` Bootstrap Protocol field.

lastmikoi, 2017-04-18:
I am currently trying to get my Raspberry Pi 3 (firmware `20170405`) to boot from network using PXE, while namely following https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net.md.

My "infrastructure" has an already existing TFTP server, running on `172.29.0.2`, while the main router and DHCP server runs on `172.29.0.254`.

It would appears that the current PXE boot ignores the `siaddr` field of the BOOTP offer that's being emitted from my DHCP server, which contains the IP address to use to reach the TFTP server. Indeed it seems to simply re-use the initial DHCP server IP address.

DHCP Offer from my DHCP server (with MAC addresses redacted):
```
(...)
Internet Protocol Version 4, Src: 172.29.0.254, Dst: 172.29.0.128
User Datagram Protocol, Src Port: 67, Dst Port: 68
Bootstrap Protocol (Offer)
    Message type: Boot Reply (2)
    Hardware type: Ethernet (0x01)
    Hardware address length: 6
    Hops: 0
    Transaction ID: 0x26f30339
    Seconds elapsed: 0
    Bootp flags: 0x0000 (Unicast)
        0... .... .... .... = Broadcast flag: Unicast
        .000 0000 0000 0000 = Reserved flags: 0x0000
    Client IP address: 0.0.0.0
    Your (client) IP address: 172.29.0.128
    Next server IP address: 172.29.0.2
    Relay agent IP address: 0.0.0.0
    Client MAC address: Raspberr_xx:xx:xx (b8:27:eb:xx:xx:xx)
    Client hardware address padding: 00000000000000000000
    Server host name not given
    Boot file name: bootcode.bin
    Magic cookie: DHCP
    Option: (53) DHCP Message Type (Offer)
        Length: 1
        DHCP: Offer (2)
    Option: (54) DHCP Server Identifier
        Length: 4
        DHCP Server Identifier: 172.29.0.254
    Option: (51) IP Address Lease Time
        Length: 4
        IP Address Lease Time: (900s) 15 minutes
    Option: (43) Vendor-Specific Information
        Length: 20
        Value: 52617370626572727920506920426f6f74202020
    Option: (1) Subnet Mask
        Length: 4
        Subnet Mask: 255.255.255.0
    Option: (255) End
    Padding: 00000000000000000000000000000000
```

Then after that the Raspberry emits the following TFTP read request:
```
(...)
Internet Protocol Version 4, Src: 172.29.0.128, Dst: 172.29.0.254
User Datagram Protocol, Src Port: 49152, Dst Port: 69
Trivial File Transfer Protocol
    Opcode: Read Request (1)
    Source File: bootcode.bin
    Type: octet
```

I have tested my DHCP+TFTP setup on x86 virtual machines (with only slight changes in DHCP options) and they boot as expected, so I believe there is an issue as far as the raspberry netboot is concerned.

ghollingworth, 2017-04-19:
There are two options for providing the TFTP server, the first is as you have used, providing option 43 with "Raspberry Pi Boot   " in the DHCP reply, the second is to provide option 66 with the IP address of the TFTP server.  This needs to be in text "192.168.1.1" for example

The Next Server option is not supposed to contain a TFTP server, it's supposed to point to a secondary DHCP server.

lastmikoi, 2017-04-19:
You are right, adding the option 66 to my ISC DHCPd configuration (through `option tftp-server-name "172.29.0.2";`) made it work.

I guess I was wrong in assuming that iPXE (running on my x86 virtual machines) could be used as a reference implementation to test against, color me enlightened 👍.

Thanks for the help ! I'll close this issue now that it has been solved on my end..

closed 2017-04-19 by lastmikoi