raspberrypi/firmware #803: Compatibility between firmware and kernel?

rodizio1, 2017-04-20:
I've recently tried to upgrade the firmware to the one from the latest Raspbian image (2017-04-10). Now the system doesn't boot anymore. It doesn't even initialize the display, HDMI monitor shows "no device connected".

Figured out that enabling  `start_x=1` in config.txt causes this.

Using this kernel from an older Raspbian version:
> Linux wifibroadcast 4.4.11-v7+ #26 SMP Wed Apr 19 01:03:25 CEST 2017 armv7l GNU/Linux

If I remember correctly, Raspbian 2017-04-10 uses 4.4.50. Is that difference the cause?

Is there any documentation on what Kernel fits with what firmware? How is this handled internally, I'd assume there is an API version inside the firmware and the kernel that needs to match?

pelwell, 2017-04-20:
There is very little coupling between Raspbian and the kernel - there's more between the VPU image (start.elf) and the kernel, but using any of our update mechanisms shouldn't keep those in sync.

1) How are you updating? I would recommend `sudo BRANCH=stable rpi-update` in this case.

2) Which model of Pi do you have?

rodizio1, 2017-04-20:
It's a Pi3. Will try other models later when I have more time.

I don't want to update the kernel because newer kernels have usb timing issues or something (I get packetloss on my wifi cards in monitor mode). So updating was done by simply copying the firmware files (*.dtb, *.bin, *.elf) over from the latest Raspbian image.

pelwell, 2017-04-20:
Newer than what? The stable branch will get you 4.4.50.

rodizio1, 2017-04-20:
Newer than the 4.4.11 (from an older Raspbian version) I use now. Have tried different ones so far. The  4.4.50 one from the 2017-04-10 Raspbian release I haven't tried yet though. 

So I'm forced to upgrade to 4.4.50 I guess.

pelwell, 2017-04-20:
As an intermediate step you could try:
```
sudo BRANCH=stable SKIP_KERNEL=1 rpi-update
```
which will update the VPU firmware and leave the kernel unchanged.

rodizio1, 2017-04-20:
wouldn't that be exactly the same as I did with copying the firmware from the latest raspbian distribution?

Just tried it with the Pi0, same result. 

 I have feedback from a user of my EZ-Wifibroadcast distribution, that copying over the firmware from an older Raspbian Version (the first one with Pi0 W support) did work for him.

So I think the incompatibility must've been introduced somewhere between those two Raspbian versions (or the firmware versions they contain, to be be precise).

My goal is to have a firmware that works with my 4.4.11 kernel and also with the new Pi0 W.

popcornmix, 2017-04-21:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo SKIP_KERNEL=1 rpi-update <hash>```
to revert back to that version. Report the first version with the problem.

rodizio1, 2017-06-18:
Still haven't got around buying a Pi0 W to do further testing. What I can tell so far is, that it has something to do with the start_x.elf firmware file. People using my EZ-Wifibroadcast distribution which uses Kernel 4.4.11 report that it works for them with the Pi0 W if they replace only the start_x.elf file from the 03-03-2017 Rasbian image.

Since there seems to be no API versioning or other way to tell which firmware works with what kernel, I guess I'll eventually have to upgrade the kernel to the one from the latest Raspbian release.

6by9, 2017-06-19:
If you replace the start*.elf files, you also need to replace the fixup*.dat files.

You appear to not be picking those up based on your comment:

> So updating was done by simply copying the firmware files (*.dtb, *.bin, *.elf) over from the latest Raspbian image.

JamesH65, 2017-12-22:
@rodizio1 Any change of status on this? Can it be closed?

rodizio1, 2018-02-21:
|You appear to not be picking those up based on your 
|comment:

I did replace everything, including the .dat files.

@JamesH65
Status is: Not all firmware versions work with all kernels. There seems to be no versioning or API or something like that regarding the interaction between kernel and firmware. Thus, there is no sure way to tell firmware X will work with kernel Y.

Consequence for me is, I'll use the exact firmwares and kernel versions included in the Raspbian Releases, as it's the only way to tell that kernel and firmware are compatible.

rodizio1, 2018-02-22:
Knowing that a slightly newer firmware will most likely work, gives a bit more flexibility, thanks for the info.