raspberrypi/firmware #805: USB boot. Boots on new. Not on updated

RogierQ, 2017-04-26:
Playing around with the usb boot option. Sounds nice!
So did update and made image and put it on usb stick, no succes
Tried other brand stick, no succes
Did again an update, made new image and so on, no succes

Them downloaded a fresh raspian, great succes!
Tried other stick, great succes!!

So fresh stick runs and updated not...
Also the bootcode.bin didnt solve it

popcornmix, 2017-04-26:
On the system that isn't working, was it raspbian jessie?
What does:
``vcgencmd version
uname -a``
report?

RogierQ, 2017-04-26:
uname -a gives:
Linux new 4.9.24-v7+ #991 SMP Sat Apr 22 20:26:57 BST 2017 armv7l GNU/Linux

vcgencmd version gives :
Copyright (c) 2012 Broadcom
version e49fda6e8d837fd3cbac4900a141dd7fba7ca3e3 (clean) (release)

popcornmix, 2017-04-26:
``vcgencmd version`` reports a date. I suspect it was around Apr 19.
Did you update with rpi-udpate?
Ping @ghollingworth

RogierQ, 2017-04-26:
first i did update with apt-get when that didnt fix it i tried it with rpi-update
Is that wrong or good to do?

popcornmix, 2017-04-26:
``apt-get`` gets you latest stable kernel/firmware
``rpi-update`` gets you latest testing kernel/firmware
As USB boot is still somewhat experimental then using rpi-update is fine (but be aware it could contain regressions).

RogierQ, 2017-04-26:
ok clear, thanks!

But is the version now the right version?

popcornmix, 2017-04-26:
Yes, it it latest version.
Can you confirm your setup? Are you using Pi3?
Are you booting without an sdcard, or with an sdcard with bootcode.bin?

RogierQ, 2017-04-26:
Jep using a pi3
Without sdcard and with, both no succes.

With same setup with brand new clean install version it works...

popcornmix, 2017-04-26:
What is ``uname -a`` and ``vcgencmd version`` of working system?
Can you also report md5sum of bootcode.bin for the working and non-working systems?

RogierQ, 2017-04-26:
ehh sorry but how do i get the md5sum?

popcornmix, 2017-04-26:
``md5sum /boot/bootcode.bin``
should do it

RogierQ, 2017-04-26:
non working gives
21b9cac81823ae4e92d0936ed306593e  /boot/bootcode.bin

the working version comes after finishing the usb stick...

RogierQ, 2017-04-26:
The working version gives 60009c193abdc91edc6439223266b00d  /boot/bootcode.bin

popcornmix, 2017-04-26:
21b9cac81823ae4e92d0936ed306593e
is version from ``rpi-update 06c104c37348e104d7bc108b8ad19697df93b589`` (Apr 19)

60009c193abdc91edc6439223266b00d
is version from ``rpi-update 3ff94f1fc459f88d3e2530542fb609643a7bd1a6`` (Apr 4)

So on the not-working system does running:
``sudo rpi-update 3ff94f1fc459f88d3e2530542fb609643a7bd1a6``
make it work?

RogierQ, 2017-04-26:
ehh you mean run on command line :
sudo rpi-update 3ff94f1fc459f88d3e2530542fb609643a7bd1a6 ?

popcornmix, 2017-04-26:
Yes. That updates firmware/kernel to a specific version (from Apr 4) which should make it match your working system.

RogierQ, 2017-04-26:
Nope... not working.... :-(

ghollingworth, 2017-05-01:
@JamesH65 Might be worth a test in the regression system for this

JamesH65, 2017-12-22:
@RogierQ Have you sorted this out now? Can it be closed?

closed 2019-11-05 by JamesH65