raspberrypi/firmware #806: MMAL doesn't support RGB image which width is over 3264?

Terminus-IMRC, 2017-04-28:
It seems that MMAL doesn't support RGB image which width is over 3264.
I tested it with vc.ril.source and vc.ril.isp.
Please try with this code: https://github.com/Terminus-IMRC/mmal-rgb-width-limit

This program creates source, isp and render components and connect them in this way:
```
source[0] -- [0]isp[0] -- [0]render
```
A random image generated by vc.ril.source is converted or passed-through by vc.ril.isp and then rendered on screen by vc.ril.render.

By default, encodings of output ports of vc.ril.source and vc.ril.isp (`ENCODING_SOURCE_OUT` and `ENCODING_ISP_OUT` respectively) are both set to I420, which works well:
[image]

If you set `ENCODING_SOURCE_OUT` to RGB-ish encoding such as RGB16 and RGB24, this error is produced:
```
mmal: mmal_vc_port_info_set: failed to set port info (3:0): EINVAL
mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0x1ecec20 (EINVAL)
main.c:71: config_port(cp_source->output[0], ENCODING_SOURCE_OUT, WIDTH, HEIGHT): EINVAL (0x00000003)
```
And if you set `ENCODING_ISP_OUT` to RGB-ish encoding, noises are rendered:
[image]

However, if you set `WIDTH` to smaller one than 3265 such as 3264, it works well.

Is 3264 the upper limit of WIDTH of RGB image? Could you please fix this?

Thank you for reading.

6by9, 2017-04-28:
Almost certainly you're not hitting a MMAL issue but a composition one.
The HVS (Hardware Video Scaler) that generates the output for the screen has to be able to read the source images in realtime, resize it, and then produce the output. There are various factors involved as to the cost of any resize, but there aren't any tricks to downscale horizontally. (Vertically you can skip lines)

Add ```dispmanx_offline=1``` to config.txt so the HVS composes to an offscreen buffer which is then rendered and I suspect it will work. Display framerate will be compromised but would confirm that it is just failing to maintain realtime rendering.

I420 is likely to work as there is less data to process (the chroma is already subsampled).

6by9, 2017-04-28:
Having said that, the console framebuffer appears not to be tearing which I would expect if the HVS were underflowing.
Can you also try adding ```start_debug=1``` to config,txt and log ```sudo log vcdbg assert``` after your test to see if there are any VPU asserts firing?

Terminus-IMRC, 2017-04-29:
Thank you for replying to me every time!

I first hit this issue when I only used `vc.ril.isp` to convert IMX219 RAW10 (3280x2464) from file to RGB buffer. i.e. No source and render components there. So I think the cause is only related to `vc.ril.isp`.

I updated the firmware by running `rpi-update` before testing. And before each test I rebooted the pi. `ENCODING_SOURCE_OUT` is set to I420 and `ENCODING_ISP_OUT` is set to RGB24 here.

* With `dispmanx_offline=0` and `WIDTH=3264`, correct image is rendered on screen with this log:
```
083016.449: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
083178.534: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
083340.638: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
083503.027: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
083665.026: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
083827.085: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
083989.391: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
084151.374: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
084313.741: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
084475.800: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
084637.986: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
084802.505: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
```
* With `dispmanx_offline=1` and `WIDTH=3264`, correct image is rendered on screen with this log:
```
070538.867: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
070671.674: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
070804.504: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
070937.256: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
071070.090: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
071202.900: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
071335.692: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
071468.563: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
[... 69 lines cut ...]
----------------
053500.324: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
053586.277: assert( (params->stride1&31)==0 ) failed; ../../../../../middleware/ISP/plat/isp_plat_hw.c::isp_plat_write_hi_res_output line 3805 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
053633.491: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
[... cut ...]
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
053853.684: assert( (params->stride1&31)==0 ) failed; ../../../../../middleware/ISP/plat/isp_plat_hw.c::isp_plat_write_hi_res_output line 3805 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
053900.900: assert( id->turbo_cookie_isp ) failed; ../../../../../middleware/openmaxil/components/isp.c::ip_process_thread line 2439 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
```

Hope it helps. Thank you for reading.

6by9, 2017-04-29:
```
053853.684: assert( (params->stride1&31)==0 ) failed; ../../../../../middleware/ISP/plat/isp_plat_hw.c::isp_plat_write_hi_res_output line 3805 rev 17af581
vcdbg_ctx_get_dump_stack: dump_stack failed
```
Looks odd and the only thing of real interest. It should be taking the value from port->format->es->video.width which you appear to be correctly aligning up to a multiple of 32 (unless I've got something wrong and that value is being ignored).
The standard RGB handling only requires the stride to be a multiple of 16 whilst the ISP needs 32. That should all be dealt with by your aligning, but it'd be worth trying width = 3281 to 3296 as the stride at 3296 would be a multiple of 32 (width=3265 gives a stride of 3280 which is only a multiple of 16, not 32).
(It being the weekend, I don't have a Pi set up at home to test issues on. Otherwise I'll pick it up on Tuesday (Monday being a bank holiday in the UK) when I'm back in the office).

Which platform are you on? The id->turbo_cookie_isp assert is either a programmatic error or something due to the specific platform. The ISP component does an overclock automatically whilst it is processing images. It may be that that is inappropriate on a specific platform, or the stride assert is causing that failure.

Thank you for such comprehensive bug reports with test apps! It makes tracking down the actual issue much easier.
BTW the issue appears to be solely with the MMAL ISP component, not MMAL itself. An edit to the title would probably be in order.

Terminus-IMRC, 2017-04-29:
It turned out that vc.ril.isp fails to convert I420 to RGB when `video.crop.width` is in range `[n * 32 + 1, n * 32 + 16]`. (`video.width` is aligned-up to `n * 32 + 32` though.)
e.g., Setting `WIDTH` to 512 and 529-544 works well but setting it to 513-528 don't:

| WIDTH (video.crop.width) | width aligned-up by 32 (video.width) | works or not |
| :-- | :-- | :-- |
| 512 | 512 | works |
| 513-528 | 544 | noisy |
| 529-544 | 544 | works |

So setting `WIDTH` to 3264 and 3281 works well but 3265 not.

I was using PiZero W. After replaced it to Pi3B, `id->turbo_cookie_isp` asserts disappeared though `(params->stride1&31)==0` asserts are still there. So the former isn't related to this isp issue.

(Please don't work too hard on holidays!)

6by9, 2017-04-29:
So the ISP hardware certainly requires a stride that is a multiple of 32 bytes. With RGB that is actually quite easy to arrange seeing as each pixel is either 2, 3 or 4 bytes.
Ah, RGB24. I vaguely recall now fun with configuring the stride on that. IL configures the stride in bytes instead of MMAL's pixels, so tried allowing multiples of 48. One to revisit and ensure that the MMAL configured value is actually applied.
The main user of the ISP component is Chrome for video decode, and that uses RGBA buffers so avoids funny stride calcs.

I'll raise the ```id->turbo_cookie_isp``` issue on Pi0 separately. It'll be unrelated but ought to be addressed.

Terminus-IMRC, 2017-04-29:
Even if `ENCODING_ISP_OUT` is RGBA, noisy image is produced on the same condition above:
[image]
width=513 and height=512 here.

Thanks for raising another issue. I should note here that I used PiZero *W*, not a plain PiZero.

popcornmix, 2017-04-29:
I've had a look. ``id->turbo_cookie_isp`` is harmless.
Basically auto turbo only occurs on some platforms (pi2/pi3).
You can make it work on pi0/pi1 with ``disable_auto_turbo=0`` in config.txt which does have the effect of avoiding the assert.
The assert is just saying that turbo mode wasn't active at the end of ISP resize, which is actually expected on pi0/pi1. I'll remove the assert.

6by9, 2017-04-29:
Thanks @popcornmix. I wasn't going to raise the turbo_cookie assert until Tuesday - I was fairly confident it wasn't critical and we're not in the office.

I'm a little bemused on RGBA failing as the Chrome use case exercised a fair number of widths and we went through a few iterations to get it right. It should be pretty obvious with the debugger.

Terminus-IMRC, 2017-04-29:
Thanks! I'll wait for it.

6by9, 2017-05-02:
So the main culprit is found - the ISP component was always ignoring the user specified stride on the output port.
Secondary issue that the source component was also ignoring a user specified stride, however you'd set it to I420 which inherently has a stride that is a multiple of 32 and therefore matched your request. It didn't work with RGB which only needs the stride to be a multple of 16.
Third quirk found which would just be firing an assert.

It has made me think about what you are asking for though. You've asked source for 3265x4096 I420. I420 is chroma subsampled, so the last one pixel becomes awkward as it is likely to get lost in the subsampling. It doesn't blow up, but may not be quite what you intend.
The camera component doesn't allow you to select a width/height that isn't a multiple of 2. I could enforce the same on source and isp, but it probably isn't worth it.

I'm just creating the patches, so the fixes should be in the next rpi-update release.

Terminus-IMRC, 2017-05-03:
Thank you for investigating the issue!

I know the last pixel becomes awkward in I420 when the width is odd.
I just asked for width=3265 simply because the glitch comes up on the width. And I originally intended to use the isp with camera (BAYER->I420 and I420->RGB), so I think prohibiting odd width on I420 makes sense.

Thanks.

6by9, 2017-05-03:
I'll worry about odd widths (YUV420 or 422) and heights (YUV420 only) at a later date as it looks like multiple components need to be checked. It was more flagging up that you may get slightly odd results at the edge of the image.

popcornmix, 2017-05-04:
Latest rpi-update firmware has some fix from @6by9

Terminus-IMRC, 2017-05-04:
Thank you for the fixes! I really appreciate it!
I confirmed the issue is fixed (though I didn't tested for bayer formats since I'm missing the code), so closing this.

I noticed the program doesn't work correctly for width>=4096 even if `dispmanx_offline=1` (Maybe a hardware limitation?). I'm creating a reproducible program and planning to open a new issue.

Thank you again for investigating the issue!

closed 2017-05-04 by Terminus-IMRC

popcornmix, 2017-05-05:
Yes, HVS (hardware video scaler) uses 12 bit values for X, Y, WIDTH, HEIGHT.
X & Y are signed so limited to 2047.
WIDTH and HEIGHT are unsigned so limited to 4095.