raspberrypi/firmware #807: DMT Mode 35 and 47 No Longer Works after Upgrade

yogiman-uk, 2017-05-01:
Hello

I have:

Raspberry Pi Model B (256MB) Model.
LG 19LS4R TV/Monitor

I have always run the monitor in it's native resolution of 1440x900@60Hz via HDMI (No audio via DMT Mode) So Audio is transferred using the audio output Jack.

The config.txt is configured using:

hdmi_group=2
hdmi_mode=47
hdmi_drive=2 (Changes make no difference on this TV)
config_hdmi_boost=4 (Changes make no difference on this TV)

Two days ago I performed a sudo apt-get update and a sudo apt-get upgrade. Until this point my display worked perfectly.

However after the upgrade was performed I rebooted the Pi.

Once rebooted my TV displayed No Signal....

I have run through and tested all of the the CEA and DMT modes that the TV/Monitor is capable of displaying and I have found that all display except for DMT modes 35 and 47, Mode 47 being the native mode for the display and worked perfectly fine until the upgrade.

Does anyone have any idea how to fix this and why the mode has suddenly become unavailable after the upgrade?

I have posted on the Pi forum and I was redirected here by DOM, one of the moderators.

Many thanks for any help you can offer.

Yogiman

popcornmix, 2017-05-01:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this problem.

yogiman-uk, 2017-05-01:
Hi

Don't know if this will help or not. I have just created another SD Card with the original image that I used which was: 2016-11-25-raspbian-jessie-lite.img

Having not had time to run updates since that install, I ran upgrade on 29/04/2017

At this point after the reboot it broke.

I can confirm that the original image listed above boot seamlessly and requires no changes to the config.txt for me to get 1440x900 mode 47.

Does this help, can I offer any further assistance?

Regards

Yogiman

popcornmix, 2017-05-01:
Well, we know the firmware from around 2016-11-25 is good and the firmware from around 2017-04-27 is bad. I need to know when it changed from good->bad.

You can revert your updated image to the older firmware with rpi-update. Using the link I gave you will find this firmware release (https://github.com/Hexxeh/rpi-firmware/commit/a6ddca1723f9122280ded7cbb68581fa60c684bf) is from around the good time, so you can revert back to it:
``sudo rpi-update a6ddca1723f9122280ded7cbb68581fa60c684bf``
Reboot and test if that is okay. Assuming it is you want to pick a firmware roughly half way between the newest good and the oldest bad firmwares. There are 56 possibilities, but as you are halving the range each step it should only take about 6 tests to identify the exact firmware.

Once we know the firmware update that caused the issue it should be much easier to find a solution.

yogiman-uk, 2017-05-01:
popcornmix

I will start the process now and return when I have an answer.

Thanks for your help, I appreciate it.

Regards

Yogiman

yogiman-uk, 2017-05-01:
Popcornmix

Having followed your great instructions above I can confirm the following:

Commits up to and including:

kernel: Bump to 4.4.43 
commit: 2b0494cd4122571280a0776c6d52eed8c76c6017

Work fine, everything is as it should be and the system boots with my normal display resolution of : 1440x900 ergo DMT Mode 47.

The system appears to break once I commit:

firmware: dispmanx: Return failure when dispmanx_resource_create fail…
…s to allocate image
commit: a12b699d642096023a2b3009bb88ca82ad0baef4
[
Link to commit here.](https://github.com/Hexxeh/rpi-firmware/commit/a12b699d642096023a2b3009bb88ca82ad0baef4)

At this point I am back to No Signal and Mode 47 stops working correctly again.  I have not tested mode 35 this mode is available to the TV/Monitor but doesn't display well, thus I never use it.

Mode 47 is the recommended resolution for this display.

Applying commit: 2b0494cd4122571280a0776c6d52eed8c76c6017

Brings the display back in the correct resolution.

I look forward to your reply

Many Thanks

Yogiman

popcornmix, 2017-05-01:
It is probably:

> firmware: hdmi: Use hdmi drive when any hdmi modes are supported
See: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=169879

Can you report output of ``tvservice -s`` with the working and non-working firmware?

yogiman-uk, 2017-05-01:
Popcornmix

With the working firmware `tvservice -s` outputs:

state 0x120006 [DVI DMT (47) RGB full 16:10], 1440x900 @ 60.00Hz, progressive

With the non-working firmware `tvservice -s` outputs:

state 0x12000a [HDMI DMT (47) RGB full 16:10], 1440x900 @ 60.00Hz, progressive

Hope this helps

Regards

Yogiman

popcornmix, 2017-05-01:
And if you add ``hdmi_drive=1`` to config.txt on the non-working firmware what does ``tvservice -s`` report?

yogiman-uk, 2017-05-01:
Adding hdmi_drive=1 to the config.txt outputs:

state 0x120006 [DVI DMT (47) RGB full 16:10], 1440x900 @ 60.00Hz, progressive

At this point my display has now returned.

So I assume the changes they have made to default hdmi_drive=2 have caused the problem, so the report in the original post that no one used hdmi_drive=1 by default is not strictly correct?

If this is the case, then can I sudo apt-get upgrade again to bring the OS fully up to date and just apply the hdmi_drive=1 to the config.

Or am I missing the point?

Thanks

Regards

yogiman

yogiman-uk, 2017-05-01:
Popcornmix

With previous firmware making changes to hdmi_drive=2 had no effect with this display as previously mentioned.

So I assume some change within the firmware now requires this.  As this screen is and has always been a pain in the rear end I would suggest the problem is really mine and not an issue for you.

So I am happy to take note and make the required changes to enable my display to work. I will to the upgrade again and see what happens, I assume at this point everything will be fine, if not I will of course let you know.

Thank you for taking your time to help me with this, I really do appreciate your assistance. I hope you enjoy the rest of your day :)

Regards

Yogiman

popcornmix, 2017-05-01:
I would expect with previous firmware that ``hdmi_drive=2`` would cause the same issue as latest firmware has. If it doesn't can you report output of ``tvservice -s`` using the old firmware with ``hdmi_drive=2`` present?

yogiman-uk, 2017-05-01:
Latest firmware

hdmi_drive=1

state 0x120006 [DVI DMT (47) RGB full 16:10], 1440x900 @ 60.00Hz, progressive

Works fine.

Changing firmware to:

firmware: dispmanx: Return failure when dispmanx_resource_create fail…
…s to allocate image
commit: a12b699d642096023a2b3009bb88ca82ad0baef4

hdmi_drive=2

Reboot gives me no display

tvservice -s outputs:

state 0x12000a [HDMI DMT (47) RGB full 16:10], 1440x900 @ 60.00Hz, progressive

hdmi_drive=1

gives me:

state 0x120006 [DVI DMT (47) RGB full 16:10], 1440x900 @ 60.00Hz, progressive

I have just checked back in my early notes for the Pi (This would be when I 1st purchased the device) and my notes clearly state that making changes to hdmi_drive appears to have no effect on the display. (This is what I am basing my earlier statement on).

So either I am completely loosing it or a previous version of the firmware made another difference.

I can tell you that when I 1st set up the pi I had to modify the config.txt to get my display to work, I found reference to an earlier post saying that although I eventually got the display to work I could only get audio via the 3.5mm jack and not the HDMI.

However at a later date I noticed when writing a new image to an SD card that once the machine booted by default no changes to the config.txt were required any more and the display just automatically changed to 1440x900 without issue, other than the fact I still needed to output audio via the 3.5mm jack.

So at this point I am somewhat confused to say the least and hope this all makes more sense to you than it does to me.....?

Thanks again

Yogiman.

popcornmix, 2017-05-01:
> I have just checked back in my early notes for the Pi (This would be when I 1st purchased the device) and my notes clearly state that making changes to hdmi_drive appears to have no effect on the display. (This is what I am basing my earlier statement on).
So either I am completely loosing it or a previous version of the firmware made another difference.

Well, all historical firmwares are available through rpi-update, so you can try to find a version that matches your recollection, but I would be a little surprised if you do.

There have been some tweaks to deciding what the desired resolution is over the years, so it is possible one of those has been beneficial in getting a picture out-of-the-box.

But if your monitor doesn't accept a DMT mode with HDMI drive, then audio through the HDMI cable will never be possible. Using a CEA resolution (e.g. 1280x720) would allow audio to be transmitted but possibly with reduced image quality.

yogiman-uk, 2017-05-01:
Having changed the firmware multiple times this afternoon I am about done with more changes.

With the latest firmware and hdmi_drive=1 I get the correct resolution and no audio through HDMI but that basically puts me back to 6 years ago where I have to remember to set parameters in the config.txt before I am able to actually see a display.

Seems like a bit of a backward step but if the changes help more people than they hinder then I am afraid I think I need to be the person who puts up with the grief.

Thank you so much again for your help, I am sure you are sick of the sight of me for one day.

Enjoy the rest of yours

Regards

Yogiman

closed 2017-05-01 by yogiman-uk

RonnySvedman, 2021-03-21:
This thread was helpful to me today. I had the same inssue with a Finlux brand 20 inch lcd tv, that has a 1440x900 panel but falsely claims 1920x1080 in its EDID, and has no CEA mode at native resolution listed. turning on hdmi_drive=1 and choosing dmt mode 47 got me a crisp and stable picture using all of the panel. Thank you.

yogiman-uk, 2021-03-21:
Glad it worked for you Ronny :-)