raspberrypi/firmware #808: MMAL: vc.ril.isp refuses some types of width of bayer image

Terminus-IMRC, 2017-05-05:
vc.ril.isp refuses some types of width of bayer image. For example, 512 and 592 pass the port format configuration but 514 fails to do that.

Please try it with this code: https://github.com/Terminus-IMRC/mmal-isp-does-not-crop
This code setups vc.ril.isp to consume bayer image on input port and feeds buffer, which is all filled with 1s, to the port.

If `ISP_IN_WIDTH` is 514 for example, configuring isp->input[0] fails with this error:
```
mmal: mmal_vc_port_info_set: failed to set port info (2:0): EINVAL
mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0x14d6af0 (EINVAL)
main.c:84:main: config_port(input, ISP_IN_ENCODING, ISP_IN_WIDTH, ISP_IN_HEIGHT): 0x00000003
```

I think all width and height which is multiple of 2 should be accepted because the bayer is 2x2 pattern here.
Is there any reason for prohibiting 514?

Thank you for reading. And sorry in advance if I'm saying wrong.

6by9, 2017-05-07:
Bayer support is still very much alpha stage, as is most stuff in vc.ril.isp.

Bayer formats, and in particular the raw10 format, have some very funny packing arrangements, so strides may not come out as expected with the simple align up. I'd need to run the code and see what numbers fall out where - I'll do that tomorrow in the office.

6by9, 2017-05-08:
I'd actually left some logging on by default which tells most of the story. ```sudo vcdbg log msg```
```
038946.804: ip_is_supported_format: 514x512, stride 704, slice 512, size 2560
038946.825: Failing stride test. 704 != desired of 672
038946.865: mmal: mmal_ril_set_port_settings: ril.isp:in:0: failed to set port definition (4) buffers 1/1/2560
```
The settings validation code was missing handling for any of the Bayer formats, so was computing a different stride (pitch) to expected. Add in the fun with the rounding to sort packing 4 pixels in 5 bytes and it just mismatched.
I've got some changes to add in those formats. I'm going to amend your app to go through all widths that are a multiple of 2 and ensure that it is happy with all values. 

You do have a bug in your code in that you're not sufficiently filling the buffers, so the ISP component is rejecting them on width=514.
```
ISPIL: Buffer(s) provided too small. in needs 360448, got 344064. out needs 835584, got 835584
```
You've made an incorrect assumption over the stride and then set ```header->length``` based on that. In the current case it is much safer to set it to ```header->alloc_len```.

Likewise there should be a mmal_util function for width to stride computation rather than rolling your own. ```mmal_encoding_width_to_stride``` would be my expected candidate. If it doesn't give the correct results then that is a bug that ought to be resolved. I'll try to find a few minutes to run through the Bayer formats to ensure they do the right thing - I'm not sure they do at the moment as they don't round up.

JamesH65, 2017-12-22:
@Terminus-IMRC @6by9 Did this ever get fixed?

6by9, 2017-12-22:
vc.ril.isp is still really only beta stage for Bayer, but keeps on getting extra bits added.

https://github.com/Hexxeh/rpi-firmware/commit/a4e1434bfa3fe9daf7413e0d78b83afa236e102b fixed the mishandling of raw10. I've had a pending change for it for a while which reworks the stride calcs to use the standard helper libraries, but it still needs some more testing before releasing into the wild.

JamesH65, 2019-01-15:
@6by9 Did the pending change ever become law?

6by9, 2019-01-15:
Yes, merged back in July 2018.

JamesH65, 2019-01-15:
Closing this issue as questions answered/issue resolved.

closed 2019-01-15 by JamesH65