raspberrypi/firmware #809: MMAL: vc.ril.isp doesn't crop output image when input bayer image's width is not aligned-up to 32?

Terminus-IMRC, 2017-05-05:
It seems that vc.ril.isp doesn't crop output image when input bayer image's width is not aligned-up to 32.

Please try it with this code: https://github.com/Terminus-IMRC/mmal-isp-does-not-crop
This program:
* creates vc.ril.isp component,
* sets the input encoding to `BAYER_S*10P` and
* sets the output encoding to `RGB24`

and then it passes buffer (which is all filled with 1s) to the input port.

For simplicity, let `ISP_OUT_WIDTH` be `592` and `ISP_OUT_HEIGHT` be `512` here.
In this case `ALIGN_UP(ISP_OUT_WIDTH, 32)` will be `608`.
If the isp correctly crops the output image, the output header->length should be `ISP_OUT_WIDTH * ISP_OUT_HEIGHT * 3 = 592 * 512 * 3 = 909312`.
However, the isp actually sets header->length to `933888`, which seems to be `ALIGN_UP(ISP_OUT_WIDTH, 32) * ISP_OUT_HEIGHT * 3 = 608 * 512 * 3 = 933888`.
So it seems that vc.ril.isp doesn't crop output image when input bayer image's width is not aligned-up to 32.

Defining `ISP_IGNORES_CROPPING` to 1 on the code aligns-up the width on `save_image()`, and makes the output image look better.
`ISP_IGNORES_CROPPING=0`:
[image]
`ISP_IGNORES_CROPPING=1`:
[image]
Note that the output image should be all white if the isp is working correctly.

Is this reproducible on your pi?

Thank you for reading. And sorry in advance if I'm saying wrong.

6by9, 2017-05-07:
Bayer support is still very much alpha stage, as is most stuff in vc.ril.isp. However this looks more like a misunderstanding on how MMAL port formats work.
The buffer is always ```port->format->es->video.width*port->format->es->video.height``` in geometry. ```port->format->es->video.crop.width``` and ```crop.height``` set the portion of that buffer that should be treated as active pixels. In theory ```crop.x``` and ```crop.y``` set the origin of that active portion, but in practice is generally ignored on the VPU.

Terminus-IMRC, 2017-08-20:
Sorry for delay. Yes, that was my misunderstanding. Now I see that the buffer is not actually cropped on memory. Thank you for your reply!

closed 2017-08-20 by Terminus-IMRC