raspberrypi/firmware #810: CEC activity while VC4 is active

anholt, 2017-05-15:
Hans Verkuil has been working on an open source HDMI CEC implementation in the vc4 driver, but it seems that the firmware is still stealing the HDMI interrupt for CEC, and spamming the bus with traffic.  Could we get the firmware's CEC code disabled when the vc4 HDMI DT node is present?

popcornmix, 2017-05-15:
Yes, let me take a look.
A CEC driver sounds interesting.

anholt, 2017-05-15:
I gave him a `mask_gpu_interrupt1` line for now, and he's got the driver working.

popcornmix, 2017-05-16:
Which interrupt did you mask?

anholt, 2017-05-16:
0x100 (HDMI)

popcornmix, 2017-05-16:
We have a config.txt option ``hdmi_ignore_cec=1``
Does that also avoid the conflict?

anholt, 2017-05-16:
I've pointed Hans at the question, who said he'd have some time to play with this stuff again tomorrow.  I'm focusing on the fbturbo-replacement mode right now.

popcornmix, 2017-05-16:
Okay, vcgencmd get_config int shows that ``mask_gpu_interrupt1`` does include 0x100 when vc4-kms-v3d overlay is used.

But that gets set after the firmware hdmi initialisation is done, and I believe that includes sending an active source over cec which is probably the unwanted behaviour. I believe the interrupt will be disabled and GPU won't interfere with CEC once the arm is running.

I suspect that setting ``hdmi_ignore_cec_init=1`` may be sufficient for your issue as that will stop the initial active source CEC command, but would be useful if that can be confirmed.

anholt, 2017-05-16:
I wonder if Hans is using u-boot, with the firmware finding the downstream DT and the u-boot loading the upstream one, so that the firmware's vc4 detection didn't work.

anholt, 2017-05-16:
>> Dom had a question on that github issue if the hdmi_ignore_cec=1
>> config.txt option works as well as the config.txt I gave you.  Could you
>> give that a try?
>> 
>
> Doesn't work. This option seems to disable the CEC bus altogether. With just this
> option the interrupt is still handled by the firmware and I get timeouts because
> of that. If I set both options, then interrupts are handled fine, but I can no
> longer detect the TV.

Looking at the fw, hdmi_ignore_cec doesn't look like it does much.  I am confused by the behavior with both options, though.

popcornmix, 2017-05-16:
If ``hdmi_ignore_cec=1`` stops the kernel side CEC driver from working it sounds like it is relying on some setup being done by the firmware which it probably shouldn't do.

``hdmi_ignore_cec=1`` makes the firmware think the physical CEC address from edid is 0xffff which will generally disable the firmware side cec state machine.
It also stops the host (arm) from selecting passive mode (used by libcec) and so controlling cec.

``hdmi_ignore_cec_init=1`` stop the initial ImageViewOn and ActiveSource CEC commands from being sent (used to encourage display to come out of standby/switch to right input).

hverkuil, 2017-05-16:
Rather than going through Eric I can just give my feedback here :-)

It is entirely possible that there is a DT mixup in my boot sequence. I'll check that tomorrow.

hdmi_ignore_cec_init=1 is handy to know: I noticed that behavior when testing and it was quite annoying. Being able to turn this off sounds very useful.

hverkuil, 2017-05-17:
I'm using u-boot (i.e. config.txt has the line 'kernel=u-boot.bin'). I had a lot of old dtb's in that same sdcard partition, but I removed them all and replaced them with the same bcm2836-rpi-2-b.dtb I use with the kernel. It still has the same behavior as I described earlier.

One note: I am currently using a kernel based on the 4.11 drm code (but 4.12 media/cec code). If going to 4.12 makes any difference, then let me know (I was planning to switch anyway).

hverkuil, 2017-05-17:
BTW, is there anyway I can tell if there is a DT mixup?

pelwell, 2017-05-17:
> BTW, is there anyway I can tell if there is a DT mixup?

The kernel exposes the contents of the DT as a virtual filesystem under `/proc/device-tree`. You can browse that using the normal Linux tools or convert it to source or binary using `dtc`:
```
dtc -I fs -O dts -o source.dts /proc/device-tree
dtc -I fs -O dtb -o binary.dtb /proc/device-tree
```
The `dtdiff` utility compares source or binary DTs, allowing you to see how the live DT differs from the DT you think you are using. Note that the firmware makes some changes to the DT before launching the kernel; fortunately the `dtmerge` utility makes most of the same changes, so you can use it to prepare a "base" DT for comparison:
```
dtmerge /boot/bcm2836-rpi-2-b.dtb base.dtb -   # The '-' means no overlay
dtdiff base.dtb binary.dtb
```

popcornmix, 2017-05-17:
If you are not using u-boot and are using ``dtoverlay=vc4-kms-v3d`` then various firmware side hooks into hdmi/hvs/v3d will be disabled and you may have fewer problems.
If not then you should disable them manually (e.g. with mask_gpu_interrupt1).

anholt, 2017-07-26:
With an upstream 4.13-rc1 DT with VCHI added, and no u-boot to confuse things, I get this from vcgencmd:

mask_gpu_interrupt0=1024
mask_gpu_interrupt1=26370

So it does look like we're getting the right things masked there.  With the pulse8 inline and cec-ctl --playback; cec-ctl -S, I see the pulse8 show up.  (without the first, it didn't show anything)

@hverkuil can we close this one?

markfirmware, 2017-09-13:
@anholt @hverkuil There is a programming system http://ultibo.org that compiles pascal into kernel7.img without any run-time dependencies. It supports a lot of the rpi periperhals. I'd like to add support to it for receiving cec button presses from an hdmi tv's remote controller. I've done it with cec-client on linux. Can you advise on where to start? Thanks, Mark

hverkuil, 2017-09-13:
@markfirmware: key presses are automatically routed to a remote control input device (assuming that's compiled into the kernel with config option MEDIA_CEC_RC, which it usually is). So if the CEC adapter is configured (cec-ctl --playback), then afterwards you can just listen to keypresses. No CEC knowledge needed, it works just like a remote control.

To be honest, I've never worked with input devices for a remote control, so you would have to do some googling for that.

The alternative is if you want to use CEC directly. In that case you need to dig into the API:

https://linuxtv.org/downloads/v4l-dvb-apis-new/uapi/cec/cec-api.html

The cec utilities maintained in v4l-utils (https://git.linuxtv.org/v4l-utils.git/) are a good starting point for example code and to experiment with CEC.

For both CEC and remote control questions you can ask those on the linux-media mailinglist. See linuxtv.org for that as well.

hverkuil, 2017-09-13:
@anholt : we can close this bug report, @markfirmware's comment is unrelated to this.

closed 2017-09-13 by popcornmix