raspberrypi/firmware #811: Support for PAL60 in config.txt/sdtv_mode

manekinekodesu, 2017-05-22:
I use my Raspberry Pi 3 to emulate consoles. And my crt tv only supports PAL/PAL60 modes. Most games were not/badly optimized for PAL 50 Hz mode. And PAL60 can be used to play NTSC content full screen and full speed, as it was meant to be played.

popcornmix, 2017-05-22:
So with your display do you get no picture when using NTSC output?

I'm not an expert on composite TV out and I'm certain if the hardware can handle this. 
Do you know exactly what the TV wants for PAL60?
Is this the NTSC resolution and framerate, but with the PAL colour carrier frequency?

deborah-c, 2017-05-22:
Wikipedia's actually a pretty good resource for this: https://en.wikipedia.org/wiki/PAL

manekinekodesu, 2017-05-23:
I get black and white picture, both with NTSC and Brazilian PAL-M.

Yes,  PAL60 has NTSC resolution and framerate, but with the PAL colour carrier frequency. 
Wikipedia's all the specs:
https://en.wikipedia.org/wiki/PAL#Multisystem_PAL_support_and_.22PAL_60.22

> A "PAL 60" signal is similar to an NTSC (525/30) signal, but with the usual PAL chrominance subcarrier at 4.43 MHz (instead of 3.58 as with NTSC and South American PAL variants) and with the PAL-specific phase alternation of the red colour difference signal between the lines.

popcornmix, 2017-05-23:
Yes, I was using wikipedia. 
The "PAL-specific phase alternation of the red colour difference signal between the lines" is the part that I'm unsure of. Nothing in the peripheral spec sounds like that so it is possible it can't be independently changed.

deborah-c, 2017-05-23:
That's more or less the definition of PAL (Phase Alternating Lines) -- the R-Y chrominance signal alternates phase between lines. If there's an NTSC/PAL switch, it should cover this, I think.

popcornmix, 2017-05-23:
@manekinekodesu could you give this firmware a test:
https://drive.google.com/uc?id=0B-6zmEDJwxZEU3V2RkxMTzktS3M&export=download
By default no change, but:
``sdtv_mode=0x22`` should give PAL 60
``sdtv_mode=0x20`` should give NTSC 50

Basically for PAL 60 I'm using normal PAL settings, but have changed resolution/framerate to NTSC.
Not sure if that is correct, but my composite TV does show a picture (with colour) in the new modes.

manekinekodesu, 2017-05-23:
Tested this on 2 pal crts(one of them supports ntsc signal over composite in but colors look off quite a bit).
Results are:
sdtv_mode=0x22 - 60Hz, fullscreen, no color (on my second tv I can see that colors actually  in NTSC format)
sdtv_mode=0x20 - 50Hz letterbox, PAL colors (just like the usual PAL)

popcornmix, 2017-05-23:
How can you tell colours are NTSC format?

Any different with https://drive.google.com/uc?id=0B-6zmEDJwxZEVmhhWjNOZGRPWHc&export=download ?

manekinekodesu, 2017-05-23:
I can tell it by the way my second TV displays NTSC colors - they are off (the green is almost blue etc)

Results from the second firmware are closer:
sdtv_mode=0x22 60 Hz check, fullscren check, PAL color check, but there is a problem: image is statically skewed:
[image]

sdtv_mode=0x20 looks like 50 Hz, monochrome, picture is also skewed, albeit in another direction:
[image]

popcornmix, 2017-05-23:
Can you try: https://drive.google.com/uc?id=0B-6zmEDJwxZENTd3bWZsMktLME0&export=download

I notice that PAL has a horizontal front porch of 20 and NTSC has 14.
Those 6 pixels seem to match the slope you are seeing.

manekinekodesu, 2017-05-24:
Yes, that fixed it. Now the picture looks totally PAL60 to me! 60 Hz, fullscreen and PAL colors!
[image]
Could progressive scan be enabled for this mode as well?

And NTSC50:
[image]
I gues this TV has not been designed with NTSC50 in mind.

popcornmix, 2017-05-24:
Great. I'm not sure how defined NTSC 50 is, so it's hard to say if the TV doesn't support it or we are generating the wrong thing.
If anyone has a source that generates NTSC 50 correctly and a TV that displays it we could be more sure.

I've tried progressive in addition to PAL 60 and get a rolling display - is that what you see?
I can't think of anything obvious that causes this and adjusting the vertical front/back porch doesn't seem to help.

manekinekodesu, 2017-05-24:
@popcornmix How do I try progressive in addition to PAL 60?

In sdtv_mode=18    I get PAL 50 and rolling display (slowly rolls downward)

popcornmix, 2017-05-25:
You add 0x10 for progressive and 0x20 for "alt mode" (i.e. flip PAL/NTSC resolution/framerate). So:
sdtv_mode=0x30 for progressive NTSC 50
sdtv_mode=0x32 for progressive PAL 60
I don't think they will work (they don't for me) but I'd be interested if you see the same as me.

manekinekodesu, 2017-05-25:
Ok, I've connected my rpi 3 to TV tuner card and recorded some test videos, hope they will be of some help to you:
I've set sdtv_mode=0x32 for progressive PAL 60
On my tv capture card I've set "video standard = PAL 60" and "vertical synchronization = Normal tracking/Fast tracking":
[image]

Then I've set "vertical synchronization = Free running mode (disable tracking)":
[image]

So this display rolling has to do something with vertical sync. I could not find Pal 60 modeline, but here is a bunch of other useful modelines [http://www.geocities.ws/podernixie/htpc/modes-en.html](http://www.geocities.ws/podernixie/htpc/modes-en.html). I don't know if they are of any help though.

What's interesting, in your second firmware, normal progressive modes 0x10-0x12 are also broken:
[image]
Should they be that way?

On my crt tv 0x32 mode looks like this:
[image]

popcornmix, 2017-05-25:
Yes, I think I spotted the "slow roll" issue with progressive mode and fixed it. Try:
https://drive.google.com/uc?id=0B-6zmEDJwxZEWW5HT0hSaFNxb0E&export=download

manekinekodesu, 2017-05-25:
Yep, modes 0x10-0x12 are fixed.

Spotted something weird in the 0x32 mode:
[image]

I've also hooked up my pal60 modded sega mega drive 2 to the tv tuner card and there were no such artifacts.

manekinekodesu, 2017-06-04:
I've tried to use 0x22 mode recently and  noticed that it runs a bit slower 59.5  (causing sound crackling ) instead of 60.00 like 0x00 mode. Then I've tried 0x20 mode which runs faster than 0x02 mode, 50.35 instead of 50.00 and it's also offset from the top a little so the bottom gets cut and warped to the top.

Maybe something else needs to be swapped, like sync pulse width for example. The 0x22 mode can be sped up to 59.94 - 60.00 by decreasing  sync pulse width and horizontal front & back porches or maybe increasing dot clock frequency.
@popcornmix You've said that you are able to manually adjust vertical porches. So, is it possible to tweak other values, as I've suggested above? Maybe it will help to rid of the rolling in progressive modes.

Also: 

> The symptom of a misplaced sync pulse is that the image is displaced on the screen, with one border excessively wide and the other side of the image wrapped around the screen edge, producing a white edge line and a band of "ghost image" on that side. A way-out-of-place vertical sync pulse can actually cause the image to roll like a TV with a mis-adjusted vertical hold (in fact, it's the same phenomenon at work).
[Black Magic and Sync Pulses](http://en.tldp.org/HOWTO/XFree86-Video-Timings-HOWTO/magic.html)

c0ommandc0m, 2017-09-06:
Hi @popcornmix , @manekinekodesu . I just got into this thread looking also for PAL 60hz.

Amazing work you're doing here!!!

Really, really needed to get things running smoothly on a SD TV here in PAL areas...

Got the latest firmware and tried 0x22. And as @manekinekodesu  posted, all good but some flashing on some aras of the screen, depending on the intensity and contrast of colors. Could it be just luminance is being updated oddly?

With 0x32 got the same lack of V sync. Both modes (PAL 60 480i / 240p) would be such a great addition. Actually, on the Gamecube & Dreamcast era it was quite common to get games that asked you if your monitor supported PAL 60 and, in case they did, they'd do the switch in order to run 'as intended'.

Tested all on my Sony KX14CP1 CRT, a very good one I must say!

If there's anything I can test or help you in any other way, please let me know!

JamesH65, 2017-12-22:
@popcornmix Taken a  quick look at the code but difficult to tell what has been changed and/or added to master branch related to this issue Any status update on this?

Sobakin76, 2018-06-14:
@popcornmix I have a good 15" composite monitor JVC TM-H150C and RPi3b, with firmware downloaded from this thread posted @May 25 it works in PAL60 mode but only in mode 0x22 (interlaced) and flickers alot, but in mode 0x32 (progressive) gets out of sync. Monitor's OSD says "PAL", tvservice -s says PAL 60Hz and NTSC (720x480) resolution and picture size and proportions are the same as NTSC original mode. This mode is quite better than original NTSC because a lot less color artifacts but more compatible for retrogaming than PAL 50Hz (in cga/ega games used color dithering in NTSC modes monitor shows gradient colors but OK in PAL and PAL60Hz, but in PAL too much spacing in top and bottom and flickering as it's only 50Hz). Also at PAL_M modes with this firmware it shows currupted picture, with original latest firmware it shows correct in luma but rainbow colors and monitor recognized it as NTSC.
I would appreciate if You add support of PAL60 progressive. I'm ready to test anything You need for it.

pietch, 2018-08-14:
I'm trying to set CVBS output to PAL-N (south america):  625 lines, 576 active lines, 50Hz, 3.58205625 MHz color carrier. I can't find any documentation.
Is this combination supported by RPi hardware?

popcornmix, 2018-08-15:
@pietch no, PAL-N is not supported.

elfoam, 2018-11-03:
I'm in the same situation as Shaum257, I have a CRT which will run 60hz but can't display the NTSC colours, I would be very happy if the PAL 60hz 288p mode could be added to the PI3b+. That would make that TV very useful when right now I can't use it for much at all... Well of course I could buy a 3b.. but I have the 3b+

Antiriad, 2018-11-10:
@popcornmix excuse me if I write here but issue #683 it's closed and this is somehow related.

In sdtv_mode=16 retrogames with a resolution of  320x240 (or 320x224) are blurry: ok, that's probably because, I suppose, RPi is outputting 720x480. 
But if I set same games to a multiple integer resolution of 640x480 (or 640x448) image is sharper but a lot (really a lot) of "rainbow effect" occurs. 

It seems that this does not happen in PAL mode, is this normal? thank you!

shaum257, 2018-11-10:
@elfoam I'm also still very interested in PAL 60 video modes on my Raspberry Pi 3B+. I've commented like 2 months ago and I'm still waiting for a satisfying answer. If only @popcornmix could reply and put a download link for the working firmware files for the RPi 3B+, I would really appreciate if he does that. But yeah I'll be still keeping check for this issue #811. Hopefully our saviour @popcornmix will reply and put a download link for the firmware files soon. If that day ever comes, then I'll be surely partying hard after!!! lol

pelwell, 2018-11-10:
If this was easy it would have been done already. popcornmix has already put quite a bit of time into this issue, along with various other PAL and NTSC variations, but we are restricted by the modes provided by the VEC block.

elfoam, 2018-11-11:
It seems like the firmware he made for the 3b does everything I want anyway, although I'm not sure the 288p pal 60 was ever 100%? I'll buy a 3b and try it anyway if the 3b+ isn't ever getting this. Like a lot of people I'm done with LCDs and going back to CRTs only

pelwell, 2018-11-11:
If it is simply a case of adding some simple changes that have been shown to work to the mainline builds, and if those changes aren't at the expense of other functionality, then I'd say you have a chance. But this is a relatively niche area and it has to compete for time with lots of other work, so don't hold your breath.

Antiriad, 2018-11-11:
@elfoam progressive PAL-60 is perfectly working on my RPi 3. Video quality is outstanding, compared to progressive NTSC with a lot of rainbow banding.
Unfortunately in PAL-60 I can't get a full screen viewport in Retroarch for NTSC games, so I really hope that what you wrote (PAL coloring method in NTSC progressive video mode) is doable. I hope that @popcornmix will read this. ;)

elfoam, 2018-11-11:
I know PAL colours with 60hz on NTSC resolution is against all standards etc but it would be perfection for composite inputs and old CRTs. It could be just mode... Awesome rather than any standard lol

Sobakin76, 2018-11-12:
> @elfoam progressive PAL-60 is perfectly working on my RPi 3. Video quality is outstanding, compared to progressive NTSC with a lot of rainbow banding.

Which settings do You use to turn PAL60 progressive mode? And which firmware?
I tried old experimental firmware and in PAL60 worked only interlaced mode and it was flickering.

Antiriad, 2018-11-12:
@Sobakin76 @elfoam I'm using latest stable release of LAKKA (www.lakka.tv) and so the firmware packed with it. 
No issues at all with PAL-60: picture is stable and colors are perfect, no rainbow banding at all (sdtv_mode=0x32).
Unfortunately it's nearly impossibile to get a viewport for NTSC games that is in fullscreen and without non-integer scaling artifacts.

Sobakin76, 2018-11-12:
> @Sobakin76 @elfoam I'm using latest stable release of LAKKA ([www.lakka.tv](http://www.lakka.tv)) and so the firmware packed with it.
> No issues at all with PAL-60: picture is stable and colors are perfect, no rainbow banding at all (sdtv_mode=0x32).

Are You sure it's output is really PAL60? My Raspbian in mode 0x32 runs really in progressive PAL50 (the same as in 0x12), just scaling 240p output to 288 (if "framebuffer_height=240")

Antiriad, 2018-11-12:
@Sobakin76 well I just read this comment of @popcornmix and I really didn't notice speed differences (50hz/60hz) with sdtv_mode=18:

> You add 0x10 for progressive and 0x20 for "alt mode" (i.e. flip PAL/NTSC resolution/framerate). So:
> sdtv_mode=0x30 for progressive NTSC 50
> sdtv_mode=0x32 for progressive PAL 60
> I don't think they will work (they don't for me) but I'd be interested if you see the same as me.

I'll check again ASAP.

elfoam, 2018-11-13:
Just check is the scrolling is smooth in Strider or Sonic the hedgehog and you'll soon see if it's 50 hz or 60hz

elfoam, 2018-11-13:
I just tested sdtv_mode=19 on my 3b+ and it's 60hz for sure, all the colours are rainbowed but otherwise that seems to have potential. Maybe the rainbow is just because my TV wont do the Brazilian colours?.. I'll try on my trinitron.  sdtv_mode=0x22 is 50hz standard interlaced PAL.  sdtv_mode=0x32 is also 50hz but 240p so neither of those are working / do what they are supposed to. If I had good colours rather than rainbows on sdtv_mode=19 would be winning though.

elfoam, 2018-11-13:
I get the same rainbow colours on the brazilian modes on one of my LCD tvs. I'll keep trying other sets.

elfoam, 2018-11-13:
Ok on the Trinitron where I can switch between PAL/NTSC 3.58 and NTSC 4.43, it's NTSC 3.58 that gives the rainbow colours. PAL and 4.43 are black and white in those mods (3 and 19). This is still all on the 3b+.  Mode 19 is 60hz but I don't know if it's working on Brazilian sets as far as colour goes. Both standard PAL 60 modes though are broken for sure on the 3b+.  I still think we only need one mode=awesome that is standard pal colours on 60hz 240p. That would be the best pi update of all time if that's even possible.

Sobakin76, 2018-11-13:
> @Antiriad @Sobakin76 I just tested sdtv_mode=19 on my 3b+ and it's 60hz for sure, all the colours are rainbowed but otherwise that seems to have potential. Maybe the rainbow is just because my TV wont do the Brazilian colours?.. I'll try on my trinitron. sdtv_mode=0x22 is 50hz standard interlaced PAL. sdtv_mode=0x32 is also 50hz but 240p so neither of those are working / do what they are supposed to. If I had good colours rather than rainbows on sdtv_mode=19 would be winning though.

50hz can't be 240p, it's 288p (@ standard 15.7kHz)
type "tvservice -s" and You'll see standard PAL 720x576 @50Hz instead of 720x480 @60Hz for NTSC, PAL60 or PAL-M.

JamesH65, 2019-01-11:
I'm a bit confused as to the state of this issue, have all the questions been answered?

elfoam, 2019-01-12:
> 
> 
> I'm a bit confused as to the state of this issue, have all the questions been answered?

@JamesH65 nah.. the video code in the PI is broken. I've moved to using the Nintendo Wii for my emulation for now

JamesH65, 2019-01-12:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

Sobakin76, 2019-01-15:
I lost all hopes here and moved to VGA666 120Hz 240p on 17" CRT VGA monitor (which I found much better image quality than composite because its true RGB and with no flicker, and its absolutely cheap solution now), but still interesting in PAL60 and/or PAL-M support in RPi firmware. I still own two composite 15kHz CRT monitors one of it 100% support PAL-M and other PAL60, and I ready to participate in testing if needed.

AlistairThom, 2019-01-19:
> I lost all hopes here and moved to VGA666 120Hz 240p on 17" CRT VGA monitor (which I found much better image quality than composite because its true RGB and with no flicker, and its absolutely cheap [...]

Unfortunately, using a solution which utilises GPIO pins to generate video isn't viable for me, and I suspect many other retro gamers in the target audience for PAL60/PAL-M, who already use GPIO pins for arcade/console controller setups. 

The prevalence of affordable/free CRT TVs ready to be given new life, makes PAL60/PAL-M from the composite port a desirable option for people searching for a cheap and straightforward entry-point to the glorious nostalgia of CRT gaming.

shaum257, 2019-01-21:
I still hope that @popcornmix releases a firmware for Raspberry Pi 3B+ for PAL60. So far I've gotten used to games running in 50HZ but the smoothness of 60HZ would be the definitive way to play retro games on a Pi 3B+. I mean if @popcornmix managed to release a firmware for the Pi 3B, then there shouldn't be any problem releasing a firmware for the Pi 3B+ for PAL60. But yeah it's still up to @popcornmix if he feels like releasing it.

elfoam, 2019-01-22:
That would be awesome JamesH65. CRTs are becoming hugely collectable right now, there is still plenty around for gamers but you have to hunt them out. Rather than have to get a SCART hat for the Pi, PAL 60 would overcome a lot of the dodgy picture quality NTSC gives over composite.  I've kinda moved on myself as I'm looking for more high end picture quality and I'm using Nintendo Wiis over component and RGB and also building some SCART hats for the pi. But for guys that are starting out on CRT gaming with a Pi PAL 60 on composite would be very nice. I'd be happy to help with any testing on the 3b+

JamesH65, 2019-01-23:
OK, so have something that appears to be PAL60, firmware here https://drive.google.com/open?id=1B-tlXmTXxJ5xYrFFS85Def2f85YSvp18

Please try out and report back.

AlistairThom, 2019-01-23:
Thanks @JamesH65

I’ll be able to test shortly. Which sd_tvmode should I use? 19?

shaum257, 2019-01-23:
@JamesH65 I tried the firmware you have published and it's still 50HZ. I even tried setting the sdtv_mode to 0x22 and it still gives me 50HZ ingame. And another thing is that you have only published the start_x.elf file whereas @popcornmix published more files than just start_x.elf for the firmware but yeah as I said, I tried the firmware without any success. It could be that I've missed something. I just don't know what else to do.

JamesH65, 2019-01-23:
To use start_x.elf  you need to have `start_x=1` in config.txt othereisse it will use start.elf. Might be some other files I need to supply, will check.

I tested mode 0x22, which should be NTSC carrier with PAL colour at 60Hz. It's the 0x2x but that is important - that swaps the NTSC/PAL settings as Dom I think described above.

JamesH65, 2019-01-23:
Dom did appear to make other changes to do with porches, I've not done that. When I tested on the composite monitor I have here,  tvservice -s claimed PAL on NTSC at 60Hz, and the sync on the monitor was correct.

shaum257, 2019-01-23:
@JamesH65 You are a genius. By following your instructions and also by setting the sdtv_mode to 0x22 it now finally shows 60HZ in the retroarch menu instead of 50HZ when going to the video options. Now I am able to play every game in 60HZ with PAL colours without any compromises and I couldn't be happier. Thank you very much for everything!!! :)

JamesH65, 2019-01-23:
I think genius is a bit strong, but I'll take it. Glad it's working. Would appreciate any other testers with different composite kit giving it a once over. 

Note, I do not believe this works in progressive mode (0x32) but worth trying.

AlistairThom, 2019-01-23:
I've only had chance to do a _very_ quick test, but so far, so good in both 0x32 progressive, and 0x22 interlaced modes on my CRT. I'll be able to dig deeper at the weekend.

Forgive the pun, but this is game changing. 

Thanks again James.

JamesH65, 2019-01-24:
That's good news, wasn't sure about progressive. Will need some testing from other people on other devices, but since this is an entirely backwards compatible change, I'm sure it will go in the release firmware fairly soon.

elfoam, 2019-01-24:
Very nice, I have had a quick look tonight but couldn't get it to run at
60hz yet on my 3b+ but PAL colours are working. My retroarch is showing
60hz but the pi is running at 50, Retroarch isn't giving me video options
though like it does on the Wii and being over 40 celcius here today.. I
can't deal. I will check it again tomorrow once it cools down to see if I
can figure it out :)

JamesH65, 2019-01-24:
If PAL colours are working and you have 0x22 for the mode, and you have start_x=1 in config.txt, then I cannot see how it cannot be running at 60Hz. 0x22 is basically NTSC60 with PAL, so will be running at 60Hz.

elfoam, 2019-01-24:
Yeah I must be doing something wrong, I hashed out start_x=1 and nothing is
any different than with it enabled. So it seems to be not loading that new
video code.It's definitely running in 50hz, I have a very good eye for
that, can see it from a mile away.

These are the relevant changes I've made on the 3b+

-rwxr-xr-x 1 root root 3941508 Jan 24 22:16 start_x.elf
-rwxr-xr-x 1 root root 4047876 Oct  8 07:40 start_x.elf.bak

start_x=1

#60hz PAL 480i
sdtv_mode=0x22

I can switch it to NTSC and it runs 60hz no problem

Maybe I need to update the firmware on the pi or something I haven't used
it in a while so it might not be on the same spec you guys are running.
I'll try that.

JamesH65, 2019-01-24:
Odd. Not sure what that can be, When you do `vcgencmd version` what numbers do you get out?

[88 more replies]

closed 2025-05-24 by manekinekodesu