raspberrypi/firmware #812: ArchLinux ARM and "vc4-kms-v3d" from the "linux-firmware" package

TheAifam5, 2017-05-23:
Hey,
Yesterday and today i had a lot of problems with that dtoverlay. I have Raspbery Pi 2 and official RPi2 Touchscreen 7" .
When I use "vc4-kms-v3d", XOrg sees the display but does not rendering - its black but everything works when i connect display to the HDMI port. XOrg doesn't reporting any issues.
But when i use "vc4-**f**kms-v3d" everything works fine but i heard that is **experimental** actually.
Also i see that resolution is 2px smaller 799x479 (should be 800x480).

More info here: https://github.com/WebReflection/benja/issues/13

Regards,
Mateusz.

popcornmix, 2017-05-24:
The vc4-kms-v3d doesn't use the firmware for display.
Issues with vc4-kms-v3d should be reported here: https://github.com/anholt/linux/issues/

WebReflection, 2017-07-18:
I can confirm, using `vc4-fkms-v3d` fixed the issue out of the box.

This can be closed.

closed 2017-07-18 by popcornmix

xaviergmail, 2018-07-26:
I still suffer from this and unfortunately fake kms gives Kivy a really tough time, where as kms doesn't. Are there any known workarounds?