raspberrypi/firmware #813: Rapsberry pi 2 - Frame buffer problem (with HDMI port)

AAlphaDEV, 2017-05-25:
Hi, I hope this is an appropriate place to ask. I'm working on a bare-metal program for the raspberry pi 2 and I have setup a frame buffer with the mailbox property interface (the interface works well) but I can't  draw anything to the screen. I tried first wih the qemu emulator, everything worked well, then I moved on a classic 1280x1024 screen (with HDMI to VGA adapter), the GPU draw the colored square but when a allocate the frame buffer nothing can be done, the screen is black (not off just black). Also I noticed that the frame buffer address looks strange : 0xfd83a000, is this normal ?
I stuck on this since few days and I don't know how I can fix it. Anyone have and idea ?

popcornmix, 2017-05-25:
Have you tried the [bare metal forum](https://www.raspberrypi.org/forums/viewforum.php?f=72)?
Bare metal code writing to the framebuffer is frequently discussed.

AAlphaDEV, 2017-05-26:
I will try, thanks for answering.

closed 2017-05-28 by AAlphaDEV