raspberrypi/firmware #814: Official 7'' screen recognised as 720x480 on RPi3

soswow, 2017-05-26:
I am not sure where to report this problem. I found couple other people posting this problem on different forums, but with no responses.
I reported this and other combined problems here https://github.com/anholt/linux/issues/99
Defining 'vc4-fkms-v3d' overlay helped with colored pixels. But I don't know why I still have 720x480. Screen looks vertically squeezed.

```
$ vcgencmd version
May 15 2017 17:01:07
Copyright (c) 2012 Broadcom
version 9469ea3706e34c4de62f38a5008f69a429b4b43e (clean) (release)

$ uname -a
Linux raspberrypi3 4.9.29-v7+ #1000 SMP Sun May 21 20:11:31 BST 2017 armv7l GNU/Linux
```

What else is needed from me?

popcornmix, 2017-05-26:
Remove any config.txt settings (i.e. no dtoverlay or other display settings) and reboot.
Report output of:
```
tvservice -s
fbset
sudo vcdbg log msg
```

soswow, 2017-05-26:
config.txt now this: (I thought memory shouldn't be a problem?)
```
dtparam=i2c_arm=on
dtparam=spi=on
dtparam=audio=on
start_x=1
gpu_mem=250
```

```
pi@raspberrypi3:~ $ tvservice -s
state 0x400000 [LCD], 800x480 @ 60.00Hz, progressive
pi@raspberrypi3:~ $ fbset

mode "800x480"
    geometry 800 480 800 480 32
    timings 0 0 0 0 0 0 0
    rgba 8/16,8/8,8/0,8/24
endmode

pi@raspberrypi3:~ $ sudo vcdbg log msg
001170.309: HDMI:EDID error reading EDID block 0 attempt 0
001171.571: HDMI:EDID error reading EDID block 0 attempt 1
001172.829: HDMI:EDID error reading EDID block 0 attempt 2
001174.086: HDMI:EDID error reading EDID block 0 attempt 3
001175.344: HDMI:EDID error reading EDID block 0 attempt 4
001176.601: HDMI:EDID error reading EDID block 0 attempt 5
001177.859: HDMI:EDID error reading EDID block 0 attempt 6
001179.116: HDMI:EDID error reading EDID block 0 attempt 7
001180.374: HDMI:EDID error reading EDID block 0 attempt 8
001181.631: HDMI:EDID error reading EDID block 0 attempt 9
001182.653: HDMI:EDID giving up on reading EDID block 0
001198.247: HDMI:Setting property pixel encoding to Default
001198.267: HDMI:Setting property pixel clock type to PAL
001198.286: HDMI:Setting property content type flag to No data
001198.305: HDMI:Setting property fuzzy format match to enabled
001742.269: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001742.294: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
001742.606: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001743.869: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001745.133: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001746.397: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001747.659: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001748.923: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001750.186: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001751.449: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001752.712: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001753.975: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001755.001: hdmi: HDMI:EDID giving up on reading EDID block 0
001755.035: hdmi: HDMI: No lookup table for resolution group 0
001755.051: hdmi: HDMI: hotplug attached with DVI support
001755.081: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001755.374: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001756.636: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001757.897: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001759.160: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001760.421: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001761.683: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001762.945: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001764.207: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001765.469: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001766.731: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001767.756: hdmi: HDMI:EDID giving up on reading EDID block 0
001770.096: hdmi: HDMI: hotplug deassert
001770.111: hdmi: HDMI: HDMI is currently off
001770.126: hdmi: HDMI: changing mode to unplugged
001770.147: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001773.197: *** Restart logging
001775.035: Read command line from file 'cmdline.txt'
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=9577b1c2-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
002115.190: Loading 'kernel7.img' to 0x8000 size 0x45d490
002118.843: No kernel trailer (run mkknlimg to fix) - assuming DT-capable
002125.385: Loading 'bcm2710-rpi-3-b.dtb' to 0x465490 size 0x44d8
002246.502: Loaded overlay 'rpi-ft5406'
002279.085: Loaded overlay 'rpi-backlight'
002310.262: dtparam: i2c_arm=on
002319.725: dtparam: spi=on
002327.602: dtparam: audio=on
003393.794: Device tree loaded to 0x2effb600 (size 0x49db)
003399.363: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
004740.506: vchiq_core: vchiq_init_state: slot_zero = 0xee580000, is_master = 1
004744.755: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
105834.934: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
105836.185: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
105837.435: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
105838.686: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
105839.937: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
105841.187: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
105842.439: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
105843.690: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
105844.941: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
105846.191: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
105847.207: hdmi: HDMI:EDID giving up on reading EDID block 0
```
(But as you can see on a screenshot from prev. issue - it shows 720x480 in config)

popcornmix, 2017-05-26:
Remove the dtparam options as well for now.

Was the screenshot captured when using either the kms or fkms dtoverlay?
Do you have the issue with both disabled?

Also report output of:
```
vcgencmd dispmanx_list
```

soswow, 2017-05-26:
```
$ vcgencmd dispmanx_list
display:0 format:XRGB8888 transform:0 layer:-127 src:0,0,800,480 dst:0,0,800,480 cost:534 lbm:0
```
Ok. It's strange. I removed everything and there is no colored pixels. But I believe when I first time opened config.txt there was kms overlay uncommented. When I replaced it with fkms problem went away, but seems like it is not there without anything.

Here are latest screenshots:
[image]
[image]

So it's clearly squashed in hight.
When I loging with VNC I get this:
[image]
Which is 800x480 =\

soswow, 2017-05-26:
Just in case I am going crazy, I took a calliper and measured some image of a square. Via VNC it's almost the same, where is on the screen it's 60.x mm v 56.y mm

popcornmix, 2017-05-26:
The official Pi LCD does not have square pixels.
The display is 155mm x  86mm = 1.802
Its resolution if 800x480 = 1.667
so the 60mm x 56mm square you see is expected.
https://www.raspberrypi.org/forums/viewtopic.php?t=120710&p=817889

soswow, 2017-05-26:
oh. ok. I was thinking this might be the case. I am not sure if I have a problem then ... 
Why is it showing to me 720x480 in the configuration UI?

popcornmix, 2017-05-26:
Is this in preferences/Configuration?
I suspect it is directly querying the HDMI/composite resolution (without hdmi attached composite will be configured for 720x480) rather than the LCD.
Sounds like a issue with raspberrypi-ui-mods. Best to report it here:
https://github.com/RPi-Distro/

soswow, 2017-05-26:
> Is this in preferences/Configuration?
Yes.

What about my config.txt had kms overlay being there from the begining? Where initial config.txt comes from? Is it possible that kms is there and it makes this pixilation problem?

popcornmix, 2017-05-26:
The kms driver is considered experimental, but provides much more functional 3D support from the desktop. I don't know why you had it enabled - it has never been a default option (but can be enabled through the configuration utility).
It is not currently recommended for the majority of users unless they need to run desktop 3D applications. If you do need to use it and have a problem (i.e. pixelisation) then the anholt repo is the place to report it as that is where development occurs.

soswow, 2017-05-26:
When I went to bed I remembered something vaguely that I might enabled it myself. =\

closed 2017-05-26 by soswow