raspberrypi/firmware #815: Raspberry pi 3 1600MHz cpu limit?

pomwtin, 2017-05-27:
Hello!

I mannaged to acheive a 1600MHz (http://hwbot.org/submission/3560621_pomwtin_hwbot_prime_broadcom_bcm2837_2202.26_pps ) stable clock on rpi3. I have a reason to believe that max CPU clock setting is locked somehow, as setting it 1MHz higher result in hold on splash screen, no matter how I rise the voltage or cool the board.

A while ago, an start.elf file was made for older model of raspberry pi, wich unlocked CPU speed to 2000MHz (posted by user dom here https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=69929 )

Would it be possible to "unlock" CPU clock for RPI3 too preety please? As it seems to have quite some more potential :)

Regards
Martin

popcornmix, 2017-05-27:
I'm dom on forum. The 2GHz limit test build didn't actually work. I reported later:

> I had a look at this. It seems there is a 3.2GHz limit on PLLs and you need at least a divide by 2 on the output to get an even duty cycle.
This means >1600MHz doesn't (trivially) work.
I've increased the limit from 1500 to 1600 with latest rpi-update, so you can give that a try.

Afraid that is still the case - 1600Mhz is still the limit.

pomwtin, 2017-05-27:
Many thanks for fast reply and explanation, I appreciate it.
I guess I'll keep myself busy with other clock settings then :)

Have a nice day,
Martin

pomwtin, 2017-05-27:
I'd like to post another notice- if force_turbo is disabled and if frequency is higher than 1600 (1610 in this example) raspbian boots into GUI.
When under heavy load command `cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq` indeed returns 1610000. This would mean that actuall current CPU clock indeed is 1610MHz, right?
Now compared to your explanation above, this would not be possible as far as I understand...
 
I'm re-opening this "issue" in hopes that you will once again find some time to give me an answer.

Regards
Martin

popcornmix, 2017-05-27:
Use "vcgencmd measure_clock arm" to determine actual arm frequency.
scaling_cur_freq will only show what was requested - it may not have succeeded.

pomwtin, 2017-05-27:
Ah... yes this makes more sense now. Actual clock stayed on 600 MHz in case of 1610MHz in config.
Thank you once again, closing this issue now :)

closed 2017-05-27 by pomwtin