raspberrypi/firmware #816: USB HD reset

kochtopf, 2017-05-27:
Getting sporadic 'usb 1-1.3: reset high-speed USB device number 4 using dwc_otg' from my boot device (sda2). Going back to 4.9.27 fixed it. It does not break debian but during the device reset nginx operation stalls (serving 499). This 'feature' came with 4.9.28 and exist up til .30.

ghollingworth, 2017-05-27:
How did you confirm the kernel version, did you only update the kernel or update both the kernel and firmware using rpi-update ?

If you updated the firmware then it is possible that the problem is caused by the firmware rather than by the kernel...  If you can confirm which firmware update caused it we can identify the change

Gordon

kochtopf, 2017-05-27:
Used sudo rpi-update (Hexxeh/rpi-firmware):
kernel: Bump to 4.9.27 bb0292107df671bbeff7165518ba4bc5b6be7d12 --> tested OK (without resets)
kernel: Bump to 4.9.29 aeb5f6166d3ef7298f6fee644ac8cebb7a8fa133 --> tested BAD
kernel: Bump to 4.9.30 b81a11258fc911170b40a0b09bbd63c84bc5ad59 --> tested BAD
Thanks for reading

ghollingworth, 2017-05-27:
OK, there was a revision to the firmware at aeb5f616 will have to check on Tuesday what that was...

Gordon

popcornmix, 2017-05-27:
@ghollingworth "Change USB PHY settings to make device mode work correctly" looks most likely.

@kochtopf Can you return to latest rpi-update kernel/firmware and then replace boot files with these:
https://drive.google.com/uc?id=0B-6zmEDJwxZEc1RDOVEzb05fQUU&export=download

That reverts the named commit - I'd be interested if it helps.

kochtopf, 2017-05-27:
@popcornmix Done that. WORKED! No USB reset after boot. Will give it some time to confirm if the resets are gone...

closed 2017-05-28 by kochtopf

popcornmix, 2017-05-30:
This revert is now in latest rpi-update firmware.

kochtopf, 2017-05-30:
Thanks for this. Installed - tested- worked!!! You guys are **incredible**!!!