raspberrypi/firmware #820: FB MBOX flag for tiled scanout

anholt, 2017-06-06:
For the changes I'm working on to remove our need for compton, I'd like to be able to scan out V3D T-tiled buffers.  Could you add a flag to the fb mbox interface we use in fkms mode (https://github.com/raspberrypi/firmware/wiki/Mailbox-framebuffer-interface) to treat the fb as a tiled buffer?

popcornmix, 2017-06-06:
I assume you want VC_IMAGE_TF_RGBX32?
Any preference for how to request it? Set bit 31 of bits per pixel?

popcornmix, 2017-06-06:
If you can test:
https://drive.google.com/uc?id=0B-6zmEDJwxZENmVPQzlXajRxc2c&export=download
should give VC_IMAGE_TF_RGBX32 if you set bit 31 of bits_per_pixel.
It also will mask interrupts in the "txp" node.

anholt, 2017-06-07:
bit 31 of bpp sounds fine.  I've got a backport to test this evening.

anholt, 2017-06-08:
https://github.com/anholt/linux/tree/rpi-4.9.y-tiling is the backport kernel branch.  I'm getting tiled garbage being scanned out, in what looks like wrong colors, but which does respond to desktop activity (selection dragging on the file manager).  I also tried doing the | BIT(31) again after the pitch change workaround.

popcornmix, 2017-06-08:
If I take top 3 commits of that tree and add this:
http://paste.ubuntu.com/24808884/
and then run with ``vc4-fkms-v3d`` overlay, will that from boot write to a t-format buffer?
I can then manually change firmware to boot in "bit-31" mode and see what it is doing.

Initially I just tested with a linear framebuffer and checked the result became garbled (but still reacting to things changing). But I guess something is not set up right.

anholt, 2017-06-12:
Nope, you need updated Mesa (which hasn't landed upstream or in Raspbian yet) to start seeing T-tiled buffers show up in the kernel.  Could you send me the diff of what you did to the firmware?  I'll poke around and try to see if we need to change anything else.  (The pitch value is particularly likely to be a problem)

popcornmix, 2017-06-12:
Sent.

JamesH65, 2017-12-22:
@anholt @popcornmix Is this fixed? Closable?

anholt, 2018-02-09:
Yeah, we're using this in 4.9 and 4.14:
```
commit 9e4d7886d567713d9ac01acd0468dd62640cd9ab
Author: Eric Anholt <eric@anholt.net>
Date:   Wed Jun 7 14:39:49 2017 -0700

    drm/vc4: Add FB modifier support to firmwarekms.
    
    Signed-off-by: Eric Anholt <eric@anholt.net>
    (cherry picked from commit 11752d73488e08aaeb65fe8289a9c016acde26c2)

```

closed 2018-02-09 by anholt