raspberrypi/firmware #821: OpenMax  discarding Frames when converting to Mpeg-ts

MuhammadAsghar, 2017-06-08:
I have encoded live video and saved this to a file. This file plays fine but when i write the encoder buffer to mpegts container it generates errors.
1- Missing Picture in access unit with size 25
2- No start code is found
3- Error Splitting the input into NAL units.
If i save this MPEG-TS to a file it does not play with vlc and omxplayer.

popcornmix, 2017-06-08:
Please explain the commands you ran to record video to a file (raspivid?)
and the commands to convert the encoded bitstream to a mpegts file.

MuhammadAsghar, 2017-06-08:
I am checking my saved file through ffprobe.
[image]

MuhammadAsghar, 2017-06-08:
https://pastebin.com/DpDZ6vZt 
Here you can find my code of mpeg-ts which i am using to create mpegts files

F5OEO, 2017-06-08:
It seems that in your generated H264 ES stream you forget to add SPS/PPS header in order to be parsed by your TS encapsulator.
You don't explain all your encoding chain ! Which tool generate your mp4 file (drone.mp4) : openmax example seems most probable regarding your debug on screen.

If you write your own TS packetizer instead of using ffmpeg it is most probably to have low latency.
Note that I work on a similar project : https://github.com/F5OEO/avc2ts

MuhammadAsghar, 2017-06-08:
F5OEO thank you. I am using similar logic as you have. but i made some changes in it.my chain is 
  camera->  splitter--->encoder 1------>mpegts
                              |
                               -------->encoder2---->mpegts
I can put my code on paste bin. if there is need to check the code

F5OEO, 2017-06-09:
In an old version, I also implement multiple encoding with splitter. I
could review your code if you like !

MuhammadAsghar, 2017-06-09:
F5OEO thank you again.  I have posted my code on Pastebin and the link to the code is
https://pastebin.com/GMTFB6KB
The problem is when i save my encoded output to file and convert this file into mpeg-ts it works fine but when i put my live video to convert it into mpeg-ts it makes problem and also the file size of mpeg-ts file is much less than the file size of encoded file.

F5OEO, 2017-06-13:
Not a firmware issue. Should be closed @popcornmix . @MuhammadAsghar could continue conversation by mail (evaristec@gmail.com) , but be kind to respect GPL when you copy/paste my code !

MuhammadAsghar, 2017-06-27:
F5OEO can be please tell me where is the problem because both h264 encoder and my mpegts code are working fine when tried separately. if i take a saved file and convert it through my mpegts code it works.

JamesH65, 2017-12-22:
Closing as this issue is not firmware related.

closed 2017-12-22 by JamesH65