raspberrypi/firmware #822: Wifi does not work

Josar, 2017-06-08:
Try to get an new RPI3 to connect to a wifi network but this does not work.

Error listed  from dmesg | grep brcmfmac :

```
[    4.337738] usbcore: registered new interface driver brcmfmac
[    4.510026] brcmfmac: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[    8.004846] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   10.564854] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   10.564868] brcmfmac: brcmf_dongle_scantime: Scan assoc time error (-110)
[   13.124833] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   13.124844] brcmfmac: brcmf_netdev_open: failed to bring up cfg80211
[   15.844838] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   18.404829] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   18.404843] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[   20.964855] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   20.964867] brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
[   23.524841] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   23.524851] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[   26.085007] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   26.085019] brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
[   28.644897] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   28.644909] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[   31.204844] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   31.204853] brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
[   33.764858] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   38.884900] brcmfmac: brcmf_dongle_scantime: Scan assoc time error (-110)
[   41.444838] brcmfmac: brcmf_netdev_open: failed to bring up cfg80211
[   44.004843] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   44.004854] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[   46.564871] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   46.564880] brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
[   49.124844] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   54.244841] brcmfmac: brcmf_dongle_scantime: Scan assoc time error (-110)
[   56.804858] brcmfmac: brcmf_netdev_open: failed to bring up cfg80211
[   59.444861] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   64.564839] brcmfmac: brcmf_dongle_scantime: Scan assoc time error (-110)
[   67.124846] brcmfmac: brcmf_netdev_open: failed to bring up cfg80211
```

I updgrade to the last Kernel which runs fine on an second RPI3. 

```
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Linux raspberrypi 4.9.31-v7+ #1005 SMP Thu Jun 8 13:02:15 BST 2017 armv7l GNU/Linux
Linux version 4.9.31-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1005 SMP Thu Jun 8 13:02:15 BST 2017

Jun  8 2017 13:29:59
Copyright (c) 2012 Broadcom
version 0784a9f7dfa6a45aa2524eeab1980da00cd93019 (tainted) (release)
```

Maybe its an faulty device? 
Or where can this come from `brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout` ?
Thanks in advance.

pelwell, 2017-06-08:
I have not seen errors like that before, especially so consistently. If the same image runs on a different Pi3 with the same power supply then I have to think it is a (very rare) hardware fault.

bdheeman, 2017-06-28:
> [root@raspberrypi:~]# uname -a
Linux raspberrypi 4.9.34-1-ARCH #1 SMP Mon Jun 26 01:32:24 UTC 2017 armv7l GNU/Linux
[root@raspberrypi:~]# lsmod | grep brcm      
brcmfmac              203321  0
brcmutil                7361  1 brcmfmac
cfg80211              541781  1 brcmfmac
[root@raspberrypi:~]# dmesg |grep brcm
[   25.295147] brcmfmac: brcmf_sdiod_regrw_helper: failed to read data F1@0x08000, err: -110
[   25.295154] brcmfmac: brcmf_chip_recognition: chip backplane type 15 is not supported
[   25.295158] brcmfmac: brcmf_sdio_probe_attach: brcmf_chip_attach failed!
[   25.295162] brcmfmac: brcmf_sdio_probe: brcmf_sdio_probe_attach failed
[   25.295231] brcmfmac: brcmf_ops_sdio_probe: F2 error, probe failed -19...
[   25.295370] usbcore: registered new interface driver brcmfmac
[root@raspberrypi:~]# 

That's what I see and WiFi stopped working here too after firmware upgrade on RPi3; the OS is ArchLinuxARM here.

Josar, 2017-07-17:
my PI had a capacitor soldered instead of an resistor. see the pictures.

Faulty (sorry for the bad quality) soldered capacitor is white instead the usually black resistors.
[image]

Picture of an working Pi.
[image]

Repaired device with much bigger 0402 resistor, 22 Ohm instead the 22.5 Ohm (0102). 
Worked instantly after reboot.
[image]

Josar, 2017-07-17:
Had a capacitor soldered instead of an resistor. See pictures.

Bad device with a white capacitor instead the usually black resistors. (Sorry for the bad quality.)
[image]
Working device.
[image]

Repaired device with 22 Ohm [0402] resistor instead the 22.5 Ohm [0201].
Worked instantly after boot. 
[image]

closed 2017-07-17 by Josar

pelwell, 2017-07-17:
Wow - that's amazing detective work. If you send the serial number of the faulty board and a shipping address to phil@raspberrypi.org I'll make sure you get a replacement.

shaf, 2017-07-21:
@Josar That is really amazing !

As a new rpi user the pi zero w works perfectly but the rpi3 exhibited the same errors so I've been racking my head for a couple of days, trying different kernels and firmware I've given up....

However on inspection it seems that my wifi is chipped :( Hugely disappointed as this is going to hold up dev

[image]

I'm sure this is the cause of my woes

floydianslips, 2018-03-25:
I seem to be having this same issue. I have no wifi interface on one Pi 3 but I can switch the SD card to another Pi 3 and it works perfectly. When hovering over networking in the top bar it says: wlan0: Not associated.

carylogan, 2021-01-05:
Thanks so much.  I've tried to track this down a few times in the past, but just had to use hardwired network.  I did not find a capacitor, but i did find that that resistor was broken and mesured 3 MOhms.   I replaced with a 20 Ohm and at least wpa_supplicant will connect.  I still need to check my configuration.   Bluetooth also started working and connected to a keyboard.  At least I know that it was a hardware problem.