raspberrypi/firmware #825: Pi3 retry network (PXE) booting

FezzFest, 2017-06-10:
Can the Raspberry Pi 3 be programmed to retry network booting if it fails to boot the first time? We have some Raspberry Pi 3's at remote sites and if a power outage occurs, the Pi requests an IP-address through DHCP before networking equipment is fully up and running. This means the DHCP-request goes unanswered and the raspberry pi sets itself to USB device mode, then waits. 

Is it possible to reprogram the Pi to retry network booting after a failed attempt and not switch to device mode? Or does this require reprogramming the boot ROM (which, as I understand, is not possible as it is readonly and part of the SoC)?

mangodan2003, 2017-06-23:
We are booting a bunch of slave rpis from a master rpi and have a similar problem described above.
We currently have to boot the master first and then the slaves once the Master has booted. A way to allow the slaves to wait a time before booting - or to retry would be nice. We are using the bootloader.bin on the SD card to avoid setting the OTP flag at present so in theory this could be updated to accommodate our needs, only we cannot do this due to the closed nature of some parts of the RPi and thus need to ask someone who reads these forums and can do it to look at this for us.

timmmmmey, 2017-07-17:
This would be really useful. Especially in enviroments with a STP-switch.

popcornmix, 2017-07-17:
Without an sdcard present this is impossible as the bootrom can't be changed.

schoerg, 2017-07-17:
@timmmmmey 
Usually there is something like "portfast", which could be used as a workaround.

mikemccabe, 2017-11-03:
This is also biting us.  We're shipping integrated units with pis behind a switch that takes some time to come up - currently when the unit is power cycled the pis attempt to boot before the switch (which provices access to PXE resources) is ready, and we have to wait for a hardware watchdog to reset the pi.@popcornmix we use SD cards with an updated bootcode.bin, as the bootrom bootcode.bin doesn't function reliably.

ghollingworth, 2017-11-07:
bootcode.bin option for sticking permanently with ethernet sounds like an easy thing to add...  What release of bootcode.bin are you using at the moment?

mikemccabe, 2017-11-07:
Thanks for your reply!

We're using the bootcode.bin from the 2017-09-07-raspbian-stretch-lite release.

$ shasum bootcode.bin
d3852e9a9bcf9b3fbacfb84725fcad2d138baa12  bootcode.bin

ghollingworth, 2017-11-13:
Can you try following the instructions and using the bootcode.bin attached to the Wiki to get UART debug out?

https://github.com/raspberrypi/firmware/wiki/USB-MSD-and-network-boot-debugging

mikemccabe, 2017-11-15:
Hi, I've set this up.  Did you also change the logic in that bootcode.bin?  Things seem to work now!  Booting fails in my test setup with the older bootcode.bin.(My test setup: Pi is connected to PXE server via an intermediate switch powered on after boot starts, thus DHCP may fail for a time even though the link is up.

mikemccabe, 2017-11-15:
I'm also encountering a related boot problem; it may belong a separate issue - let me know.  (But now I have a tool to diagnose it, which is fantastic!)

The issue is that with full PXE boot (bootcode.bin only on SD card) if there's a USB storage device of any kind, the boot process halts.  (Green light blinks 3? times, repeatedly.)  We want to keep a homogenous PXE environment on all our devices, but some have USB keys for storage.

It doesn't matter if the USB key is FAT, blank (`dd if=/dev/zero`-ed out), ext4, NTFS, etc.  ('FAT32 partition decoded successfully' is always there.)

It looks from the PXE trace above that full back-out retries are happening, could it make sense to re-try from a different device (ethernet) on the USB hub? 

Here's the trace:

> Raspberry Pi Bootcode
> 
> Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
> Trying USB
> Hub device found at addr 4, enumerating HUB
> Initialise hub
> Found 5 ports, multi_tt = 1
> Setting interface 0
> Enabling PORT POWER on port 1
> Enabling PORT POWER on port 2
> Enabling PORT POWER on port 3
> Enabling PORT POWER on port 4
> Enabling PORT POWER on port 5
> Waiting for devices to respond to reset
> Found device on port 1
> Found highspeed device
> Device found: type = Ethernet adapter, addr = 5
> Found device on port 4
> Found highspeed device
> Device found: type = Mass storage, addr = 6
> Trying booting from MSD device addr 6
> Retrieving the config descriptor
> Found ep_in = 00000002, ep_out = 00000001
> GetMaxLun returned 0, max_lun = 0
> Sending and retrying TEST_UNIT_READY
> Sending SCSI_COMMAND_INQUIRY
> Sending READ_CAPACITY
> init_msd returned 0
> USB device configuration set
> Initisalised mass storage, block size = 512, block count = 62668800
> MBR read and decoded successfully
> FAT32 partition decoded successfully
> 
> 
> 
> 
> Raspberry Pi Bootcode
> 

(hang)

mikemccabe, 2017-11-15:
An option like the 'timeout' file (`touch pxe`) could also work for the above issue.  Obviously not appropriate to a bootrom approach.

mikemccabe, 2017-11-29:
I've moved comments describing a related issue to #907 .

The debug bootcode.bin you supplied seems to have resolved this issue, can we expect those changes to land in the mainline code?

ghollingworth, 2017-12-15:
They will end up there, it's taking some time putting together an update for various reasons...

JamesH65, 2019-01-15:
@ghollingworth Missed this one when cataloging PXE/netboot issues.

closed 2019-01-17 by ghollingworth