raspberrypi/firmware #826: Original Raspberry pi setting wrong screen dimentions for DVI monitor

thedjnK, 2017-06-13:
I've had an arch linux arm raspberry pi for a number of years without issue, the config file was empty and the monitor configuration was always correctly determined at boot up. I upgraded everything earlier (it hasn't been updated in a long time so I'm not sure where the commit that causes this issue is) and now when it boots I see only a small part of the rainbow image that's zoomed in, and all the text output from the system booting is magnified by possibly 500 times (the text is so large and blurry it is impossible to read). I later tried using the newest raspbian image from the raspberry pi site and that has the same problem. I have to correct it by putting this in the config.txt file:

    hdmi_drive=1
    hdmi_group=2
    hdmi_mode=16

I'm using a normal HDMI cable to connect between the raspberry pi with a HDMI-to-HDMI adapter on the end connecting to the monitor which is a Digimate L1916 19" TFT LCD Monitor, specs available on https://www.amazon.co.uk/d/Monitors/Digimate-L1916-19-TFT-LCD-Black-Silver/B0007LYGDY

I do not see this issue with a raspberry pi 3 (although the firmware hasn't been updated in a few months so if that was to be upgraded and if the issue arises from that I do not know).

popcornmix, 2017-06-13:
Can you report output of:
```
vcgencmd version
tvservice -s
tvservice -m CEA
tvservcie -m DMT
```
on a pi that behaves as desired and a pi that doesn't (connected to same TV).

thedjnK, 2017-06-13:
Non-working original raspberry pi with latest firmware:

    ./vcgencmd version
    Jun  8 2017 13:31:01                                                                                                                                                                                           
    Copyright (c) 2012 Broadcom                                                                                                                                                                                    
    version 0784a9f7dfa6a45aa2524eeab1980da00cd93019 (tainted) (release)

    ./tvservice -s
    state 0x12000a [HDMI DMT (4) RGB full 4:3], 640x480 @ 60.00Hz, progressive

    ./tvservice -m CEA
    Group CEA has 2 modes:
               mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive 
               mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive 

    ./tvservice -m DMT
    Group DMT has 1 modes:
      (prefer) mode 4: 640x480 @ 60Hz 4:3, clock:25MHz progressive

Working raspberry pi 3 with old firmware:

    vcgencmd version
    Dec  9 2016 15:11:26 
    Copyright (c) 2012 Broadcom
    version 2e557d8dac70add28597c3b449cb52c34588d818 (clean) (release)

    tvservice -s
    state 0x120006 [DVI DMT (16) RGB full 4:3], 1024x768 @ 60.00Hz, progressive

    tvservice -m CEA
    Group CEA has 0 modes:

    tvservice -m DMT
    Group DMT has 12 modes:
               mode 4: 640x480 @ 60Hz 4:3, clock:25MHz progressive 
               mode 5: 640x480 @ 72Hz 4:3, clock:31MHz progressive 
               mode 6: 640x480 @ 75Hz 4:3, clock:31MHz progressive 
               mode 8: 800x600 @ 56Hz 4:3, clock:36MHz progressive 
               mode 9: 800x600 @ 60Hz 4:3, clock:40MHz progressive 
               mode 10: 800x600 @ 72Hz 4:3, clock:50MHz progressive 
               mode 11: 800x600 @ 75Hz 4:3, clock:49MHz progressive 
      (prefer) mode 16: 1024x768 @ 60Hz 4:3, clock:65MHz progressive 
               mode 17: 1024x768 @ 70Hz 4:3, clock:75MHz progressive 
               mode 18: 1024x768 @ 75Hz 4:3, clock:78MHz progressive 
               mode 35: 1280x1024 @ 60Hz 5:4, clock:108MHz progressive 
               mode 36: 1280x1024 @ 75Hz 5:4, clock:135MHz progressive

popcornmix, 2017-06-13:
Can you also report output of
```
sudo vcdbg log msg
```
on both.

thedjnK, 2017-06-13:
Non-working:

    Unable to determine the value of __LOG_START
    Unable to read logging_header from 0x00000000

Switched from arch linux arm to raspbian and re-ran it non-working:

    000893.194: HDMI:EDID error reading EDID block 0 attempt 0
    000894.458: HDMI:EDID error reading EDID block 0 attempt 1
    000895.717: HDMI:EDID error reading EDID block 0 attempt 2
    000896.975: HDMI:EDID error reading EDID block 0 attempt 3
    000898.234: HDMI:EDID error reading EDID block 0 attempt 4
    000899.492: HDMI:EDID error reading EDID block 0 attempt 5
    000900.751: HDMI:EDID error reading EDID block 0 attempt 6
    000902.009: HDMI:EDID error reading EDID block 0 attempt 7
    000903.268: HDMI:EDID error reading EDID block 0 attempt 8
    000904.526: HDMI:EDID error reading EDID block 0 attempt 9
    000905.548: HDMI:EDID giving up on reading EDID block 0
    000919.328: HDMI:Setting property pixel encoding to Default
    000919.348: HDMI:Setting property pixel clock type to PAL
    000919.367: HDMI:Setting property content type flag to No data
    000919.387: HDMI:Setting property fuzzy format match to enabled
    000927.576: gpioman: gpioman_get_pin_num: pin WL_LPO_CLK not defined
    000927.600: gpioman: gpioman_get_pin_num: pin BT_ON not defined
    000927.620: gpioman: gpioman_get_pin_num: pin WL_ON not defined
    001099.729: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
    001100.058: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
    001101.416: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
    001102.770: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
    001104.123: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
    001105.477: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
    001106.830: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
    001108.183: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
    001109.536: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
    001110.889: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
    001112.243: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
    001113.269: hdmi: HDMI:EDID giving up on reading EDID block 0
    001113.302: hdmi: HDMI: No lookup table for resolution group 0
    001113.320: hdmi: HDMI: hotplug attached with DVI support
    001113.360: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
    001113.763: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
    001115.121: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
    001116.476: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
    001117.831: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
    001119.186: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
    001120.541: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
    001121.896: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
    001123.251: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
    001124.607: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
    001125.961: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
    001126.988: hdmi: HDMI:EDID giving up on reading EDID block 0
    001129.429: hdmi: HDMI: power_on to DMT mode VGA@60
    001132.564: hdmi: HDMI: Action callback added to queue to happen at frame 2
    001132.588: hdmi: HDMI: Action stop_3d_mode added to queue to happen at frame 2
    001132.608: hdmi: HDMI: Action unmute added to queue to happen at frame 3
    001132.636: hdmi: HDMI: Action cec_init added to queue to happen at frame 3
    001198.930: *** Restart logging
    001201.231: Read command line from file 'cmdline.txt'
    dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p6 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
    001446.315: Loading 'kernel.img' to 0x8000 size 0x3f1128
    001446.363: No kernel trailer (run mkknlimg to fix) - assuming DT-capable
    001448.401: Loading 'bcm2708-rpi-b.dtb' to 0x3f9128 size 0x36ba
    002580.288: Device tree loaded to 0x1bfec400 (size 0x3b2f)
    002581.571: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
    003622.030: vchiq_core: vchiq_init_state: slot_zero = 0x5b880000, is_master = 1
    003626.421: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
    003631.743: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000004

Working:

[... 30 lines cut ...]
    001206.599: hdmi: HDMI:EDID found DMT format: code 10, 800x600p @ 72 Hz in established timing I/II
    001206.641: hdmi: HDMI:EDID found DMT format: code 11, 800x600p @ 75 Hz in established timing I/II
    001206.684: hdmi: HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
    001206.727: hdmi: HDMI:EDID found DMT format: code 17, 1024x768p @ 70 Hz in established timing I/II
    001206.767: hdmi: HDMI:EDID found DMT format: code 18, 1024x768p @ 75 Hz in established timing I/II
    001206.813: hdmi: HDMI:EDID found DMT format: code 36, 1280x1024p @ 75 Hz in established timing I/II
    001206.906: hdmi: HDMI:EDID standard timings block x 8: 0x0101 0101 0101 0101 0101 0101 0101 0101 
    001206.946: hdmi: HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
    001207.085: hdmi: HDMI:EDID preferred mode remained as DMT (35) 1280x1024p @ 60 Hz with pixel clock 108 MHz
    001207.103: hdmi: HDMI: hotplug attached with DVI support
[... cut ...]
    001492.438: Loading 'bcm2710-rpi-3-b.dtb' to 0x411058 size 0x3e78
    001589.143: dtparam: audio=off
    001742.442: Device tree loaded to 0x2efebc00 (size 0x4388)
    001744.759: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
    002499.474: vchiq_core: vchiq_init_state: slot_zero = 0xee880000, is_master = 1
    002503.912: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
    002509.096: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000023

popcornmix, 2017-06-13:
So the non-working Pi cannot read the EDID of the display which describes the supported resolutions.
Normally that is a hardware issue (e.g. a faulty TV, or cable that's not fully wired).

Is everything the same in the two cases you are testing apart from the Pi and the sdcard?
e.g. same HDMI cable, same socket on monitor?
Do you have a spare sdcard you can try a latest raspbian on the Pi3?

thedjnK, 2017-06-13:
I've got an original 256MB RAM raspberry pi model B which I'm currently busy compiling something on but I will test with when it's finished, the non-working test is with a 512MB RAM raspberry pi model B, the working test is with a raspberry pi 3. Everything is the same in the tests, generic micro-USB cable for power, the same HDMI cable and HDMI > DVI connector and monitor are used, same USB mouse and keyboard, the non-working tests is with a 16GB PNY 'elite performance' SD card (which was working fine without a config.txt on this setup before I upgraded all the packages whereby it stopped working) and the raspberry pi 3 is using a 32GB sandisk microUSB.

I've booted the working raspberry pi 3 board with the SD card from the non-working original raspberry pi using a transflash-to-SD adapter and it's working fine with the display. The output of the `sudo vcdbg log msg` command is the same as with the older firmware when booted on the raspberry pi 3 so it looks to be an issue only with the original raspberry pi.

thedjnK, 2017-06-14:
I've just tested the SD card on the 256MB raspberry pi and it works fine... What's even stranger is I've tried it again on the 512MB rasberry pi where it wasn't working earlier and it's working fine on that now too. Restarted and cold booted multiple times on both boards on both arch linux arm and raspbian and I'm not able to get it to fail, it's detecting the monitor fine and all modes and selecting 1280x1024 which is displaying fine. So with that I'll close this and put it down to something strange that can no longer be reproduced.

closed 2017-06-14 by thedjnK