raspberrypi/firmware #827: tvservice gives wrong answer for RPF Official 7" Display

DougieLawson, 2017-06-15:
Running ```tvservice -s``` gets

```
pi@challenger:~$ tvservice -s
state 0x40001 [NTSC 4:3], 720x480 @ 60.00Hz, interlaced
pi@challenger:~$
```
Which doesn't seem to match the advertised resolution for the RPF Official Display.

popcornmix, 2017-06-15:
That is describing the resolution of composite connection.
Is LCD connected and working? Is it the primary or secondary display?

With default config.txt settings and display connected (so primary) I get:
```
pi@domnfs:~ $ tvservice  -s
state 0x400000 [LCD], 800x480 @ 60.00Hz, progressive
```

DougieLawson, 2017-06-15:
The display works perfectly. It is the only display device connected to that RPi3B.

Plus or minus https://github.com/raspberrypi/linux/issues/2071 which means I can't control the backlight.

DougieLawson, 2017-06-15:
Sorry forgot this essential piece
```
pi@challenger:~$ vcgencmd version
Jun 13 2017 14:49:57
Copyright (c) 2012 Broadcom
version 22340f772ce101854acbc46045358dae832dd513 (tainted) (release)
pi@challenger:~$ uname -a
Linux challenger 4.9.31-v7+ #1007 SMP Tue Jun 13 15:14:24 BST 2017 armv7l GNU/Linux
pi@challenger:~$
```

DougieLawson, 2017-06-16:
Looks like https://github.com/Hexxeh/rpi-firmware/commit/9def95edfc121e3762a9ff520befbee9557436e1 has fixed this.

popcornmix, 2017-06-16:
That latest update was kernel only, so I don't see it fixing the issue.
If you see the issue again, output of:
```
sudo vcdbg log msg
vcgencmd dispmanx_list
```
may be interesting.

JamesH65, 2017-12-22:
@DougieLawson Hi Dougie, is this fixed? Please close if so.

JamesH65, 2019-01-11:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

DougieLawson, 2019-01-11:
At the moment I don't know 100%. The last experiments before Xmas were with that display and @JamesH65's dual screen stuff (which I never got working). It looks like I've got 
```hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=16
``` 
in /boot/config.txt for that dual screen stuff right now. I'll have to sit down and play with that Raspberry with the screen tomorrow and update or close this issue.

DougieLawson, 2019-01-21:
Finally got round to checking this.
```state 0x400000 [LCD], 800x480 @ 60.00Hz, progressive```
is what I'm getting today. That's running with BRANCH=next ```4.14.93-v7+ #1191```.

closed 2019-01-21 by DougieLawson