raspberrypi/firmware #829: hdmi_force_hotplug=1 doesn't work if RPi can't read EDID

danijelt, 2017-06-20:
In our project, we use various TV brands and models.  
On some of them, under specific circumstances (which differ between brands), RPi can't read EDID from TV, and if it can't read EDID file, HDMI hotplug doesn't work and there is no picture on HDMI, even with `hdmi_force_hotplug=1`.

If I fake the EDID with `hdmi_ignore_edid=0xa5000080`, put `edid.dat` on the SD card, or reboot RPi once the TV is in condition where EDID is readable, it will work even if HDMI is plugged in after booting kernel.

Is this expected?

popcornmix, 2017-06-20:
No. Please report output of:
```
vcgencmd version
tvservice -s
vcgencmd get_config int
vcdbg log msg
```
when the tv is working (e.g. with edid from file) and not working.
(In both cases with edid unreadable).

JamesH65, 2017-12-22:
@danijelt Did you have a chance to get the debug out needed to diagnose this? Or have you worked around it somehow?

danijelt, 2017-12-22:
@JamesH65 Sorry for the wait. It appears that the problem is gone and I'm not able to reproduce it anymore. Since we had various firmware versions for TV sets while we tested, it's possible that I just came across one that was faulty, and/or missed something in config.txt during all that mess. If I ever see the problem again and reproduce it, I'll reopen this issue with required information.

closed 2017-12-22 by danijelt