raspberrypi/firmware #831: Wrong sleep frequency for BCM43340 on GPCLK2?

lategoodbye, 2017-06-25:
According to the dt-blob.dts the BCM43340 of the Raspberry Pi Zero W seems to uses GPCLK2 as the external Low-Power Oscillator. Looking at the datasheet the Low-Power Oscillator for the BCM43340 chip should have the frequency 32768 Hz. But if i dump the clock settings the frequency is set to 32000 Hz:

```
root@raspberrypi:/sys/kernel/debug# cat clk/clk_summary
   clock                         enable_cnt  prepare_cnt        rate   accuracy   phase
----------------------------------------------------------------------------------------
 uart1_pclk                               0            0   125000000          0 0
 uart0_pclk                               0            0     3000000          0 0
 apb_pclk                                 0            0   126000000          0 0
 otg                                      0            0   480000000          0 0
 osc                                      6            6    19200000          0 0
    gp2                                   1            1       32000          0 0
```

Is this intended or a bug?
May be this is related to the wifi power managment issues.

pelwell, 2017-06-26:
That looks like a bug, probably due to a misinterpretation of 32KHz somewhere along the line.

I'll be surprised if it changes the behaviour in any significant way, but we'll get it fixed.

pelwell, 2017-06-27:
A firmware with the correct clock rate is now available via rpi-update. Please test and close if appropriate.

lategoodbye, 2017-06-28:
Since the power management of the brcmfmac is completly disabled i don't expect any changes regarding to Wifi. But i can confirm the LPO frequency is correct now:

 ```
Linux raspberrypi 4.9.34+ #1013 Sun Jun 25 16:41:15 BST 2017 armv6l GNU/Linux

  clock                         enable_cnt  prepare_cnt        rate   accuracy   phase
----------------------------------------------------------------------------------------
 uart1_pclk                               0            0   125000000          0 0
 uart0_pclk                               0            0     3000000          0 0
 apb_pclk                                 0            0   126000000          0 0
 otg                                      0            0   480000000          0 0
 osc                                      6            6    19200000          0 0
    gp2                                   1            1       32768          0 0
```

closed 2017-06-28 by lategoodbye