raspberrypi/firmware #832: MMAL camera image overlay Example?

AlexanderShniperson, 2017-06-29:
Hi,
can some one make an example of creation overlay image over camera preview, like it does in [Python Picamera](https://github.com/waveform80/picamera)?
p.s. it is hard to understand Python source code

6by9, 2017-06-29:
userland issue, not firmware.

Overlay over preview (not recording it) is just creating a MMAL video_render component, setting the display rectangle and layer (```MMAL_PARAMETER_DISPLAYREGION```) appropriately, and sending it a buffer with your image in it.

AlexanderShniperson, 2017-06-29:
thanks for reply, will open it at userland
the general problem is: i am from Java world, so i don't exactly understand what you mean, i need simple solution

6by9, 2017-06-29:
https://github.com/raspberrypi/userland/issues/405
Please close this issue.

closed 2017-06-29 by AlexanderShniperson