raspberrypi/firmware #834: Question regarding vcos_abort and recoverabilty after

anfractuosity, 2017-07-01:
Hi,

I'm just wondering, I've just taken a skim of the code surrounding the 'vcos_abort: Halting' 
message.

Would I be right in thinking the vcos_abort is fatal, with no way of recovering from at all? Without only doing a system reboot?

Or is it possible to restart the videocore without re-starting the whole system?

Please forgive my naive questions. 

Thanks in advance!

Chris

popcornmix, 2017-07-02:
vcos_abort is fatal - a reboot will be required.
It should never occur - if you find a case where it does occur then the bug must be fixed so vcos_abort no longer occurs.

anfractuosity, 2017-07-02:
Cheers @popcornmix thanks for your reply, that's very helpful!

I just got it recently while streaming with a 1m camera cable, which I realise is pushing the hardware to it's limits, so is likely just due to the length of cable I'm assuming.  Unfortunately the issue is sporadic, like it may occur hours into streaming.  I'll see if I can possibly add some debugging information into raspivid, which might let me see when it occurs.

popcornmix, 2017-07-02:
Is this triggering on arm or firmware side? (vcos is used on both sides).
What message do you see?

anfractuosity, 2017-07-02:
So I notice the stream stops. But raspivid and ffmpeg may remain still open.

Sorry I'm not sure about if it's triggering on arm or firmware side, is there a simple way I can tell?

I then did:

sudo vcdbg log msg

And got messages like the below:
```
3480690.655: vcos_abort: Halting
3482637.188: mmal: fail_destroy: Timed out waiting to destroy ril.camera
```

popcornmix, 2017-07-02:
If error comes from vcdbg then it is firmware side.
Can you add ``start_debug=1`` to config.txt and reboot.
Then report complete output of:
```
vcgencmd version
sudo vcdbg log msg
sudo vcdbg log assert
sudo vcdbg log ex
```

anfractuosity, 2017-07-02:
Sure, I'll definitely do that. It might be a while before I get the error again, but I'll post when I do. And log that information.

6by9, 2017-07-03:
vcos_abort will abort the thread that found the situation it judged fatal, although others should continue running. If the thread in question is one of the main IPC threads, then that GPU service will be blocked until reboot.

There are a couple that I know of, but only get tripped in very specific situations:

1. In the MMAL core is when the buffer management gets screwed up and a component returns a buffer that the core didn't think it had. I've seen that with the rawcam component and know I need to look at it at some point.
2. When running video_encode with a very low bitrate. If rate control decides that the last frame has to be skipped because it is out of bits, then on close the codec complains bitterly due to in/out counters not matching, fails to close, and I think calls vcos_abort.
3. General stack/heap trashing, but that should never happen.

I've checked, and there are no vcos_demands in the main camera component or imaging subsystem. It's possible that one of the other libraries abort, but that would be unusual.
That makes the most likely cause a buffer overflow. A long camera cable shouldn't be able to induce that under any circumstances.
Are you using any particular raspivid options in your captures?

anfractuosity, 2017-07-03:
@6by9 Thanks a lot for your pointers.  I _think_ I did recall seeing a message regarding MMAL in the bash console, when the crash had occurred - annoyingly I don't think I logged that message.   I've been streaming for 16 hours now and it hasn't crashed at the moment.  But it did definitely crash the other day using the 1m camera cable.  As soon as it crashes again, I'll post here.  The bitrate I'm using is 1Mbps:

```
raspivid --ISO 800 -v -o - -t 0 -fps 30 -b 1000000 | ffmpeg -re -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -f h264 -r 30 -i - -vcodec copy -acodec aac -ab 128k -g 50 -strict experimental -f flv rtmp://a.rtmp.youtube.com/live2/[code]
```

6by9, 2017-07-04:
The default resolution for raspivid is 1080P. Trying to encode that at 1Mbit/s is what I'd call optimistic! Rate control is very likely skipping quite a few frames although it will depend on the level of movement in the scene.
My memory says the problem in the codec only occurs on shutdown, so it shouldn't abort mid-stream due to this, but it is one possible cause.

Can you either increase the bitrate or drop the resolution?

anfractuosity, 2017-07-04:
Oh very interesting, sure I'll do that!  I was using 1Mbps as our total upstream was only 2Mbps, but it just got upgraded today to around 11Mbps :), if I used 5Mbps for the stream ,would that be more reasonable?

6by9, 2017-07-04:
Doing the basic maths.
1Mbit/s = 125kbytes/s.
30 frames/s, so if equally split, 125/30 = 4.07kB per 1920x1080 frame. That's not much.

Now admittedly the H264 codec will be sending one I-frame per 60 frames by default, and P-frames should be much much smaller as they are only deltas, but even if you allocated all of the bandwidth for those 2 seconds to the I-frame, 250kB for a 2MPix (effective) JPEG isn't going to be brilliant quality. If there is any significant motion then it will very easily exceed the target bitrate and is likely to drop frames.

For comparison, the BBC HD channels on satellite are apparently running 1080i at an average bitrate of 4.5-5Mbit/s, but peaking up to 12Mbit/s. They're also using significantly more advanced encoders which support things such as B-frames, so can get better image quality for the same bandwidth. Your average camera phone will be set to 12-18Mbit/s for 1080P.
(Data from http://en.digitalbitrate.com/dtv.php?mux=10847&pid=6941&live=69&lang=en - quote an interesting site in a geeky way!)

anfractuosity, 2017-07-04:
That's very interesting thanks!  That site is very cool too.  

Tomorrow one thing I'm going to try, is see if I can get it to crash in the same fashion, with say 1Mbps or maybe even lower, by dancing around a lot in front of the camera, most of the time the camera isn't capturing anything normally, as it's just to catch wildlife, which haven't really been appearing recently.

6by9, 2017-07-04:
I had a quick look before I left the office. I'm not convinced that the codec will misbehave badly enough to trigger an abort based on this, and certainly not mid-encode. I'll try to reproduce tomorrow if I can find a few spare minutes. 

If you're really not getting that much motion I'd suggest you add something like -g 300 to insert one I-frame every 10 seconds and thereby reduce the disparity between I and P frames. It does make the stream less robust to corruption, but there's always a compromise.
Youtube do make some recommendations, but I can't see them at the moment. However I did trip over https://support.google.com/youtube/answer/1722171?hl=en-GB in my search - recommended bitrate for 1080P is 8Mbit/s.
..
It is there: "Closed GOP. GOP of half the frame rate.", so they are recommending 60, although that doesn't seem to preclude a longer value (GOP = Group Of Pictures, or the repeating pattern of I, P, and B frames).

anfractuosity, 2017-07-04:
I just tried setting the bitrate to '100000', and waving my hand in front of the camera.

Although it didn't crash, when I killed the stream.

I noticed : 
4092921.298: mmalsrv: send_buffer_to_host: tx failed:size 292 st -1

In:
sudo vcdbg log msg

JamesH65, 2019-01-11:
Any further comments from any participents?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2019-11-05 by JamesH65