raspberrypi/firmware #835: Kernel 4.9.28-v7+ no longer has twofish modules

mohakshah, 2017-07-03:
After I updated to debian stretch, I can no longer access my HDD encrypted with twofish using LUKS. Is there anyway I can get the twofish modules?

pelwell, 2017-07-03:
I can't see any evidence of RPi kernel configurations ever enabling the CONFIG_CRYPTO_TWOFISH option. What was the kernel version string (`uname -a`) for the last working system? Are you sure it was a Raspbian build, or had you perhaps installed a custom kernel from somewhere?

mohakshah, 2017-07-03:
That could be it! I installed Raspbian Jessie using [raspbian-ua-netinst](https://github.com/debian-pi/raspbian-ua-netinst) and then upgraded to Raspbian Stretch by updating my `/etc/apt/sources.list`

closed 2017-07-12 by mohakshah