raspberrypi/firmware #836: 240P over COMPOSITE - can't adjust front/back porch?

Dochartaigh, 2017-07-04:
I'm using a 240p setup over Composite RCA (on a CRT TV) as described here: https://github.com/raspberrypi/firmware/issues/683

I've always used hdmi_timings to adjust screen size, centering, and front and back porches. How is this accomplished when using "sdtv_mode=16" in /boot/config.txt?

My first problem is centering the image. After that I can't figure out how to get rid of the black bars on the top and bottom of my screen (tried 6x CRT's so I know it's not them - AND the multicolored block when you first boot-up is completely full-screen top-to-bottom so I know it's capable of filling the screen). Thank you.

popcornmix, 2017-07-04:
You can't adjust the composite timings. That is a hdmi only feature.

Have you tried disabling/adjusting overscan?
Try ``disable_overscan=1``. If that results in missing pixels then increase ``overscan_left``/``overscan_right``/``overscan_top``/``overscan_bottom``
https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

Dochartaigh, 2017-07-04:
When using the Pi2SCART for RGB output from the Raspberry Pi 3 (with hdmi_timings) we learned to stay away from touching overscan and any of those parameters as it increases (exponentially in my experience) artifacts and scaling issues on our CRT's when playing video games (and basically making the picture look absolutely horrible).

Are things different since this is a completely different video output method? I will try it right now but just had a bear of a time getting things setup correctly with the Pi2SCART and it seems like I need to re-learn everything once again to use this slightly different video mode. Thanks.

popcornmix, 2017-07-04:
Overscan is enabled by default, so it doesn't really make sense to say don't touch it because overscan settings cause scaling artefacts.
The overscan settings reduce the framebuffer dimensions by the number of pixels specified, so it is just replacing edge pixels with black pixels - it doesn't affect scaling.
If you saw an issue it may be due to the way the game/emulator was scaling, but standard framebuffer won't have this issue.

Dochartaigh, 2017-07-04:
It's been a while since I got mine all setup but I'm pretty positive it was on a per-emulator basis (and sometimes per-ROM if you're playing Arcade/MAME with a BUNCH of mixed resolutions).

For instance if the system is set to 320x240 (which I understand this 240p over composite tweak ISN'T) I was seeing scaling issues and other weird things if the viewport height was set to 242 instead of the correct 240 for example

ektgit, 2017-11-18:
You can try this shader:  https://github.com/ektgit/snap-shader-240p