raspberrypi/firmware #837: screen tearing when using display_rotate

christophevg, 2017-07-05:
My RPI 3 setup (fully updated and upgraded)  consists of an HDMI screen in portrait mode. So I used the `display_rotate` option in `boot/config.txt` to rotate the entire desktop.

This option/feature causes _screen tearing_ (https://en.wikipedia.org/wiki/Screen_tearing).  After some Googling, I found https://raspberrypi.stackexchange.com/questions/37280/distortion-while-moving-objects-in-portrait-mode, which describes the problem and proposes a solution to rotate the Xserver. This is indeed _a_ solution, but the result is suffering from similar artefacts when updating the screen's content.

> The screen tearing problem was most apparent when I wanted to display a 1080p movie using omxplayer. Only with the default display rotation, the video quality is good. As soon as I change the display orientation using the config.txt option, the video suffers terribly from the tearing.

Other sources address this problem to `VSYNC` (https://www.raspberrypi.org/forums/viewtopic.php?f=7&t=4451), but I don't seem to be able to find any way to influence this from my side.

My main question regarding this issue is: Is there a way to use the `display_rotate` option and avoid the screen tearing?

Thanks for your insights and feedback,
Christophe VG

popcornmix, 2017-07-06:
No. display_rotate is designed to copy/transpose a live buffer and so will suffer from tearing.

In the future we will be moving to an arm side kms driver and that should have the ability to rotate the display using standard X mechanisms which should allow rotation without tearing.

dividuum, 2017-07-10:
Just adding: You can avoid tearing this if you rotate the video renderer output only instead of rotating the complete output with the `display_rotate` config.txt parameter. I'm pretty sure omxplayer has a --rotate argument. This will rotate the video only and you'll get no tearing.

popcornmix, 2017-07-10:
Yes, if it only video you care about then see https://github.com/popcornmix/omxplayer/issues/553

JamesH65, 2017-12-22:
The latest firmware has an extra config.txt option which may help with tearing a little bit. Here's all the display change config.txt options I added.

display_lcd_rotate -specifies rotate of LCD only.
display_hdmi_rotate - rotate of HDMI/composite only.
extra_transpose_buffer - allocates an extra transpose buffer when rotation of a display is required- can help with tearing

There is probably still  a chance of tearing (I'm working on a fix for that) but it may be reduced.

riverscn, 2018-10-10:
extra_transpose_buffer=1 still have tearing from my experience.

JamesH65, 2018-10-10:
Is it reduced by using the extra_transpose_buffer config item?

riverscn, 2018-10-10:
I use a DPI connected HVGA(480x640).
display_lcd_rotate=1
extra_transpose_buffer=1
Simple test method: turn on "Black frame insertion" in RetroArch.
[image]
(extra_transpose_buffer=0)
[image]
(extra_transpose_buffer=1)
Tearing is still there, but different (smaller?).

JamesH65, 2018-10-10:
Hmm, I don't have retroarch. Or that screen, so not easy for me to look at.

riverscn, 2018-10-10:
I hope kms driver come out soon 👍

JamesH65, 2018-10-10:
You cam use the beta version now. It's been out for ages.

lwpz, 2018-12-29:
I'm currently using kernel 4.4.50, are the new config.txt options working in this kernel version?? I'm facing similar issues.
Is it possible to advise where to look for the KMS drivers?
Thanks.

JamesH65, 2018-12-29:
You enable the KMS driver using raspi-config, but I suspect 4.4.50 might be too old. We are currently in testing of 4.19, and current release is 4.14, so you are a long way behind.

lwpz, 2018-12-29:
Ok. Thanks. Does it mean that the new config.txt settings are also only available in 4.14+ kernel?

JamesH65, 2018-12-29:
It's actually the firmware on the VC4 that parses config.txt for the transpose buffer, so if that is up to date, you should have that option. The latest KMS (name?) code needs a fairly recent kernel I believe.

lwpz, 2018-12-29:
I see. I'm using a CM3 compute module purchased in Jan 2018. Is there anyway to check the firmware of the VC4? 

Else, I guess I could just try the commands to see if it does anything.

Thanks.

popcornmix, 2019-01-02:
``vcgencmd version`` should report the date of the firmware.

lwpz, 2019-01-03:
> `vcgencmd version` should report the date of the firmware.

Ok. I'll check this.

lwpz, 2019-01-03:
> I see. I'm using a CM3 compute module purchased in Jan 2018. Is there anyway to check the firmware of the VC4? Else, I guess I could just try the commands to see if it does anything. Thanks.
> […](#)

We had checked and found that kernel 4.4.50 does not support the above extra options. Kernel 4.14 will support it. For the KMS drivers, it's not working in 4.4.50, but working in 4.14. However, tearing issue is still there. 

Is it possible to modify the HDMI driver to solve this issue? Just wondering....

JamesH65, 2019-01-03:
Can you copy&paste the results of the following to here so I can confirm the transpose buffers are working.

`sudo vcdbg reloc `

You should get entries for "transpose buffer0" and "transpose buffer1"

lwpz, 2019-01-04:
> should have mentioned, this is for the standard display drivers, not KMS, so using display_lcd_rotate and extra_transpose_buffer in config.txt

Yes. I understood, but enabling the KMS drivers didn't show any (improved) effect to tearing, even with or without the additional commands. The one that made a difference in the GUI desktop was basically kernel 4.14 with extra_transpose_buffer =1.

lwpz, 2019-01-04:
> Can you copy&paste the results of the following to here so I can confirm the transpose buffers are working.
> 
> `sudo vcdbg reloc `
> 
> You should get entries for "transpose buffer0" and "transpose buffer1"

Ok. I'll check for both 4.4.50 and 4.14.

lwpz, 2019-01-04:
> > Can you copy&paste the results of the following to here so I can confirm the transpose buffers are working.
> > `sudo vcdbg reloc `
> > You should get entries for "transpose buffer0" and "transpose buffer1"
> 
> Ok. I'll check for both 4.4.50 and 4.14.

Please see the screenshots of both `vcgencmd version` and `sudo vcdbg reloc `

[image]

[image]

Looks like 4.14 has a transpose buffer0, even without the extra_transponse_buffer=1, but 4.4.50 has neither.

Comments welcome.

JamesH65, 2019-01-04:
OK, so 4.14 (actually, its the version of firmware that is important, 4.14 has a much more recent firmware. You could use the 4.14 firmware on 4.4 I think) is allocating a transpose buffer, you would get two transpose buffers if you have extra_transpose_buffer set.  I would expect two transpose buffers to improve the observed tearing effect, although in my tests here, when running omxplayer over the framebuffer on rotated HDMI screen, I still get some tearing. I do not get the dark band, its only vaguely noticeable in the corner of the played back video.

lwpz, 2019-01-04:
Do you mean the firmware inside the VC4 chip? Couldn’t really make out the firmware version, as the numbers (hex) were pretty long. It’s different tho. Did you want me to swap the kernel version and have a look?

If the firmware is on the chip, is there anyway to program the VC4?

Resolution wise we’re not even close to 1920x1080. I’d guess that’d explain why our moving images are more responsive to the extra buffer?

JamesH65, 2019-01-04:
The firmware is the start*.elf files in the boot folder. They are loaded on to the VC4 on startup. They should be backwards compatible, so you could take the start*.elf files from a 4.14 and put on a 4.4 and its should still work. However, I would backup the old files just in case as its not something we generally check. vcgencmd version will display the version number of the firmware (and date etc).

lwpz, 2019-01-04:
Ok. I’ll give it a try and see. Thank you for your help. Let you know soon.

popcornmix, 2019-01-04:
Note: copy start*.elf, fixup*.elf and bootcode.bin.

lwpz, 2019-01-07:
> Note: copy start*.elf, fixup*.elf and bootcode.bin.

I checked a sample directory and it does not seem to have fixup*.elf files, but only fixup*.dat. Do I need to copy that?

JamesH65, 2019-01-07:
Sorry, they are fixup*.dat, not elf.

lwpz, 2019-01-07:
We had tested what you mentioned and replaced start*.elf, fixup*.dat & bootcode.bin. The option is now working. There are 2 additional buffers 0 & 1. Need both of them, so that there is no tearing seen. Will do more testing and update later. Thanks!

JamesH65, 2019-01-07:
Glad its working. I'm still seeing some issues (under different circumstance) so will continue to look in to this.

JamesH65, 2019-01-07:
So, I have now added the ability to have up to two extra transpose buffers. This seems to fix a tearing issue I was seeing on rotated HDMI displays with omxplayer running on the top. It's been accepted in to our firmware tree and should be available in rpi-update in the next few days. 

If you are happy with the current situation, feel free to close this is issue.

lwpz, 2019-01-08:
Thanks James! We will try to run omxplayer to see if tearing occurs. Is it possible to share the resolution of your display and also what is the GPU memory setting? Not sure if resolution and colour depth (6 bit or 8 bit) has anything to do with it.

Is it possible to assume that we can do the same as per 4.14? I think the latest stable version now is 4.14.90? If we just grab the relevant files from the boot directory. Would it be possible?

lwpz, 2019-01-08:
We see the same issue with the 4.14 firmware files. Please see below. 

[image]

On other applications in QT, we also see the same effect, when we're swiping and scrolling.
[image]

My finger was moving the screen and it's tearing.. 

Could you advise how to get it relevant files from Github?
Thanks

lwpz, 2019-01-08:
> I'm just using a normal desktop 1920x1200 monitor, rotated, then playing 1080p video over the top.
> 
> Until I added the extra extra transpose buffer, I was getting slight tearing depending on where on the display the video was played. Note each transpose buffer takes 9MB, so using this will reduce the overall memory available.
> 
> The extra extra patch has been applied to the firmware tree, will be available in rpi-update in a few days. Not available in github as its closed source. I could probably let you have a custom version if it is urgent. Let me know.

I see. We're using only up to SVGA (800 x 600) resolution, so I guess it is not.

We're now seeing transpose buffer0 & transpose buffer1. I presume there will be another transpose buffer on top the existing 2??

No issue. I understand that it is a closed source, due to licensing. I was thinking that it should be part of the /firmware section in github? In any case, if you could provide a version to us it'll be good, since we're doing some tests now. We can just confirm it for you?

Thanks.

lwpz, 2019-01-08:
> The tranpose buffers are only necessary on rotated displays. I also use the default GPU memory settings.

Rotated display only --> Got it!

Default should be 64MB right? Theoretically, would increase GPU to 128MB be necessary?

JamesH65, 2019-01-08:
Depends what other stuff is going on. If you are using the camera, you'll probably need to increase it, or using H264, maybe. Try it and see what happens.

JamesH65, 2019-01-08:
Yes, another transpose buffer will be added if you use extra_transpose_buffer=2 on the latest firmware. Each TB is about 10MB on the display I use - the size of the buffer is the size of the display, so a smaller display requires smaller buffers.

lwpz, 2019-01-08:
> Yes, another transpose buffer will be added if you use extra_transpose_buffer=2 on the latest firmware. Each TB is about 10MB on the display I use - the size of the buffer is the size of the display, so a smaller display requires smaller buffers.

Ok. I can try it out, if you're able to share it with me? 

Size of buffer would make sense..

JamesH65, 2019-01-08:
Have uploaded the firmware set to here. This is all the different firmwares you would get on a normal install (standard, full, cutdown, debug). Not sure which one you are using, probably *_x.*

https://drive.google.com/open?id=1fsvwhbE1cxjJMT_L8_sypy8zwdiAIQc5

use extra_transpose_buffer=2 in config.txt

lwpz, 2019-01-08:
Thanks. We'll update tomorrow. Preliminary impression it looks like tearing is not present now. Update again tomorrow.

Just a check. We tried the KMS drivers, but the tearing was still present and we still had to add the transpose buffer. If you need us to help you to test it, please do advise. We can allocate resources to do so.

JamesH65, 2019-01-08:
I have no knowledge of how the KMS driver works with this stuff. I would not expect the extra buffers to have any effect, I do not think that the KMS driver uses them, but could be wrong.

popcornmix, 2019-01-08:
``vc4_kms_v3d`` will not be affected by any firmware code changes. Any tearing solution will need to be handled on arm side.
``vc4_fkms_v3d`` does use the firmware for displaying framebuffer, so may well be affected by this change.

lwpz, 2019-01-09:
> `vc4_kms_v3d` will not be affected by any firmware code changes. Any tearing solution will need to be handled on arm side.
> `vc4_fkms_v3d` does use the firmware for displaying framebuffer, so may well be affected by this change.

Ok. I understand, which means that this buffer is probably mandatory to solve this issue, since it's GPU firmware related.

lwpz, 2019-01-09:
Ok. Looks like the tearing issue is solved, using extra_transpose_buffer=2 and latest firmware (dated 7 Jan 2019). Let us run through and update by tomorrow.

Can I close the case if I  wasn't the one, whom opened it?

JamesH65, 2019-01-09:
I'll close it. Thanks for testing. The code should be in rpi-update within a week or so, once it is you can move away from the custom firware you have at the moment. There's also a minor fix to this stuff, I didn't release all the buffers correctly on shutdown which isn't in your code, but will be in the release. I don't think it will affect you  in the meantime.

closed 2019-01-09 by JamesH65

popcornmix, 2019-01-09:
Now in latest rpi-update firmware

lwpz, 2019-01-10:
@JamesH65 @popcornmix 
Thank you for your help. Always glad to do testing with the hardware. If you have anything else, please let me know.

ijash, 2019-07-30:
i agree...... it happened to me too

JamesH65, 2019-07-30:
@ijash This issue has been closed for some time - do you still have a problem with the fixes outlined above?

ijash, 2019-07-30:
> @ijash This issue has been closed for some time - do you still have a problem with the fixes outlined above?

sorry, solved it with `extra_transpose_buffer=2`

ijash, 2019-07-30:
> @ijash This issue has been closed for some time - do you still have a problem with the fixes outlined above?

been updating it to latest firmware with `rpi-update` dated to this post and did the `extra_transpose_buffer=2` in config.txt but still tearing.

JamesH65, 2019-07-31:
Are you using the vc4-fkms-v3d driver? If you don't need it, turn it off to see if that helps. The code path that uses extra_transpose_buffer is not used with this driver IIRC.

ijash, 2019-07-31:
> Are you using the vc4-fkms-v3d driver? If you don't need it, turn it off to see if that helps. The code path that uses extra_transpose_buffer is not used with this driver IIRC.

no, i'm not using that driver.

shutchinson, 2019-08-11:
Hi all,
I have a Dell P2418HT 24" touchscreen connected to a 3b+ and am using `rotate_display=1` in `/boot/config.txt`. The display rotates appropriately, but there is a slight flicker/tearing when opening new windows as the image is, seemingly, rotating. Is this a similar issue?

lwpz, 2019-08-11:
Have you tried the extra transpose buffer in the config.txt.

Does it show the symptom in non rotated mode? 

⁣Regards lwpz