raspberrypi/firmware #845: Abnormal Display on HDMI-to-DVI

georgezlei, 2017-07-21:
Pi3 connected to the display using an HDMI-to-DVI cable. EDID information is unreadable. 

When using Raspbian 4.1.*. The full screen can be initialized correctly. And the correct resolution can be configured by manually setting up in /boot/config.txt.

After a fresh installation of the latest version (4.9.35), only part of the screen can be lighted up now. All screens including splash screen and NOOBS are all displayed within the boundary of this region. And any manual configuration of resolution that is bigger than the default 640x480 will cause the blank screen.

[image]

It is obviously not a configuration issue because NOOBS can't use the full screen either.

georgezlei, 2017-07-22:
Some updates. I tried a few history releases today and found the problem actually starts from NOOBS v2.2.0. NOOBS v2.1.0 can use the full screen by default, and it can be configured to the correct screen resolution.

popcornmix, 2017-07-22:
Can you report output of
``tvservice -s``
when display is good and when it is bad?

georgezlei, 2017-08-07:
In NOOBS v2.2.0 (problematic version)
$tvservice -s
state 0x12000a [HDMI DMT (4) RGB full 4:3], 640x480 @ 60.00Hz, progressive

georgezlei, 2017-08-07:
In NOOBS v2.2.0 (good version)
```
$tvservice -s
state 0x120006 [DVI DMT (4) RGB full 4:3], 640x480 @ 60.00Hz, progressive
```

after setting
```
hdmi_group=2
hdmi_mode=58
```
```
$tvservice -s
state 0x120006 [DVI DMT (58) RGB full 16:10], 1680x1050 @ 60.00Hz, progressive
```

JamesH65, 2019-01-11:
Are you still suffering from a problem here?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2019-11-05 by JamesH65