raspberrypi/firmware #846: Regression with Linux ≥ 4.12: local-mac-address no longer applied

stapelberg, 2017-07-22:
In the Linux 4.12 release, the device tree file changed:

```diff
--- linux-4.11.8/arch/arm/boot/dts/bcm283x-rpi-smsc9514.dtsi	2017-06-29 13:03:17.000000000 +0200
+++ linux-4.12/arch/arm/boot/dts/bcm283x-rpi-smsc9514.dtsi	2017-07-03 01:07:02.000000000 +0200
@@ -1,6 +1,6 @@
 / {
 	aliases {
-		ethernet = &ethernet;
+		ethernet0 = &ethernet;
 	};
 };
```

(Upstream commit https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=10b6c0c2e2bb8cd1be682f8d36ef597e3419cb88)

This results in the `local-mac-address` change implemented in issue #613 no longer having any effect, i.e. my Raspberry Pi starts with a random MAC address.

Could you please support both, the `ethernet` and the `ethernet0` alias, to make this feature work with Linux ≥ 4.12? Thank you very much!

pelwell, 2017-07-22:
@swarren You approved the addition of the ethernet alias and are involved in U-Boot - can you explain why U-Boot now seems to require this change, just as it was a justification for the original addition of the alias?

@anholt This change breaks our "ABI", but you gave it a review - any comments?

Obviously we have to dance to the beat of the upstream drum so I'll change the firmware if I have to, but the current support is under 6 months old and I'd like to think it won't need changing again.

pelwell, 2017-07-24:
Note that this doesn't affect downstream RPi builds because the firmware looks for DT symbols and aliases and we build the DTBs with symbols enabled:
```diff
--- a/arch/arm/boot/dts/Makefile
+++ b/arch/arm/boot/dts/Makefile
[...]
+# Enable fixups to support overlays on BCM2835 platforms
+ifeq ($(CONFIG_ARCH_BCM2835),y)
+       DTC_FLAGS ?= -@ -H epapr
+       dts-dirs += overlays
+endif
```
I've put a patch in the pipeline that will try `ethernet0` as a fallback when `ethernet` isn't found.

swarren, 2017-07-26:
Quoting from the upstream kernel DT patch's description:
> An alias name should have an index number even when it is the only of its type.
> This allows U-Boot to add the local-mac-address property. Otherwise U-Boot
> skips the alias.

Unfortunately, I'm not familiar with the U-Boot issue that text refers to. I wasn't involved in the change so don't have enough context. All I can really say is that: yes aliases probably should have index numbers, so the upstream DT change does make sense. I don't know for sure whether it's a U-Boot bug (or not) that it (now?) only works with an alias with an index. I'd rather expect it to look for ethernet0 and fall back to ethernet if not found, but who knows. There have been lots of DT-related changes in U-Boot recently and frankly the implications of those changes aren't always validated against every use case. Sorry this isn't terribly helpful.

P.S. Sorry for the slow reply; I was on vacation.

anholt, 2017-07-26:
Would just adding another alias for "ethernet" back keep compatibility?  Would someone like to submit that patch?

It would be great if those that know the firmware interactions with the kernel better would participate on linux-rpi-kernel -- I'm just trying to do the best I can with the patches that come in, in time taken from my actual job (graphics), with a year-old copy of the firmware code that I can't get updated because I don't have VPN access any more.

pelwell, 2017-07-27:
Keeping the original alias as well would have avoided the problem, but you make a fair point about monitoring and participating in the mailing list - it's an area I could improve in.

lategoodbye, 2017-07-30:
linux-rpi-kernel sometimes has very big delays (1 week+) because of the stupid recipient limitation. An another option would be to add pelwell as a designated reviewer for BCM2835.

JamesH65, 2019-01-11:
4.12 is not pretty old, is this issue still relevent?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

stapelberg, 2019-01-11:
It’s fixed from my perspective, thanks to the alias which was added in https://github.com/raspberrypi/firmware/issues/846#issuecomment-317393159

closed 2019-01-11 by stapelberg