raspberrypi/firmware #849: 1676ddb broke booting with mainline DTB

notro, 2017-07-28:
I'm using mainline 4.13.0-rc1 on Pi1 with:
```
device_tree=bcm2835-rpi-b-plus.dtb
```
and overlays enabled: https://www.raspberrypi.org/forums/viewtopic.php?p=1159639#p1159639

When booting this is the only thing I get:
```
Uncompressing Linux... done, booting the kernel.
```

If I remove device_tree= and the downstream DTB is used (not sure which version), it boots up (but no dwc2 etc.).

The previous commit bdddae2 is fine.

Kernel and dtb: http://tronnes.org/downloads/bcm2835-rpi-b-plus.zip

popcornmix, 2017-08-01:
I've reproduced. I think this will fix it (and will be in next firmware update)
https://drive.google.com/uc?id=0B-6zmEDJwxZETGRVazhQRXlzazg&export=download

There was a fix that disabled firmware side display_rotate when using vc4-kms-v3d driver
in the firmware that caused your problem. The fix effectively disabled the firmware display
earlier when vc4 driver was detected. That caused the width/height passed to kernel
(through cmdline and device tree) to be zero which seemed to cause your boot failure.

I've got a new fix that fixes the display_rotate issue in a less visible way.

notro, 2017-08-02:
Thanks, it's booting now.

closed 2017-08-02 by notro

pelwell, 2017-08-07:
I just ran foul of this on an upstream 32-bit kernel. Can you push the fix to rpi-update?

popcornmix, 2017-08-08:
rpi-update should now include the fix from test firmware.