raspberrypi/firmware #852: Bluetooth causes ticks noise on analog sound output(jack)

ShFil119, 2017-08-02:
It looks like old, but still alive bug.
https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=30669

P33M, 2017-08-02:
What model of Raspberry Pi are you using where this issue is exhibited?

ShFil119, 2017-08-02:
It's Rpi 3 with Raspian lite. It's working as sound server in lan network. For this purpose I use pulseaudio pulseaudio-module-zeroconf and avahi-daemon.

ShFil119, 2017-08-02:
It's Raspberry 3 with raspbian lite. It's working as sound server in lan network. For this purpose I use pulseaudio pulseaudio-module-zeroconf and avahi-daemon.

P33M, 2017-08-02:
So the onboard Bluetooth, correct? What peripherals are paired via Bluetooth - if you unpair them, does the problem go away completely - i.e. with Bluetooth active but not used is the audio disruption not present?

ShFil119, 2017-08-02:
I haven't been using rpi's bluetooth. 
Raspbian has default default enabled bluetooth service and after disabling it, problem gone.

JamesH65, 2019-01-11:
Closing this issue as questions answered/issue resolved.

closed 2019-01-11 by JamesH65