raspberrypi/firmware #853: New firmware doesn't boot kernel Image of arm64 with device_tree

SandPox, 2017-08-05:
Hello

Commit 1676ddb0792f1d1678efed0399c55128bc8203a6 broke ability to boot arm64 kernel Image (upstream kernel.org kernel), the firmware commit before that booted fine.

I've tested current firmware version and tracing back to that revision and found it's broken from that git revision.

Test environment:
I compiled kernel 4.12.4 from kernel.org with *gcc-linaro-6.3.1-2017.05-x86_64_aarch64-linux-gnu* to get Image and bcm2837-rpi-3-b.dtb
I put both into /boot partition with config.txt options:
device_tree=bcm2837-rpi-3-b.dtb
kernel=Image

popcornmix, 2017-08-05:
Probably this issue: https://github.com/raspberrypi/firmware/issues/849
Can you confirm that firmware posted there resolves the issue?

SandPox, 2017-08-05:
Yes, the firmware posted there did resolve the issue

But there is no power on USB ports, neither the commit bdddae23fa06822e182a180744f05185196439db have any power, I'll try to tracing some more to see which commit that broke USB power.

closed 2017-08-05 by SandPox

popcornmix, 2017-08-08:
rpi-update should now include the fix from test firmware.

SandPox, 2017-08-08:
oh yes, I do confirm the current firmware fixed the problem, thanks