raspberrypi/firmware #854: RPi3 USB boot mode: fails to initialize Microchip's USB2640 or USB2244 devices

craigjwvup, 2017-08-08:
Having successfully booted my RPi3 over usb with a generic usb to sd card reader, I thought i would give some others a try :)
Seeing that RPi has made good use of the LAN95xx product line, i decided to design an embedded usb hub/sd card reader using the Microchip product line. I have 2 evaluation boards from Microchip (EVB-USB2640 & EVB-USB2240), both chip-sets contain usb to sd card bridges, however the 2640 also includes an additional 2 port USB 2.0 hub.
Ive tried to boot through the board(s) sd card and integrated 2 port USB 2.0 hub but no joy. Both show no indication of activity or port power control is being toggled to the downstream usb ports (USB2640). 
Ive tried:
latest RASPBIAN JESSIE LITE (July 2017)
latest bootcode.bin
multiple sd cards
Self and Bus powered configurations.
adding program_usb_boot_timeout to the config.txt
changing various register settings of the devices.

Ive been crunching on this problem for about a week now and im at a loss, any suggestions?
Thanks

popcornmix, 2017-08-08:
Ping @ghollingworth

ghollingworth, 2017-08-09:
What happens if you plug your evaluation boards into a running Raspberry Pi?  Can you post here the result of doing:

sudo lsusb -v

I would assume the reason it doesn't work is because the device doesn't appear as a USB mass storage device or that it requires some unusual communication (if there are multiple volumes it will only try volume 0 for example)

craigjwvup, 2017-08-09:
Thanks for the quick response.  
Below is the output of lsusb -v also include a diagram of the bus and a snippet from the usb2640 datasheet. Is the note shown in the datasheet what you are describing in your previous post as "multiple volumes?" Or are you referring to mass storage volumes?
note: the Bus 001 Device 007: ID 05e3:0741 Genesys Logic, Inc. is the generic sd card reader that works for USB boot when attached directly to the RPi's LAN9514 HUB

 
Bus 001 Device 011: ID 0424:2240 Standard Microsystems Corp.
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0424 Standard Microsystems Corp.
  idProduct          0x2240
  bcdDevice            1.98
  iManufacturer           1 Generic
  iProduct                2 Ultra Fast Media
  iSerial                 3 000000225001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               96mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  bNumConfigurations      1
[... 434 lines cut ...]
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered

[image]

[image]

ghollingworth, 2017-08-15:
There's nothing that jumps out immediately here...  As far as I can tell it should enumerate correctly.

You've tried it with bootcode.bin only on an SD card?  Have you tried it with a compute module?  It would then remove the 9514 from the equation...

Gordon

craigjwvup, 2017-08-15:
I do have a few compute modules (v1.1) and the breakout board. After a few try's with the CM I couldn't get things working, probably a setup issue on my end with the boot gpio jumper settings....
So moving to the next option, I removed the LAN9514 from my RPi3 and re-routed the BCM2837 USB signal traces to the upper usb port of J11 (see image).  

With the LAN9514 removed from the equation, I successfully tested the board modification by booting to my (working) generic usb card reader. Unfortunately, the USB2640 and USB2244 behaviors have remained unchanged :(  
I also tried the bootcode.bin only on the sd cards, no change.

[image]

lurch, 2017-08-16:
I think Gordon meant bootcode.bin only on the SD card connected to the Pi3's on-board SD slot? (with the rest of the files still on your USB-connected SD card)
See https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/

craigjwvup, 2017-08-16:
Hey Lurch, thanks for your input. The documentation you referenced is one of the tutorials I used to get the usbboot mode working with my generic card reader. As to the bootcode.bin file on the primary sd card, this would cause the pi to boot to the primary sd according to https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow.md
and without the supporting OS files the pi would just sit and blink the green ACT led, off hand i think its a series of 4 blinks followed by a pause then repeated.
It is true that you need to boot from the primary sd first in order to do a one time setup of the usb boot mode via the OTP bit. 

My understanding is that, once this is accomplished the primary sd can be removed entirely from the system (or be blank) so the boot flow continues by searching each enabled method for a storage volume with the .bin file. Once found the pi boots from that storage volume.

lurch, 2017-08-16:
@craigjwvup Ahh, I found a better reference: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/
"...and in situations where the Pi 3 fails to boot..." ;-)

pelwell, 2017-08-16:
> As to the bootcode.bin file on the primary sd card, this would cause the pi to boot to the primary sd according to ...

Actually it just says:

> If enabled: check primary SD for bootcode.bin on GPIO 48-53
> * Success - Boot

The documented boot flow ends once it has found and loaded the second-stage loader (bootcode.bin) - the rest is just software - so both documents are correct.

ghollingworth, 2017-08-16:
No Craig has done the thing I wanted to understand, so it's something specifically about the USB2640 and USB2244 devices that the bootcode doesn't like.

To move forward I can add some debug to the bootcode.bin only version of the code to identify the problem but there's nothing I can do about the bootrom.  It's possible there is a hardware fix that could be done (it's possible it's a waking from reset type problem for example) though so it's still worth investigating!

Let me start by adding some serial debug (I assume you've got a serial USB cable you could plug into the Pi?)...  Give me a day or two...

Gordon

craigjwvup, 2017-08-16:
Thanks Gordon, your clarification and efforts are much appreciated!

Yes, I do have the ability to serial debug. I also have a fair amount of other diagnostic equipment if necessary...though sadly not a hardware USB protocol/packet analyzer. 
Meanwhile, Ive opened a support ticket with microchip and im digging deeper into the hardware/firmware fixes that are controllable on my end. I'll let you know if i dig something up that might shed more light on the issue(s).

ghollingworth, 2017-08-16:
Shame, the USB analyser would tell us a lot, although the debug output should be enough...

ghollingworth, 2017-08-18:
[bootcode.zip](https://github.com/raspberrypi/firmware/files/1234386/bootcode.zip)
OK,

Added debug output...  Attaching a bootcode.bin here

Now put this on an SD card with a single file called 'uart' it should now start outputting at 115200 from the usual pins on the Pi (GPIO 14 and 15)

Gordon

craigjwvup, 2017-08-21:
sorry for the delay, out for the eclipse. Quick question, do you want the SD card containing only the debugged bootbin and uart file in the pi's primary slot? or downstream of the USB2640?

ghollingworth, 2017-08-22:
Yes in the primary slot... This should then boot normally using bootcode.bin but because the rest of the files (start.elf, config.txt) are not on there then it'll enumerate all the devices and should give lots of information

craigjwvup, 2017-08-22:
K, im having trouble getting anything out on the serial port. 
Heres what ive done:
SD card contains the 2 files (see pic) 
Tried 2 different sd cards and RPi 3's, no joy.
Led activity is solid red with 4 short green flashes followed by a long green then repeated indefinably.
RPi's are consuming ~ 200mA during this process (except the RPi3 with the LAN chip removed)
serial baud 115200
My usb serial cable is connected correctly and is working, I can see the serial console/kernel output if i switch to a sd card containing the os with serial enabled.

[image]

ghollingworth, 2017-08-22:
You've got the serial connected to the right gpio pins?

craigjwvup, 2017-08-22:
I'm pretty sure, my usb/serial ftdi cable is connected to the RPi3 like so:

FTDI................................................RPi3
Tx                      ->               RxD0 (GPIO15 Pin10)
Rx                      ->               TxD0 (GPIO14 Pin8)
Gnd                   ->                Gnd (via Pin14)

for sanity i disconnected the FTDI and through on logic sniffer, both pins remain low with no activity at all 
[image]

ghollingworth, 2017-08-23:
[bootcode.zip](https://github.com/raspberrypi/firmware/files/1244800/bootcode.zip)
OK,

Just tested on a Pi 3 and a compute modules (previously only tested on CM3), and it seems to work fine for me, just in case there was a copy paste issue, I've rebuilt it (there is also some more debug available)...

Looking at the way I uploaded it, I think I may have uploaded the wrong thing anyway, which will explain it!

Gordon

craigjwvup, 2017-08-23:
Tried the new file, no luck. after lunch in an hour or so, ill try it with my CM and post the results. meanwhile here is what ive done thus far:
using a stock RPi3 with OTP set and verified to be 
17:3020000a 
My steps for sd card and powerup:
1.using chrome on win7 download new bootcode.zip from your last post
2.using 7zip, unzip the bootcode.bin to newly formatted (FAT32) sd card (note: formtting was done with SDFormatter program)
3. add a new file called uart with no file extension to the sd (tried it with win and Linux for generating the uart file)
4. insert sd card into RPi Primary solt
5. power up the RPi with RUN Pin Pulled low
6. Start Logic capture
7. Release the RPi RUN Pin
8. Wait for Serial output.....

attached are 2 logic outputs of the RPi GPIO14-15 and ACT LED (taken from Q6)
the first logic was taken using a sd with **normal OS + enable_uart=1** 
[image]
 
the next logic was taken with sd **containing only bootcode.bin+uart file**
[image]

Ive also tried different SD cards, no luck...

craigjwvup, 2017-08-23:
K, tried it with my CM's no luck.
I can see activity on the serial only if an OS is present on the emmc, same result as posted above except the CM will hold the ACT LED high continuously when the emmc contains only the .bin and uart file

ghollingworth, 2017-08-24:
Ah...  Make sure the uart file has something in it...

Gordon

ghollingworth, 2017-08-24:
[bootcode.zip](https://github.com/raspberrypi/firmware/files/1248321/bootcode.zip)

Problem was that it ignored empty files...  So doing 'touch uart' wouldn't work you'd have to edit the file and add content.  This one will work with an empty file!

My bad  ;)

craigjwvup, 2017-08-24:
No worries, now were getting somewhere!
Here is my hardware setup I'm using and the relevant dumps for both the RPi3, followed by the CM.
The CM has 2 dumps due to the single USB connection.   
fyi I'll be out of my office 5pm MST today and will be gone for the weekend, let me know if you need me to do some more debugging before then, if not ill be back mon morning.
[image]

.............................................................RPi3 uart dump......................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Found device on port 2
Found highspeed device
Device found: type = Mass storage, addr = 6
Found device on port 3
Found highspeed device
Hub device found at addr 7, enumerating HUB
Initialise hub
Found 3 ports

Raspberry Pi Bootcode

.............................................................CM uart dump-USB2244......................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Device found: type = Mass storage, addr = 4
Trying booting from MSD device addr 4
Failed initialising mass storage device
.............................................................CM uart dump-USB2640......................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 3 ports

Raspberry Pi Bootcode

ghollingworth, 2017-08-25:
Can you try this version:

I think I understand the problem with the hub, I think it's because it's a single TT hub and it is failing the set_interface call I make on the device (it's not actually necessary on a single TT hub)

The attached has a little more diagnosis and skips the set_interface...

[bootcode.zip](https://github.com/raspberrypi/firmware/files/1251537/bootcode.zip)

craigjwvup, 2017-08-28:
Cool! this time round the port power leds are turning on on the downstream devices!

.............................................................RPi3 uart dump.....................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 0, enabling interface 0
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Found device on port 2
Found highspeed device
Hub device found at addr 6, enumerating HUB
Initialise hub
Found 3 ports, multi_tt = 0, enabling interface 0
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Mass storage, addr = 7
Found device on port 2
Found highspeed device
Device found: type = Mass storage, addr = 8
Found device on port 3
Found highspeed device
Device found: type = Mass storage, addr = 9
Trying booting from MSD device addr 7
Failed initialising mass storage device
Trying booting from MSD device addr 8
Failed initialising mass storage device
Trying booting from MSD device addr 9
Failed initialising mass storage device
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5

.............................................................CM uart dump-USB2244......................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Device found: type = Mass storage, addr = 4
Trying booting from MSD device addr 4
Failed initialising mass storage device

.............................................................CM uart dump-USB2640......................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 3 ports, multi_tt = 0, enabling interface 0
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Mass storage, addr = 5
Found device on port 2
Found highspeed device
Device found: type = Mass storage, addr = 6
Trying booting from MSD device addr 5
Failed initialising mass storage device
Trying booting from MSD device addr 6
Failed initialising mass storage device

ghollingworth, 2017-09-04:
Sorry, back from holiday in New York!

Next step is to actually get the MSD booting working, I don't know if it's just timing out or whether there is a further bug in the initialisation.  I'm going to have to add extra debug logging for you...

Will return tomorrow

ghollingworth, 2017-09-05:
I've added some extra debug information to bootcode.bin, hopefully this should give us some information about why it's not able to initialise the MSD.

[bootcode.zip](https://github.com/raspberrypi/firmware/files/1276873/bootcode.zip)

craigjwvup, 2017-09-06:
Hope the New York trip was fun :)

Tried your file. The RPi3 and CM failed to power up the downstream devices. also i needed to add something to the uart file so things would work. Sure you sent the right bootbin? Below are the dumps just in case.
.............................................................RPi3 uart dump.....................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Found device on port 2
Found highspeed device
Hub device found at addr 6, enumerating HUB
Initialise hub
Found 3 ports

Raspberry Pi Bootcode

.............................................................CM uart dump-USB2244......................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Device found: type = Mass storage, addr = 4
Trying booting from MSD device addr 4
Failed initialising mass storage device

.............................................................CM uart dump-USB2640......................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 3 ports

Raspberry Pi Bootcode

ghollingworth, 2017-09-06:
I've checked this and am happy this is the right version of the file...

[bootcode.zip](https://github.com/raspberrypi/firmware/files/1280540/bootcode.zip)

craigjwvup, 2017-09-06:
k, now the downstream devices are being powered on. here is the dump using your last bootbin
.............................................................RPi3 uart dump.....................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Found device on port 2
Found highspeed device
Hub device found at addr 6, enumerating HUB
Initialise hub
Found 3 ports, multi_tt = 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Mass storage, addr = 7
Found device on port 2
Found highspeed device
Device found: type = Mass storage, addr = 8
Found device on port 3
Found highspeed device
Device found: type = Mass storage, addr = 9
Trying booting from MSD device addr 7
Retrieving the config descriptor
Found ep_in = 00000002, ep_out = 00000002
GetMaxLun returned 0, max_lun = 0
Sending and retrying TEST_UNIT_READY
Failed to test volume, probably not ready
init_msd returned -2
Failed initialising mass storage device
Trying booting from MSD device addr 8
Retrieving the config descriptor
Failed to read short config descriptor -2
Failed initialising mass storage device
Trying booting from MSD device addr 9
Retrieving the config descriptor
Failed to read short config descriptor -2
Failed initialising mass storage device
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5
Trying booting from Ethernet device addr 5

.............................................................CM uart dump-USB2244......................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Device found: type = Mass storage, addr = 4
Trying booting from MSD device addr 4
Retrieving the config descriptor
Found ep_in = 00000002, ep_out = 00000002
GetMaxLun returned 0, max_lun = 0
Sending and retrying TEST_UNIT_READY
Failed to test volume, probably not ready
init_msd returned -2
Failed initialising mass storage device

.............................................................CM uart dump-USB2640......................................................................
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 3 ports, multi_tt = 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Mass storage, addr = 5
Found device on port 2
Found highspeed device
Device found: type = Mass storage, addr = 6
Trying booting from MSD device addr 5
Retrieving the config descriptor
Found ep_in = 00000002, ep_out = 00000002
GetMaxLun returned 0, max_lun = 0
Sending and retrying TEST_UNIT_READY
Failed to test volume, probably not ready
init_msd returned -2
Failed initialising mass storage device
Trying booting from MSD device addr 6
Retrieving the config descriptor
Failed to read short config descriptor -2
Failed initialising mass storage device

ghollingworth, 2017-09-07:
Looks like the mass storage isn't responding to the test_unit_ready SCSI command, not sure if this is a timing thing (just timing out too quickly).  To get any further I'd need to attach a protocol analyser...

In the end though it's never going to be able to boot from the device without have an eMMC or SD card on board...

craigjwvup, 2017-09-07:
I forgot to mention, since the port powers are being enabled  (changes you made to the bootcode.bin that fixed the multi_tt issue) ive installed 2017-07-05-raspbian-jessie-lite on the downstream SD cards
  
K, Im weighing out some options here... I guess they depend on the following questions and probably more ....

1. during the time when "Sending and retrying TEST_UNIT_READY", what does the program flow / timing look like eg(timeout, num of retrys, etc.) before "Failed initialising mass storage device"?

2. can the RPi's BCMxxxx internal boot code be adapted to facilitate single TT hubs?

ghollingworth, 2017-09-08:
1. It sends a TEST_UNIT_READY command and waits for the response from the drive for 1ms.  if this times out it sends a SCSI_REQUEST_SENSE command then repeats the TEST_UNIT_READY.  This continues for a maximum for 5s

2. No, it would require taping out the chip again!

It would be better to use a different device that does work instead...

craigjwvup, 2017-09-11:
K, ill run the timing past the microchip team and see if there is something that can be changed. Meanwhile, Its probably alright to close the issue?
Thanks for the support and again your efforts are much appreciated :)

closed 2017-09-14 by popcornmix