raspberrypi/firmware #855: How to detect if hdmi display is turned on or off?

phmio, 2017-08-16:
Hi,

I wonder if there is a way at boot to detect if an hdmi display is turned on or not.

My Pi 2 is connected to an HDMI display and to the 7" Rasp. lcd display(DSI). 
What i need is if the HDMI is ON then Pi used it, but if it is OFF then Pi use the 7" Rasp lcd.

I have tried to play with those commands but they always return the same result even the HDMI display is ON or OFF :(
`$ tvservice -n`
`device_name=SAM-SAMSUNG`
`$ tvservice -s`
`state 0x12000a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive`

popcornmix, 2017-08-18:
Depends if the specific TV deasserts hotplug when switched off. Some do, some don't.

The number after state will change if hotplug is deasserted (if it doesn't change, try unplugging TV from mains, or unplugging HDMI cable - I suspect the state number will change).

phmio, 2017-08-21:
Ok popcornmix, my TV seems not to deassert hotplug when switched off :(

The state only change if i unplug/plug the HDMI cable.
`tvservice -M`
`Starting to monitor for HDMI events`
`[I] HDMI cable is unplugged`
`[I] HDMI is attached`

 and the state changes from previous (0x12000a) to:
`tvservice -n`
`[E] No device present`
`tvservice -s`
`state 0x120009 [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive` 

So to avoid manually unplug/plug each time, a workaround is maybe to make a multi-boot (eg with noobs) and as Pi use 7" Lcd by default, then select my app from the 2 choices:
- Boot my app with TV (disable 7"LCD by setting ignore_lcd=1 in config.txt)
- Boot my app with 7"LCD ( set ignore_lcd=0 and display_default_lcd=1 in config.txt)

Is there a more elegant solution?

dtrunk90, 2018-01-15:
Any news here? I need the information if the active TV source is HDMI1 (where my RasPi is connected to). I've found this while doing research: https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface But I don't have any idea how this interface can be called.

popcornmix, 2018-01-15:
@dtrunk90 can you explain what you want specifically?
Are you saying you sometimes boot with HDMI connected and sometimes without HDMI connected and want to tell which case you are in?

popcornmix, 2018-01-15:
Sorry, but what do you want from the firmware?
What command do you want to type in and what response are you hoping to get back?

dtrunk90, 2018-01-15:
The information if HDMI signals are sent or not. It it is working this way. I guessed no signals are sent if the input of my TV isn't set to the input of my RasPi (HDMI1).

dtrunk90, 2018-01-16:
Nevermind. Just had to start cec-client on my RPi to be able to control Kodi using my TV remote. :-)

closed 2019-01-11 by JamesH65