raspberrypi/firmware #857: EGL/GLES missing pkgconfig to facilitate building

psyke83, 2017-08-20:
Hi,

On Debian stretch, the libraspberrypi0 and libraspberrypi-dev packages now use renamed Broadcom libraries for GLES/EGL/OpenVG, but the packages don't include the requisite pkg-config files (which are present in the repository) to enable successful configuration.

Would you be able to package https://github.com/raspberrypi/firmware/tree/master/hardfp/opt/vc/lib/pkgconfig/brcm*.pc in future deb packages? Installing to /opt/vc/lib/pkgconfig will not be sufficient, as pkg-config looks for configuration in  /usr/lib/pkgconfig, /usr/share/pkgconfig, /usr/local/lib/pkgconfig and /usr/local/share/pkgconfig. They either need to be installed in a standard path or the PKG_CONFIG_PATH
environment variable needs to be set by default for Raspbian.

Here's an example on an issue that's arisen due to the stretch upgrade: https://github.com/libretro/RetroArch/pull/5351

psyke83, 2017-08-20:
I've sent a PR related to this issue: https://github.com/raspberrypi/userland/pull/418

If you merge the PR above and include /opt/vc/lib/pkgconfig/* in libraspberrypi-dev, it will be possible to build against the Broadcom libraries by setting PKG_CONFIG_PATH=/opt/vc/lib/pkgconfig. This will also override the generic glesv2, egl and vg pkgconfig in /usr/lib/*/pkgconfig, ensuring that the new library names are always used for linking.

XECDesign, 2017-08-21:
Thanks. This will be added in the next apt firmware release.

XECDesign, 2017-08-21:
Looking at your PR, I see you want to include glesv2.pc, although in the apt release of the firmware, I was only going to go for brcm*.pc versions.

psyke83, 2017-08-21:
Let me explain the intent of the PR more clearly.

On jessie, I'm running libraspberrypi0 1.20170703-1, which contains both the originally-named libraries alongside the renamed versions:

pi@retropie:~ $ dpkg -S GLESv2.so
libraspberrypi0: /opt/vc/lib/libGLESv2.so
libraspberrypi0: /opt/vc/lib/libbrcmGLESv2.so

This is fine with the existing pkgconfig:
pi@retropie:~ $ PKG_CONFIG_PATH=/opt/vc/lib/pkgconfig pkg-config --libs glesv2
-L/opt/vc/lib -lGLESv2 -lbcm_host -lvcos -lvchiq_arm -pthread
pi@retropie:~ $ PKG_CONFIG_PATH=/opt/vc/lib/pkgconfig pkg-config --libs brcmglesv2
-L/opt/vc/lib -lbrcmGLESv2 -lbcm_host -lvcos -lvchiq_arm -pthread

However, on stretch with firmware 1.20170811:

pi@retropie:~ $ dpkg -S GLESv2.so
libraspberrypi0: /opt/vc/lib/libbrcmGLESv2.so

The system will no longer be able to link GLESv2 via /opt/vc/lib due to the original library now being removed.

However, with my PR:
pi@retropie:~ $ PKG_CONFIG_PATH=/opt/vc/lib/pkgconfig pkg-config --libs glesv2
-L/opt/vc/lib -lbrcmGLESv2 -lbcm_host -lvcos -lvchiq_arm -pthread
pi@retropie:~ $ PKG_CONFIG_PATH=/opt/vc/lib/pkgconfig pkg-config --libs brcmglesv2
-L/opt/vc/lib -lbrcmGLESv2 -lbcm_host -lvcos -lvchiq_arm -pthread

This facilitates compatibility with software that runs on multiple platforms so that gles, egl and vg can continued to be specified in automake configurations. Even on RPI, such software will continue to link against mesa libraries by default, when the PKG_CONFIG_PATH is unset.

XECDesign, 2017-08-22:
If there is going to be a distinction between glesv2 to refer to mesa's lib and brcmglesv2 to refer to the broadcom lib, why muddy the water with something like `PKG_CONFIG_PATH=/opt/vc/lib/pkgconfig pkg-config --libs glesv2`? This seems like an opportunity to draw a clear distinction between the two.

psyke83, 2017-08-22:
If that's your position, I can understand. But surely you agree that the existing glesv2, egl and vg .pc files are broken? I can rework the PR to delete the files if you want, or simply close it, if you intend to manually omit those files from the deb packages.

XECDesign, 2017-08-22:
What in the git repos is up to @pelwell and @popcornmix, but I think the apt package will ship with the brcm*.pc only, unless there's a reason to include the other ones. I may of course be missing something.

psyke83, 2017-08-22:
That's not as optimal as I would've hoped, but definitely would be appreciated.

Just to clarify, though, you don't mean literally "brcm*.pc"? Those configurations all depend on bcm_host.pc at the minimum, so it's probably best to include everything except egl.pc, glesv2.pc and vg.pc.

diogomsantos, 2017-08-23:
I have been compilling/making Kodi packages and also had problems because of this.

The libraspberrypi0 definitly needs to include the pkg-config files missing, although since now it only includes the broadcom firmware files, makes more sense to include only the respective ones and include the pkg-config of the open-source/mesa driver in the respective packages.
Also notice that rpi-update how it is now, includes the pkg-config files but also creates /opt/vc/lib/libEGL.so, libGLESv2, etc which it shouldn't because it can cause linking problems at compile time. This should be only in /usr/lib/arm-linux-gnueabih/*, and created only by the mesa packages.

XECDesign, 2017-08-24:
Any problem with this approach?
https://github.com/RPi-Distro/firmware/commit/2440d3312fd900452b4e91b4a42325cbc75c1e2d

psyke83, 2017-08-24:
This looks good - thank you! Would it not be prudent to do the same for the armel packages? I use a Pi 3, so it doesn't matter to me, but it's probably a good idea to keep things consistent for both ABI types.

XECDesign, 2017-08-25:
The armel parts of the package are a bit vestigial. We don't maintain, build or use anything related to armel, at least as far as the apt repo goes.

psyke83, 2017-08-25:
Ah, I didn't scroll far enough down to see the recent armel packages were not present. Thanks for the assistance on this matter - I think the issue can be closed now.

closed 2017-08-25 by psyke83