raspberrypi/firmware #858: MMAL: IMX219: Exposure won't be controlled when using capture and preview ports simultaneously

Terminus-IMRC, 2017-08-21:
It seems that the firmware doesn't control exposure settings for IMX219 when:

* camera's preview and capture ports are used simultaneously and
* MMAL_PARAMETER_CAPTURE is set frequently.

Please try with this code: https://github.com/Terminus-IMRC/mmal-preview-and-capture-together
This program setups a camera component, renders the preview stream on the right side of your screen and the capture stream on the left side of that:

```
+---------+---------+
|         |         |
| preview | capture |
|         |         |
+---------+---------+
       /_____\
```

By default, the program works well, i.e., exposure will be controlled.
However, if `INTERVAL_BETWEEN_CAPTURES` is set to `100` for example, exposure will not controlled.

Is this reproducible on your pi?
Sorry in advance if I'm using MMAL wrong.

JamesH65, 2019-01-11:
This I think is expected, 100ms between captures is simply too short a time to run any exposure calculation loops. 10 fps for stills is probably too fast for captures anyway.

@6by9 any comment?

Terminus-IMRC, 2019-01-14:
I see.  Then we'll use the preview port only and have it split into some frames.  Thanks for your replies!

closed 2019-01-14 by Terminus-IMRC