raspberrypi/firmware #859: PXE Boot wont work with RPi 3

unix0r, 2017-08-21:
Hey Guys,, I'm trying to boot a Raspberry Pi 3 via Network but it does not work.

I followed the instructions of the raspberry pi blog. 
The Pi is configured to boot 
```
pi@pxe:~ $ vcgencmd otp_dump | grep 17:
17:3020000a
```
The TFTP Server shows on a folder that contains all files from /boot (copied) and a NFS Server serves the root Filesystem. The /etc/fstab contains only /proc and the cmdline.txt shows on the nfs server:

```
port=0
dhcp-range=192.168.0.255,proxy
log-dhcp
enable-tftp
tftp-root=/tftpboot
pxe-service=0,"Raspberry Pi Boot"
```

> dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/nfs nfsroot=192.168.0.2:/nfs/client1 rw ip=dhcp rootwait elevator=deadline

The booting Pi is requesting the files:

> 11:52:03.620870 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 49)
    192.168.0.103.49152 > 192.168.0.2.tftp:  21 RRQ "bootcode.bin" octet
11:52:03.670327 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 48)
    192.168.0.103.49152 > 192.168.0.2.tftp:  20 RRQ "bootsig.bin" octet
11:52:05.778664 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 57)
    192.168.0.103.49154 > 192.168.0.2.tftp:  29 RRQ "autoboot.txt" octet tsize 0
11:52:06.778753 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 55)
    192.168.0.103.49155 > 192.168.0.2.tftp:  27 RRQ "config.txt" octet tsize 0
11:52:07.779113 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 57)
    192.168.0.103.49156 > 192.168.0.2.tftp:  29 RRQ "recovery.elf" octet tsize 0
11:52:09.646980 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 54)
    192.168.0.103.49157 > 192.168.0.2.tftp:  26 RRQ "start.elf" octet tsize 0
11:52:10.647108 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 54)
    192.168.0.103.49158 > 192.168.0.2.tftp:  26 RRQ "fixup.dat" octet tsize 0

But the TFTP Server only serves the bootcode.bin:

> Aug  9 12:13:01 pxe dnsmasq-dhcp[618]: 653460281 available DHCP subnet: 192.168.0.255/255.255.255.0
Aug  9 12:13:01 pxe dnsmasq-dhcp[618]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
Aug  9 12:13:01 pxe dnsmasq-dhcp[618]: 653460281 PXE(eth0) b8:27:eb:f4:bb:43 proxy
Aug  9 12:13:01 pxe dnsmasq-dhcp[618]: 653460281 tags: eth0
Aug  9 12:13:01 pxe dnsmasq-dhcp[618]: 653460281 bootfile name: bootcode.bin
Aug  9 12:13:01 pxe dnsmasq-dhcp[618]: 653460281 broadcast response
Aug  9 12:13:01 pxe dnsmasq-dhcp[618]: 653460281 sent size:  1 option: 53 message-type  2
Aug  9 12:13:01 pxe dnsmasq-dhcp[618]: 653460281 sent size:  4 option: 54 server-identifier  192.168.0.2
Aug  9 12:13:01 pxe dnsmasq-dhcp[618]: 653460281 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Aug  9 12:13:01 pxe dnsmasq-dhcp[618]: 653460281 sent size: 17 option: 97 client-machine-id  00:44:44:44:44:44:44:44:44:44:44:44:44:44...
Aug  9 12:13:01 pxe dnsmasq-dhcp[618]: 653460281 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
Aug  9 12:16:15 pxe dnsmasq-tftp[618]: file /tftpboot/bootsig.bin not found
Aug  9 12:16:15 pxe dnsmasq-tftp[618]: sent /tftpboot/bootcode.bin to 192.168.0.103

After that, the ACT LED is blinking, but the RPi is not booted correctly.
I tried it also with the new raspbian stretch today.
Can anyone help?

Thank you

ghollingworth, 2017-08-21:
What is the led blinking pattern, 4 flashes, 4 long +4 short etc?

Assuming it's 4 flashes which means can't load start.elf

But it does look like dnsmasq is ignoring the subsequent requests for files, wondering whether the version has changed for stretch...

Gordon

unix0r, 2017-08-21:
The blinking pattern is 4 long and 4 short.
I tried it with dnsmasq as a tftp server with the default version of raspbian jessi, stretch and with the newest version of the website.
I also tried it with atftpd, but it resulted in the same behaviour.

The start.elf file is located in the /tftpboot directory:

> pi@pxe:~ $ ls -l /tftpboot/
total 21232
-rwxr-xr-x 1 pi pi   15660 Aug 21 09:04 bcm2708-rpi-0-w.dtb
-rwxr-xr-x 1 pi pi   15197 Aug 21 09:04 bcm2708-rpi-b.dtb
-rwxr-xr-x 1 pi pi   15456 Aug 21 09:04 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x 1 pi pi   14916 Aug 21 09:04 bcm2708-rpi-cm.dtb
-rwxr-xr-x 1 pi pi   16523 Aug 21 09:04 bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 pi pi   17624 Aug 21 09:04 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 pi pi   16380 Aug 21 09:04 bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 pi pi   50248 Aug 21 09:04 bootcode.bin
-rwxr-xr-x 1 pi pi     142 Aug 21 09:49 cmdline.txt
-rwxr-xr-x 1 pi pi    1614 Aug 21 09:04 config.txt
-rwxr-xr-x 1 pi pi   18693 Aug 21 09:04 COPYING.linux
-rwxr-xr-x 1 pi pi    2594 Aug 21 09:04 fixup_cd.dat
-rwxr-xr-x 1 pi pi    6688 Aug 21 09:04 fixup.dat
-rwxr-xr-x 1 pi pi    9834 Aug 21 09:04 fixup_db.dat
-rwxr-xr-x 1 pi pi    9830 Aug 21 09:04 fixup_x.dat
-rwxr-xr-x 1 pi pi     145 Aug 21 09:04 issue.txt
-rwxr-xr-x 1 pi pi 4581264 Aug 21 09:04 kernel7.img
-rwxr-xr-x 1 pi pi 4380840 Aug 21 09:04 kernel.img
-rwxr-xr-x 1 pi pi    1494 Aug 21 09:04 LICENCE.broadcom
-rwxr-xr-x 1 pi pi   18974 Aug 21 09:04 LICENSE.oracle
drwxr-xr-x 2 pi pi    4096 Aug 21 09:04 overlays
-rwxr-xr-x 1 pi pi  666404 Aug 21 09:04 start_cd.elf
-rwxr-xr-x 1 pi pi 5007204 Aug 21 09:04 start_db.elf
-rwxr-xr-x 1 pi pi 2868132 Aug 21 09:04 start.elf
-rwxr-xr-x 1 pi pi 3952100 Aug 21 09:04 start_x.elf

ghollingworth, 2017-08-21:
Can you tcpdump everything to a cap file?

Gordon

ghollingworth, 2017-08-21:
That doesn't contain the replies to the tftp requests, what filter did you put on there?

Gordon

ghollingworth, 2017-08-21:
Also it might be worth seeing how it works with dnsmasq v2.77 

git clone git://thekelleys.org.uk/dnsmasq.git 
cd dnsmasq
make
sudo make install

Should do it...  This has a couple of new options:

--dhcp-reply-delay=2

This will delay the dhcp reply from dnsmasq for two seconds so it appears after the standard IP address response.  This will stop the Pi from getting stuck in a timeout loop!

Gordon

unix0r, 2017-08-22:
The filter was:
`sudo tcpdump -i enxb827ebf4bb43 port not ssh -w trace3.cap`

and the behaviour with --dhcp-reply-delay=2 is the same.

> Aug 22 06:53:55 pxe dnsmasq-dhcp[1055]: 653460281 available DHCP subnet: 192.168.5.255/255.255.255.0
> Aug 22 06:53:55 pxe dnsmasq-dhcp[1055]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
> Aug 22 06:53:55 pxe dnsmasq-dhcp[1055]: 653460281 PXE(enxb827ebf4bb43) b8:27:eb:7a:ec:5c proxy
> Aug 22 06:53:55 pxe dnsmasq-dhcp[1055]: 653460281 tags: enxb827ebf4bb43
> Aug 22 06:53:55 pxe dnsmasq-dhcp[1055]: 653460281 reply delay: 2
> Aug 22 06:53:57 pxe dnsmasq-dhcp[1055]: 653460281 broadcast response
> Aug 22 06:53:57 pxe dnsmasq-dhcp[1055]: 653460281 sent size:  1 option: 53 message-type  2
> Aug 22 06:53:57 pxe dnsmasq-dhcp[1055]: 653460281 sent size:  4 option: 54 server-identifier  192.168.5.2
> Aug 22 06:53:57 pxe dnsmasq-dhcp[1055]: 653460281 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
> Aug 22 06:53:57 pxe dnsmasq-dhcp[1055]: 653460281 sent size: 17 option: 97 client-machine-id  00:44:44:44:44:44:44:44:44:44:44:44:44:44...
> Aug 22 06:53:57 pxe dnsmasq-dhcp[1055]: 653460281 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
> Aug 22 06:53:57 pxe dnsmasq-tftp[1055]: sent /tftpboot/bootcode.bin to 192.168.5.101
> Aug 22 06:53:57 pxe dnsmasq-tftp[1055]: file /tftpboot/bootsig.bin not found

> 06:53:55.448652 IP 0.0.0.0.bootpc > 255.255.255.255.bootps: BOOTP/DHCP, Request from b8:27:eb:7a:ec:5c (oui Unknown), length 320
> 06:53:55.466281 IP 192.168.5.1.bootps > 255.255.255.255.bootpc: BOOTP/DHCP, Reply, length 548
> 06:53:57.452464 IP 192.168.5.2.bootps > 255.255.255.255.bootpc: BOOTP/DHCP, Reply, length 314
> 06:53:57.453383 IP 192.168.5.101.49152 > 192.168.5.2.tftp:  21 RRQ "bootcode.bin" octet
> 06:53:57.503575 IP 192.168.5.101.49152 > 192.168.5.2.tftp:  20 RRQ "bootsig.bin" octet
> 06:53:59.611725 IP 192.168.5.101.49154 > 192.168.5.2.tftp:  29 RRQ "autoboot.txt" octet tsize 0
> 06:54:00.611813 IP 192.168.5.101.49155 > 192.168.5.2.tftp:  27 RRQ "config.txt" octet tsize 0
> 06:54:01.612184 IP 192.168.5.101.49156 > 192.168.5.2.tftp:  29 RRQ "recovery.elf" octet tsize 0
> 06:54:02.612196 IP 192.168.5.101.49157 > 192.168.5.2.tftp:  26 RRQ "start.elf" octet tsize 0
> 06:54:04.006761 IP 192.168.5.101.49158 > 192.168.5.2.tftp:  26 RRQ "fixup.dat" octet tsize 0
> 

Thank you for your help.

ghollingworth, 2017-08-22:
Ah, OK, looking at the packets it looks like the ethernet address is broken in the 2nd stage, for some reason it starts writing to address 00:00:00:00:00:00 which is why the server doesn't respond.

Not sure if this is something that has broken in the latest bootcode.bin build or if it's something about your network which causes it...  Might be worth going back a couple of months and downloading bootcode.bin directly from github to see if it changed at all in the last six months or so.

unix0r, 2017-08-22:
yeah i think it is caused by the switch. I tried it with several different switches and all showed the same behaviour.
But i tested it with a direct connection between RPi and PXE RPi now and it works!

ghollingworth, 2017-08-23:
Can you try again with the attached bootcode.bin (put it into the tftpboot directory), I think the problem is that the arp messages from the client Raspberry Pi are going missing...  If you can throw this one into the tftpboot directory:

[bootcode.zip](https://github.com/raspberrypi/firmware/files/1245930/bootcode.zip)

It should now output debug messages on pin 8 of the GPIO connector, if you've got a suitable serial adapter?

If not then just tcpdump it, a failed ARP will show up as an ethernet address of aa:aa:aa:aa:aa:aa

Gordon

unix0r, 2017-08-28:
A quickt test shows, that the ethernet addresses are all aa:aa:aa:aa:aa:aa after the bootcode.bin was delivered an bootsig failed. So the booting pi cant connect to the pxe server.
But connected to the same unmanaged switch the pi boots normally but then the filesystem cant be mounted as in issue: https://github.com/raspberrypi/firmware/issues/863

ghollingworth, 2017-09-05:
OK, so looks like the managed switch is filtering your ARP requests out for some reason, I wonder if this is another problem with the STP. (Or a problem with the ARP, although the ARP packet looks valid...)

You could try disabling STP in the switch to see if this is the problem although it's strange that it is only filtering this one packet type...

Gordon

fedorovvl, 2017-09-08:
I have same problem with cisco SF220 switch. boot process stop after bootsig.bin requested
> 22:31:33.534055 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 348)
    0.0.0.0.bootpc > 255.255.255.255.bootps: [no cksum] BOOTP/DHCP, Request from b8:27:eb:1c:ff:5b (oui Unknown), length 320, xid 0x26f30339, Flags [none] (0x0000)
          Client-Ethernet-Address b8:27:eb:1c:ff:5b (oui Unknown)
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message Option 53, length 1: Discover
            Parameter-Request Option 55, length 12:
              Vendor-Option, Vendor-Class, BF, Option 128
              Option 129, Option 130, Option 131, Option 132
              Option 133, Option 134, Option 135, TFTP
            ARCH Option 93, length 2: 0
            NDI Option 94, length 3: 1.2.1
            GUID Option 97, length 17: 0.68.68.68.68.68.68.68.68.68.68.68.68.68.68.68.68
            Vendor-Class Option 60, length 32: "PXEClient:Arch:00000:UNDI:002001"
            END Option 255, length 0
22:31:33.534331 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.0.3 tell mge-smoapp, length 28
22:31:34.535505 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    mge-smoapp.bootps > 192.168.0.3.bootpc: [udp sum ok] BOOTP/DHCP, Reply, length 300, xid 0x26f30339, Flags [none] (0x0000)
          Your-IP 192.168.0.3
          Server-IP mge-smoapp
          Client-Ethernet-Address b8:27:eb:1c:ff:5b (oui Unknown)
          file "pxelinux.0"
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message Option 53, length 1: Offer
            Server-ID Option 54, length 4: mge-smoapp
            Lease-Time Option 51, length 4: 600
            Vendor-Option Option 43, length 12: 49.57.50.46.49.54.56.46.48.46.49.48
            TFTP Option 66, length 20: "Raspberry Pi Boot   "
            Subnet-Mask Option 1, length 4: 255.255.255.0
            END Option 255, length 0
            PAD Option 0, length 0, occurs 2
22:31:34.536221 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has mge-smoapp tell 192.168.0.3, length 46
22:31:34.536250 IP (tos 0x0, ttl 64, id 13831, offset 0, flags [DF], proto ICMP (1), length 48)
    mge-smoapp > 192.168.0.3: ICMP echo request, id 24591, seq 0, length 28
22:31:34.536260 ARP, Ethernet (len 6), IPv4 (len 4), Reply mge-smoapp is-at 00:15:5d:2a:54:16 (oui Unknown), length 28
22:31:34.536567 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 49)
    192.168.0.3.49152 > mge-smoapp.tftp: [no cksum]  21 RRQ "bootcode.bin" octet
22:31:34.540613 IP (tos 0x0, ttl 64, id 19468, offset 0, flags [none], proto UDP (17), length 544)
    mge-smoapp.36667 > 192.168.0.3.49152: [bad udp cksum 0x837b -> 0x2b38!] UDP, length 516
22:31:34.540995 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
    192.168.0.3.49152 > mge-smoapp.36667: [no cksum] UDP, length 4
22:31:34.541037 IP (tos 0x0, ttl 64, id 19469, offset 0, flags [none], proto UDP (17), length 544)
    mge-smoapp.36667 > 192.168.0.3.49152: [bad udp cksum 0x837b -> 0x5fc7!] UDP, length 516
22:31:34.541316 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
    192.168.0.3.49152 > mge-smoapp.36667: [no cksum] UDP, length 4
22:31:34.541351 IP (tos 0x0, ttl 64, id 19470, offset 0, flags [none], proto UDP (17), length 544)
    mge-smoapp.36667 > 192.168.0.3.49152: [bad udp cksum 0x837b -> 0x6da6!] UDP, length 516
22:31:34.541634 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
    192.168.0.3.49152 > mge-smoapp.36667: [no cksum] UDP, length 4
22:31:34.541676 IP (tos 0x0, ttl 64, id 19471, offset 0, flags [none], proto UDP (17), length 544)
    mge-smoapp.36667 > 192.168.0.3.49152: [bad udp cksum 0x837b -> 0x41af!] UDP, length 516
22:31:34.542031 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
    192.168.0.3.49152 > mge-smoapp.36667: [no cksum] UDP, length 4
22:31:34.542052 IP (tos 0x0, ttl 64, id 19472, offset 0, flags [none], proto UDP (17), length 544)
    mge-smoapp.36667 > 192.168.0.3.49152: [bad udp cksum 0x837b -> 0x064b!] UDP, length 516
22:31:34.542312 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
    192.168.0.3.49152 > mge-smoapp.36667: [no cksum] UDP, length 4
22:31:34.542340 IP (tos 0x0, ttl 64, id 19473, offset 0, flags [none], proto UDP (17), length 544)
    mge-smoapp.36667 > 192.168.0.3.49152: [bad udp cksum 0x837b -> 0x390a!] UDP, length 516
22:31:34.542617 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
    192.168.0.3.49152 > mge-smoapp.36667: [no cksum] UDP, length 4
22:31:34.542637 IP (tos 0x0, ttl 64, id 19474, offset 0, flags [none], proto UDP (17), length 544)
    mge-smoapp.36667 > 192.168.0.3.49152: [bad udp cksum 0x837b -> 0xbc6a!] UDP, length 516
22:31:34.542913 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
    192.168.0.3.49152 > mge-smoapp.36667: [no cksum] UDP, length 4
22:31:34.542935 IP (tos 0x0, ttl 64, id 19475, offset 0, flags [none], proto UDP (17), length 544)
    mge-smoapp.36667 > 192.168.0.3.49152: [bad udp cksum 0x837b -> 0x5308!] UDP, length 516
22:31:34.543185 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
[... 26 lines cut ...]
[... cut ...]
    mge-smoapp.59240 > 192.168.0.3.49152: [bad udp cksum 0x818a -> 0x95f9!] UDP, length 19
22:31:39.540730 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.0.3 tell mge-smoapp, length 28
22:31:40.542721 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.0.3 tell mge-smoapp, length 28
22:31:41.544711 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.0.3 tell mge-smoapp, length 28

Disabling stp globally didn't help. with cisco catalist 2960 all work, and without sf220 all work too )
How else can I try to bring it to work?

fedorovvl, 2017-09-08:
ghollingworth, can you modify bootcode.bin for my with mac 00:15:5d:2a:54:16 or tell me how you doing this?

unix0r, 2017-09-13:
I was trying it with unmanaged switches.

@fedorovvl 
The Bootcode.bin was just for research purpose only to see if the problem is associated with the failing ARP requests.

aaronk6, 2018-03-28:
@unix0r Were you able to solve this? I think I’m seeing the same issue.

The tftp server only sees requests for bootcode.bin and bootsig.bin:

```
# tail -f /var/log/syslog
Mar 28 23:47:53 homeserver-new tftpd[2970]: tftpd: trying to get file: bootcode.bin
Mar 28 23:47:53 homeserver-new tftpd[2970]: tftpd: serving file from /tftpboot
Mar 28 23:47:53 homeserver-new tftpd[2972]: tftpd: trying to get file: bootsig.bin
Mar 28 23:47:53 homeserver-new tftpd[2972]: tftpd: serving file from /tftpboot
````

The Pi seems to request the other files, but apparently those requests don’t reach the tftp server:

```
sudo tcpdump -vvv -i switch0 ether host b8:27:eb:fc:55:61
23:47:47.353551 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    10.0.2.254.bootps > led-pi.bootpc: [udp sum ok] BOOTP/DHCP, Reply, length 300, xid 0x26f30339, Flags [none] (0x0000)
	  Your-IP led-pi
	  Client-Ethernet-Address b8:27:eb:fc:55:61 (oui Unknown)
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Offer
	    Server-ID Option 54, length 4: 10.0.2.254
	    Lease-Time Option 51, length 4: 86400
	    Vendor-Option Option 43, length 20: 82.97.115.112.98.101.114.114.121.32.80.105.32.66.111.111.116.32.32.32
	    TFTP Option 66, length 10: "10.0.2.159"
	    Subnet-Mask Option 1, length 4: 255.255.255.0
	    END Option 255, length 0
	    PAD Option 0, length 0, occurs 4
23:47:53.254391 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has homeserver-new tell led-pi, length 46
23:47:55.976362 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 57)
    led-pi.49154 > homeserver-new.tftp: [no cksum]  29 RRQ "autoboot.txt" octet tsize 0
23:47:57.355621 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 55)
    led-pi.49155 > homeserver-new.tftp: [no cksum]  27 RRQ "config.txt" octet tsize 0
23:47:59.267729 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 57)
    led-pi.49156 > homeserver-new.tftp: [no cksum]  29 RRQ "recovery.elf" octet tsize 0
23:48:00.939198 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 54)
    led-pi.49157 > homeserver-new.tftp: [no cksum]  26 RRQ "start.elf" octet tsize 0
23:48:01.939333 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 54)
    led-pi.49158 > homeserver-new.tftp: [no cksum]  26 RRQ "fixup.dat" octet tsize 0
```

I have a managed switch (Ubiquiti EdgeSwitch 8-port). I’ve disabled STP, as I read in a another thread that this could potentially cause issues. A similar setup was working for me a couple of month ago. I think back then I was using an unmanaged switch, so I believe it’s somehow related to the switch.

I’ll try to narrow this down by replacing the switch next, however, it would be highly appreciated if you could give me a quick update whether you made any progress on this matter! Thanks :)

pelwell, 2018-03-28:
What are homeserver-new and led-pi's IP addresses?

aaronk6, 2018-03-28:
Hey, thanks for the quick reply! Actually I just solved this: it turned out that an old router I was using as a switch (Fritzbox 7270) was causing this. After replacing it with an standard switch, the setup works perfectly fine, even with the managed EdgeSwitch in between. Sorry for the noise! 🙈

spiderpug, 2018-04-07:
Hi, I'm having a similar issue, I guess. I'm also using a FritzBox as router and my DHCP/TFTP server is a Synology NAS. This is my dump:

```
00:54:00.557089 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 348)
    0.0.0.0.bootpc > 255.255.255.255.bootps: [no cksum] BOOTP/DHCP, Request from b8:27:eb:5e:4e:6d (oui Unknown), length 320, xid 0x26f30339, Flags [none] (0x0000)
	  Client-Ethernet-Address b8:27:eb:5e:4e:6d (oui Unknown)
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Discover
	    Parameter-Request Option 55, length 12:
	      Vendor-Option, Vendor-Class, BF, Option 128
	      Option 129, Option 130, Option 131, Option 132
	      Option 133, Option 134, Option 135, TFTP
	    ARCH Option 93, length 2: 0
	    NDI Option 94, length 3: 1.2.1
	    GUID Option 97, length 17: 0.68.68.68.68.68.68.68.68.68.68.68.68.68.68.68.68
	    Vendor-Class Option 60, length 32: "PXEClient:Arch:00000:UNDI:002001"
	    END Option 255, length 0
00:54:00.558437 IP (tos 0xc0, ttl 64, id 5032, offset 0, flags [none], proto UDP (17), length 406)
    ds216j.fritz.box.bootps > Xbian.fritz.box.bootpc: [bad udp cksum 0xe6fb -> 0x1d13!] BOOTP/DHCP, Reply, length 378, xid 0x26f30339, Flags [none] (0x0000)
	  Your-IP Xbian.fritz.box
	  Server-IP ds216j.fritz.box
	  Client-Ethernet-Address b8:27:eb:5e:4e:6d (oui Unknown)
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Offer
	    Server-ID Option 54, length 4: ds216j.fritz.box
	    Lease-Time Option 51, length 4: 86400
	    BF Option 67, length 13: "bootcode.bin^@"
	    RN Option 58, length 4: 43200
	    RB Option 59, length 4: 75600
	    Subnet-Mask Option 1, length 4: 255.255.255.0
	    BR Option 28, length 4: 192.168.178.255
	    TFTP Option 66, length 14: "192.168.178.2^@"
	    Vendor-Class Option 60, length 9: "PXEClient"
	    GUID Option 97, length 17: 0.68.68.68.68.68.68.68.68.68.68.68.68.68.68.68.68
	    Vendor-Option Option 43, length 35: 6.1.3.10.4.0.80.88.69.9.23.0.0.20.82.97.115.112.98.101.114.114.121.32.80.105.32.66.111.111.116.32.32.32.255
	    END Option 255, length 0
00:54:04.331793 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 49)
    Xbian.fritz.box.49152 > ds216j.fritz.box.tftp: [no cksum]  21 RRQ "bootcode.bin" octet
00:54:04.349135 IP (tos 0x0, ttl 64, id 5144, offset 0, flags [DF], proto UDP (17), length 544)
    ds216j.fritz.box.tftp > Xbian.fritz.box.49152: [bad udp cksum 0xe785 -> 0x5624!]  516 DATA block 1
00:54:04.349414 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
    Xbian.fritz.box.49152 > ds216j.fritz.box.tftp: [no cksum]  4 ACK block 1
00:54:04.349515 IP (tos 0x0, ttl 64, id 5145, offset 0, flags [DF], proto UDP (17), length 544)
    ds216j.fritz.box.tftp > Xbian.fritz.box.49152: [bad udp cksum 0xe785 -> 0x2bef!]  516 DATA block 2
00:54:04.349782 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
    Xbian.fritz.box.49152 > ds216j.fritz.box.tftp: [no cksum]  4 ACK block 2
00:54:04.349865 IP (tos 0x0, ttl 64, id 5146, offset 0, flags [DF], proto UDP (17), length 544)
    ds216j.fritz.box.tftp > Xbian.fritz.box.49152: [bad udp cksum 0xe785 -> 0x6eb2!]  516 DATA block 3
00:54:04.350138 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
    Xbian.fritz.box.49152 > ds216j.fritz.box.tftp: [no cksum]  4 ACK block 3
...
00:54:04.382035 IP (tos 0x0, ttl 64, id 5242, offset 0, flags [DF], proto UDP (17), length 108)
    ds216j.fritz.box.tftp > Xbian.fritz.box.49152: [bad udp cksum 0xe5d1 -> 0x9c32!]  80 DATA block 99
00:54:04.382212 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 32)
    Xbian.fritz.box.49152 > ds216j.fritz.box.tftp: [no cksum]  4 ACK block 99
00:54:04.382236 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 48)
    Xbian.fritz.box.49152 > ds216j.fritz.box.tftp: [no cksum]  20 RRQ "bootsig.bin" octet
00:54:04.383259 IP (tos 0x0, ttl 64, id 5243, offset 0, flags [DF], proto UDP (17), length 55)
    ds216j.fritz.box.tftp > Xbian.fritz.box.49152: [bad udp cksum 0xe59c -> 0xa434!]  27 ERROR ENOTFOUND "No Such File/No Access"
00:54:07.340818 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 57)
    Xbian.fritz.box.49154 > ds216j.fritz.box.tftp: [no cksum]  29 RRQ "autoboot.txt" octet tsize 0
00:54:09.351138 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 55)
    Xbian.fritz.box.49155 > ds216j.fritz.box.tftp: [no cksum]  27 RRQ "config.txt" octet tsize 0
00:54:10.732272 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 57)
    Xbian.fritz.box.49156 > ds216j.fritz.box.tftp: [no cksum]  29 RRQ "recovery.elf" octet tsize 0
00:54:11.732270 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 54)
    Xbian.fritz.box.49157 > ds216j.fritz.box.tftp: [no cksum]  26 RRQ "start.elf" octet tsize 0
00:54:12.732382 IP (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 54)
    Xbian.fritz.box.49158 > ds216j.fritz.box.tftp: [no cksum]  26 RRQ "fixup.dat" octet tsize 0
00:54:21.845416 IP (tos 0x0, ttl 64, id 53519, offset 0, flags [none], proto UDP (17), length 328)
```

after receiving the `fixup.dat` the pi is only blinking: 3 short, 1 very long.

I was using the boot files or the latest `master` and with them it stops after sending the `bootcode.bin` (never requests the other files).

With https://github.com/raspberrypi/firmware/issues/958#issuecomment-375692517 it produces the log above.

What else could I try? Thanks!

cupples, 2018-04-28:
Has anyone played with the switchport ”link debounce” option on the Cisco’s? I just hit this at the office and the intermediate switche “solved” the issue, but isn’t a solution. I may try poking at it this weekend at home but thought it was worth asking first.

paulenuta, 2018-10-20:
I have a similar issue.          **[Solved by making a simple private network for the PXE environment]**
I have read and apply:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net_tutorial.md
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net.md
 
I am using „Linux nfs-pi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux” from 2018-10-09-raspbian-stretch-lite.zip.
My network has a Linux gateway for the internet and a windows 2k8R2 server for dhcp and dns, plus a managed switch, plus, on my bench, an unmanaged switch.

I have created folders like this:
```
root@nfs-pi:/home/pi# ls -al /tftpboot/
total 22156
drwxrwxrwx  5 root root    4096 Oct 18 14:48 .
drwxr-xr-x 24 root root    4096 Oct 18 09:56 ..
drwxr-xr-x  3 root root    4096 Oct 18 14:48 000000005e5a8e32
drwxr-xr-x  3 root root    4096 Oct 18 14:48 5e5a8e32
-rwxr-xr-x  1 root root   23315 Sep 19 21:06 bcm2708-rpi-0-w.dtb
-rwxr-xr-x  1 root root   22812 Sep 19 21:06 bcm2708-rpi-b.dtb
-rwxr-xr-x  1 root root   23071 Sep 19 21:06 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x  1 root root   22589 Sep 19 21:06 bcm2708-rpi-cm.dtb
-rwxr-xr-x  1 root root   24115 Sep 19 21:06 bcm2709-rpi-2-b.dtb
-rwxr-xr-x  1 root root   25311 Sep 19 21:06 bcm2710-rpi-3-b.dtb
-rwxr-xr-x  1 root root   25574 Sep 19 21:06 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root   24087 Sep 19 21:06 bcm2710-rpi-cm3.dtb
-rwxr-xr-x  1 root root   52116 Sep 25 12:06 bootcode.bin
-rwxr-xr-x  1 root root     142 Oct 18 08:29 cmdline.txt
-rwxr-xr-x  1 root root    1971 Oct 18 09:31 config.txt
-rwxr-xr-x  1 root root   18693 Mar  9  2018 COPYING.linux
-rwxr-xr-x  1 root root    2618 Sep 25 12:06 fixup_cd.dat
-rwxr-xr-x  1 root root    6660 Sep 25 12:06 fixup.dat
-rwxr-xr-x  1 root root    9893 Sep 25 12:06 fixup_db.dat
-rwxr-xr-x  1 root root    9889 Sep 25 12:06 fixup_x.dat
-rwxr-xr-x  1 root root     145 Oct  9 15:33 issue.txt
-rwxr-xr-x  1 root root 4935528 Sep 25 12:06 kernel7.img
-rwxr-xr-x  1 root root 4685760 Sep 25 12:06 kernel.img
-rwxr-xr-x  1 root root    1494 Mar  9  2018 LICENCE.broadcom
-rwxr-xr-x  1 root root   18974 Oct  9 15:33 LICENSE.oracle
drwxr-xr-x  2 root root    4096 Oct  9 14:50 overlays
-rwxr-xr-x  1 root root  677764 Sep 25 12:06 start_cd.elf
-rwxr-xr-x  1 root root 5108996 Sep 25 12:06 start_db.elf
-rwxr-xr-x  1 root root 2847044 Sep 25 12:06 start.elf
-rwxr-xr-x  1 root root 4047876 Sep 25 12:06 start_x.elf
```
 
```
root@nfs-pi:/home/pi# ls -al /nfs/client1/
total 88
drwxr-xr-x 22 root root 4096 Oct 18 08:32 .
drwxr-xr-x  3 root root 4096 Oct 18 08:32 ..
drwxr-xr-x  2 root root 4096 Oct 18 08:17 bin
drwxr-xr-x  2 root root 4096 Oct 18 08:31 boot
drwxr-xr-x  2 root root 4096 Oct  9 14:47 debootstrap
drwxr-xr-x  2 root root 4096 Oct 18 08:31 dev
drwxr-xr-x 89 root root 4096 Oct 18 10:04 etc
drwxr-xr-x  3 root root 4096 Oct 18 08:23 home
drwxr-xr-x 17 root root 4096 Oct 18 08:17 lib
drwx------  2 root root 4096 Oct  9 15:31 lost+found
drwxr-xr-x  2 root root 4096 Oct  9 14:37 media
drwxr-xr-x  2 root root 4096 Oct  9 14:37 mnt
drwxr-xr-x  4 root root 4096 Oct 18 08:26 opt
dr-xr-xr-x  2 root root 4096 Jan  1  1970 proc
drwx------  6 root root 4096 Oct 18 08:24 root
drwxr-xr-x  2 root root 4096 Oct 18 08:26 run
drwxr-xr-x  2 root root 4096 Oct 18 08:17 sbin
drwxr-xr-x  2 root root 4096 Oct  9 14:37 srv
dr-xr-xr-x  2 root root 4096 Jan  1  1970 sys
drwxrwxrwt 10 root root 4096 Oct 18 08:48 tmp
drwxr-xr-x 11 root root 4096 Oct 18 08:26 usr
drwxr-xr-x 11 root root 4096 Oct  9 15:33 var
```
[... 343 lines cut ...]
Oct 19 20:10:44 nfs-pi dnsmasq-tftp[895]: error 0 Early terminate received from 192.168.2.101
Oct 19 20:10:44 nfs-pi dnsmasq-tftp[895]: failed sending /tftpboot/5e5a8e32/kernel7.img to 192.168.2.101
Oct 19 20:10:44 nfs-pi dnsmasq-tftp[895]: file /tftpboot/5e5a8e32/armstub8-32.bin not found
Oct 19 20:10:44 nfs-pi dnsmasq-tftp[895]: file /tftpboot/5e5a8e32/armstub7.bin not found
Oct 19 20:10:44 nfs-pi dnsmasq-tftp[895]: file /tftpboot/5e5a8e32/armstub.bin not found
Oct 19 20:10:48 nfs-pi dnsmasq-tftp[895]: sent /tftpboot/5e5a8e32/kernel7.img to 192.168.2.101
Oct 19 20:10:48 nfs-pi dnsmasq-tftp[895]: sent /tftpboot/5e5a8e32/bcm2710-rpi-3-b.dtb to 192.168.2.101
Oct 19 20:10:49 nfs-pi dnsmasq-tftp[895]: sent /tftpboot/5e5a8e32/overlays/rpi-ft5406.dtbo to 192.168.2.101
Oct 19 20:10:49 nfs-pi dnsmasq-tftp[895]: sent /tftpboot/5e5a8e32/overlays/rpi-backlight.dtbo to 192.168.2.101
Oct 19 20:10:49 nfs-pi dnsmasq-tftp[895]: sent /tftpboot/5e5a8e32/config.txt to 192.168.2.101
Oct 19 20:10:49 nfs-pi dnsmasq-tftp[895]: sent /tftpboot/5e5a8e32/overlays/gpio-poweroff.dtbo to 192.168.2.101
Oct 19 20:10:49 nfs-pi dnsmasq-tftp[895]: sent /tftpboot/5e5a8e32/overlays/i2c-rtc.dtbo to 192.168.2.101
Oct 19 20:10:49 nfs-pi dnsmasq-tftp[895]: sent /tftpboot/5e5a8e32/overlays/gpio-fan.dtbo to 192.168.2.101
Oct 19 20:10:49 nfs-pi dnsmasq-tftp[895]: sent /tftpboot/5e5a8e32/overlays/pi3-disable-bt.dtbo to 192.168.2.101
Oct 19 20:10:56 nfs-pi dnsmasq-dhcp[895]: 3427905433 available DHCP subnet: 192.168.2.255/255.255.255.0
Oct 19 20:10:56 nfs-pi dnsmasq-dhcp[895]: 3427905433 available DHCP subnet: 192.168.2.255/255.255.255.0
Oct 19 20:10:56 nfs-pi rpc.mountd[438]: authenticated mount request from 192.168.2.101:739 for /nfs/client1 (/nfs/client1)
Oct 19 20:11:04 nfs-pi dnsmasq-dhcp[895]: 3575090559 available DHCP subnet: 192.168.2.255/255.255.255.0
Oct 19 20:11:04 nfs-pi dnsmasq-dhcp[895]: 3575090559 vendor class: dhcpcd-6.11.5:Linux-4.14.71-v7+:armv7l:BCM2835
Oct 19 20:11:04 nfs-pi dnsmasq-dhcp[895]: 3575090559 client provides name: pxe-pi-clone01
```

Now It is working fine on this dedicated network.

Paul

unix0r, 2018-10-23:
The new Raspberry Pi 3b+ handles PXE like a charm.

Tested with isc-dhcp and dnsmasq.
Closed.

closed 2018-10-23 by unix0r

Flole998, 2018-11-25:
@unix0r Unfortunately it doesn't. I am using isc-dhcp and the Raspberry never acknowledges the DHCP Offer (this is not okay in the first place). To make things worse, it only requests the bootcode.bin and bootcode.sig, then it sends an ARP Request to get the IP of the TFTP Server (which is on a different subnet, so this is another violation of the IP Standard) but never get's a reply. In this case it should ask the default Gateway as that knows how to route the request but that never happens, instead aa:aa.... is the receiver (another violation of the standard).

This is just completely broken. I suggest to actually read the standard and then follow it, there is a reason it's the worldwide standard and by violating it you are breaking lot's of stuff! I really hope this can be fixed in the bootcode.bin, otherwise this is a huge issue.

ghollingworth, 2018-11-25:
Since most of the code you are describing is in the bootrom then no it can't be fixed in bootcode.bin  but many other people have got this working fine.  I would suggest using a fixed DHCP range for your Raspberry Pis and serve those using dnsmasq, this would enable you to reproduce the way the code was developed and will give you a better option.

Either that or use PiServer to set up a server on an x86 or Raspberry Pi.

Flole998, 2018-11-25:
It is impossible that this will work through a Router because the raspberry is simply ignoring the default gateway (as I described before). But if this is really part of the bootrom (there was a slight chance as the bootcode.bin was loaded at that point in time).

Also after a kernel panic it gets stuck after *serialnumer*/start.elf is supposed to be loaded, no matter of that's successful or not.

This code is broken really bad and if this can't be fixed, there are quite a few devices that we have to return. I will talk to our sales representative on Monday about that. And to be honest I'm not even feeling sorry if such a bad code gets punished with quite a few returned devices (and the cost associated with that).

Flole998, 2018-11-25:
I am testing on a 3B+....

ghollingworth, 2018-11-25:
Well if the gateway is advertised in the DHCP reply then it should use it...  Obviously it'll still do an ARP for the gateway itself but shouldn't ARP for the TFTP server, just go through the gateway.

In the end Mythic Beasts who run a few hundred Pi 3s on an isc-dhcp setup have no issues except with failed packets which occasionally means a Pi won't start up.  In this situation they monitor the Pi and if it doesn't start they power it off and back on again (using a PoE managed switch to control the power).

If you can't get this working you could instead using an SD card containing only the bootcode.bin file.  It would then be possible to apply fixes to the bootcode.bin file (and output UART debug so we can see what is happening).

Once booted you can then disable the SD card peripheral to stop it from being accessed by the user (to avoid it ever getting corrupted)

Flole998, 2018-11-25:
I just double-checked to make sure that it is inded a pi3b+ that I am testing with.

At first it works perfectly fine for the bootcode.bin and bootcode.sig (might have gotten that name wrong), it's using the gateway there and absolutely no issues, but for the next file it is not working correctly (after loading the bootcode.bin the ARP Request for the TFTP Server is beeing sent). Thats why I thought that the loading of that file might already be part of the bootcode.bin and so that it could get fixed even without a costly replacement. I got around that by giving the gateway an alias on the Raspberry Side so it responds with it's MAC to that request, but this is just really bad and I would not consider this a fix at all.

I do not have any issues with missed packets at all.

In addition to all that, after the device has been reset by the watchdog, it gets stuck after trying to load the serialnumer/start.elf which is an even bigger problem as this means all the devices will get stuck if the NFS Server goes down (or the connections are interrupted). Definitely not practical to start walking around and reset all of the devices.

I tried to use SD Cards at first, but right now I have 2 of my 10 Test-Cards on my desk that are no longer working at all, even though they just contained the bootloader and kernel. They behave like a piece of plastic, not being recognized at all anymore.

In case this helps:
`# sha1sum /boot/bootcode.bin
cedaeb8daa7728e9857d199b4f03d84936499b7f  /boot/bootcode.bin
# /opt/vc/bin/vcgencmd version
Nov  4 2018 16:31:07
Copyright (c) 2012 Broadcom
version ed5baf9520a3c4ca82ba38594b898f0c0446da66 (clean) (release)`

ghollingworth, 2018-11-25:
Well in that case we can do something about it...  So first we could probably just trying sorting out the problem with the ARP request.

I'd suggest first of all connecting a UART serial debug to the Pi using GPIO13,14 on pins 8 and 10 of the GPIO connector.  https://pinout.xyz/pinout/uart

Then do the magic to bootcode.bin to enable early UART output

sed -i -e "s/BOOT_UART=0/BOOT_UART=1/" bootcode.bin

Keep a backup of the original file just in case.  It should output at 115200-8-n-1

Then post the UART log and we'll see what you get

Flole998, 2018-11-25:
Alright, so the first thing I have done is simply removed my ARP Hack so that it ARP-Requests an IP that is not on that subnet. The log is very simple for that:

`<\r>Raspberry Pi Bootcode<\n><\r><\n>
<\r>USB ethernet boot<\n><\r>No response from ARP request<\n><\r><\n>
`

Now talking about the issue after a watchdog reset: I added my Workaround again and then let it boot, then I started the logging and made the watchdog trigger:

`<\r>Raspberry Pi Bootcode<\n><\r><\n>
<\r>USB ethernet boot<\n><\r>Done ARP for x.x.x.x got 10:21:19:02:88:98<\n><\r><\n>
<\r><\n>
<\r><\n>
<\r><\n>
<\r>Raspberry Pi Bootcode<\n><\r>`

and now it's stuck there, last file requested was *myserial*/start.elf

pelwell, 2018-11-25:
Two (probably) related issues for comparison: https://github.com/raspberrypi/firmware/issues/937 and  https://github.com/raspberrypi/firmware/issues/975

Flole998, 2018-11-25:
The first issue doesn't seem to be related, in that one it doesn't seem to work even when the TFTP Server is on the same Subnet. The second one is also not related, my DHCP Discover works perfectly fine, the bootcode.bin is being requested and delivered, which is why we are (thankfully) talking about an issue here that can be fixed in software.

fedorovvl, 2018-11-26:
if it help someone..
we use about 400 rpi3 with pxe+nfsroot for a 1 year in production with following scheme:
6 subnets on tftp/dhcp server with AA:AA:AA:AA:AA:AA mac on each interface
```
eth0.10: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.10  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::a8aa:aaff:feaa:aaaa  prefixlen 64  scopeid 0x20<link>
        ether aa:aa:aa:aa:aa:aa  txqueuelen 1000  (Ethernet)
        RX packets 6619806  bytes 499133814 (476.0 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 66137  bytes 2779178 (2.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
eth0.20: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.10  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::a8aa:aaff:feaa:aaaa  prefixlen 64  scopeid 0x20<link>
        ether aa:aa:aa:aa:aa:aa  txqueuelen 1000  (Ethernet)
        RX packets 6619806  bytes 499133814 (476.0 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 66137  bytes 2779178 (2.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
.......
```
6 subnet declaration on dhcp server
```
subnet 192.168.1.0 netmask 255.255.255.0 {
    option routers 192.168.1.1;
    option option-43 "192.168.1.10";
    option tftp-server-name "192.168.1.10";
    next-server 192.168.1.10;
    option root-path "192.168.1.10:/opt/nfsroot,vers=3";
    pool {
        range 192.168.1.11 192.168.1.250;
    }
}
subnet 192.168.2.0 netmask 255.255.255.0 {
    option routers 192.168.2.1;
    option option-43 "192.168.2.10";
    option tftp-server-name "192.168.2.10";
    next-server 192.168.2.10;
    option root-path "192.168.2.10:/opt/nfsroot,vers=3";
    pool {
        range 192.168.2.11 192.168.2.250;
    }
}
```
same for exports
```
/opt/nfsroot 192.168.1.0/24(rw,sync,no_subtree_check,no_root_squash)
/opt/nfsroot 192.168.2.0/24(rw,sync,no_subtree_check,no_root_squash)
```
cmdline.txt
```
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/nfs rw ip=dhcp rootwait elevator=deadline
```
and finally! we use bootcode from comment above, coz it use AA mac when ARP fail

Flole998, 2018-11-26:
The "default" bootcode uses the aa:... MAC aswell, but seriously, that aa:... thing is just extremely bad. A MAC is also supposed to be unique....

If you get the chance, could you please try to cause a kernel panic on one of them and wait for it to reboot and see if it get's past requesting *serialnumber/start.elf? That's an issue I am experiencing even with the aa;... mac set

fedorovvl, 2018-11-27:
my pi successfully boots after panic.. here is a log
```
Nov 27 08:03:52 bootserver dhcpd: DHCPDISCOVER from b8:27:eb:xx:xx:xx via ens160
Nov 27 08:03:53 bootserver dhcpd: DHCPOFFER on 192.168.1.63 to b8:27:eb:xx:xx:xx via ens160
Nov 27 08:03:53 bootserver in.tftpd[3947]: Client ::ffff:192.168.1.63 finished bootcode.bin
Nov 27 08:03:53 bootserver in.tftpd[3951]: Client ::ffff:192.168.1.63 finished config.txt
Nov 27 08:03:55 bootserver in.tftpd[3953]: Client ::ffff:192.168.1.63 finished start.elf
Nov 27 08:03:55 bootserver in.tftpd[3954]: Client ::ffff:192.168.1.63 finished fixup.dat
Nov 27 08:03:55 bootserver in.tftpd[3956]: Client ::ffff:192.168.1.63 finished config.txt
Nov 27 08:03:55 bootserver in.tftpd[3959]: Client ::ffff:192.168.1.63 finished config.txt
Nov 27 08:03:56 bootserver in.tftpd[3961]: Client ::ffff:192.168.1.63 finished cmdline.txt
Nov 27 08:04:00 bootserver in.tftpd[3967]: Client ::ffff:192.168.1.63 finished kernel8.img
Nov 27 08:04:00 bootserver in.tftpd[3968]: Client ::ffff:192.168.1.63 finished bcm2710-rpi-3-b.dtb
Nov 27 08:04:00 bootserver in.tftpd[3969]: Client ::ffff:192.168.1.63 finished config.txt
Nov 27 08:04:00 bootserver in.tftpd[3970]: Client ::ffff:192.168.1.63 finished overlays/w1-gpio.dtbo
Nov 27 08:04:00 bootserver in.tftpd[3971]: Client ::ffff:192.168.1.63 finished overlays/pi3-disable-wifi.dtbo
Nov 27 08:04:00 bootserver in.tftpd[3972]: Client ::ffff:192.168.1.63 finished overlays/vc4-fkms-v3d.dtbo
Nov 27 08:04:12 bootserver dhcpd: DHCPDISCOVER from b8:27:eb:xx:xx:xx via ens160
Nov 27 08:04:12 bootserver dhcpd: DHCPOFFER on 192.168.1.63 to b8:27:eb:xx:xx:xx via ens160
Nov 27 08:04:12 bootserver dhcpd: DHCPREQUEST for 192.168.1.63 (192.168.44.2) from b8:27:eb:xx:xx:xx via ens160
Nov 27 08:04:12 bootserver dhcpd: DHCPACK on 192.168.1.63 to b8:27:eb:xx:xx:xx via ens160
Nov 27 08:04:12 bootserver rpc.mountd[1029]: refused mount request from 192.168.1.63 for /opt/fakenfs (/): not exported
Nov 27 08:04:17 bootserver rpc.mountd[1029]: refused mount request from 192.168.1.63 for /opt/fakenfs (/): not exported
Nov 27 08:04:27 bootserver rpc.mountd[1029]: refused mount request from 192.168.1.63 for /opt/fakenfs (/): not exported
Nov 27 08:04:48 bootserver rpc.mountd[1029]: refused mount request from 192.168.1.63 for /opt/fakenfs (/): not exported
Nov 27 08:05:18 bootserver rpc.mountd[1029]: refused mount request from 192.168.1.63 for /opt/fakenfs (/): not exported
Nov 27 08:05:48 bootserver rpc.mountd[1029]: refused mount request from 192.168.1.63 for /opt/fakenfs (/): not exported

!panic!

Nov 27 08:06:18 bootserver dhcpd: DHCPDISCOVER from b8:27:eb:xx:xx:xx via ens160
Nov 27 08:06:19 bootserver dhcpd: DHCPOFFER on 192.168.1.63 to b8:27:eb:xx:xx:xx via ens160
Nov 27 08:06:19 bootserver in.tftpd[4081]: Client ::ffff:192.168.1.63 finished bootcode.bin
Nov 27 08:06:19 bootserver in.tftpd[4085]: Client ::ffff:192.168.1.63 finished config.txt
Nov 27 08:06:21 bootserver in.tftpd[4087]: Client ::ffff:192.168.1.63 finished start.elf
Nov 27 08:06:21 bootserver in.tftpd[4088]: Client ::ffff:192.168.1.63 finished fixup.dat
Nov 27 08:06:22 bootserver in.tftpd[4090]: Client ::ffff:192.168.1.63 finished config.txt
Nov 27 08:06:22 bootserver in.tftpd[4093]: Client ::ffff:192.168.1.63 finished config.txt
Nov 27 08:06:22 bootserver in.tftpd[4095]: Client ::ffff:192.168.1.63 finished cmdline.txt
Nov 27 08:06:26 bootserver in.tftpd[4101]: Client ::ffff:192.168.1.63 finished kernel8.img
Nov 27 08:06:26 bootserver in.tftpd[4102]: Client ::ffff:192.168.1.63 finished bcm2710-rpi-3-b.dtb
Nov 27 08:06:26 bootserver in.tftpd[4103]: Client ::ffff:192.168.1.63 finished config.txt
Nov 27 08:06:26 bootserver in.tftpd[4104]: Client ::ffff:192.168.1.63 finished overlays/w1-gpio.dtbo
Nov 27 08:06:26 bootserver in.tftpd[4105]: Client ::ffff:192.168.1.63 finished overlays/pi3-disable-wifi.dtbo
Nov 27 08:06:26 bootserver in.tftpd[4106]: Client ::ffff:192.168.1.63 finished overlays/vc4-fkms-v3d.dtbo
```

Flole998, 2018-11-27:
That does indeed look perfectly fine. I will do a Wireshark capture again tomorrow to see what's going on here in this case. I'm causing the panic by writing "c" to sysrq and I have the watchdog enabled, just in case that matters. Could you please also post the checksum of the bootcode.bin, just to be sure. If i remember correctly there was no request to config.txt before the *serialnumber*/start.elf.

Flole998, 2018-12-05:
The issue I am having is also in #1078 and in there a few other issues are mentioned aswell. The bootcode.bin needs to be updated to include a fix for that, it's ridiculous that the bootrom (which is harder to update) is doing this correctly and the bootcode.bin which can be updated easily is still broken!