raspberrypi/firmware #860: Bluetooth Firmware MAC Address Issue

lsgunth, 2017-08-22:
While upgrading from Jessie to Stretch, I found the latest Bluetooth firmware has a weird issue with the MAC Address it initializes the bluetooth adapter with.

With bluez-firmware version 1.2-3+rpi2, the mac address is set to 43:43:A1:12:1F:AC. This may be constant across hardware seeing I've found reference to it on the internet [1]. Besides that, this weird MAC address breaks  bluetooth PAN because the bnep adapter gets rejected as a valid device when adding it to bridge. This is because the mac address is actually a multicast address (bit 1 is set in the first octet) and thus fails the 'is_valid_ether_addr' check in the kernel.

With bluez-firmware version 1.2-3+rpi1, the mac address is what it was originally (with a proper B8:27:EB: prefix). However, this version of the firmware doesn't seem to work correctly with the rest of the bluetooth stack in Stretch.

Thanks,

Logan

[1] https://www.raspberrypi.org/forums/viewtopic.php?t=148305&start=100

asavah, 2017-08-22:
I can confim this too.

pelwell, 2017-08-23:
The new Bluetooth firmware seems to be ignoring the MAC address programmed into its OTP- the address you have now is a default for that chip (4343 A1 ...). Fortunately there is a way to set a MAC address using hciattach. [Here](https://drive.google.com/file/d/0B_P-i4u-SLBXRFZKOEFVU2xrcFE/view?usp=sharing) is a modified version of the /usr/bin/btuart script that should set the MAC address correctly.

If you want to try it, download it to the home directory of user `pi` and:
```
sudo mv /usr/bin/btuart{,.orig}
sudo cp ~pi/btuart /usr/bin/btuart
sudo reboot
```
To return to the shipping version, which I recommend before updating the pi-bluetooth package, run:
```
sudo mv /usr/bin/btuart{.orig,}
```

asavah, 2017-08-23:
Thx, modified btuart indeed works as expected here, bt mac now is the good one.

lsgunth, 2017-08-23:
Cool, thanks. This workaround solves my problems too.

closed 2017-08-23 by lsgunth