raspberrypi/firmware #861: Netboot never attempting with Raspberry Pi 3

andyshinn, 2017-08-26:
I've read a bunch of the threads on netboot / PXE boot failing. But they all seem slightly different than my problem and none of those resolutions have worked.

Basically, it just never seems to request any DHCP at boot or even attempt to do a netboot (it just boots right into Raspbian or the green light flashes 4 times).

What I've tried:
* Verified `vcgencmd otp_dump | grep 17:` output is `17:3020000a`.
* Run `sudo rpi-update` to get latest stable.
* Run `sudo BRANCH=next rpi-update` to get next branch.
* Format SD card as FAT32 and include only `bootcode.bin` from `master` branch.
* Format SD card as FAT32 and include only `bootcode.bin` from `next` branch.
* Format SD card as FAT32 and include only `bootcode.bin` from `pelwell` attachment in https://github.com/raspberrypi/firmware/issues/764.

None of the boots after these steps attempt any DHCP request at boot (I see the first DHCP packet when Raspbian boots instead). Any ideas what to try next?

unix0r, 2017-08-28:
Have you tried a different switch? A couple of Switches dop the first packets send from the booting pi.
I tried it with several Switches. There might be also a problem with the Spanning Tree Protocol that drops these packets.

ghollingworth, 2017-09-05:
If you remove the SD card completely and then power up the Pi, do the LEDs on the Ethernet come on after a short period?  Make sure no other USB things are plugged in just in case this is causing a problem.

closed 2017-09-05 by andyshinn