raspberrypi/firmware #864: bootcount feature?

mllapps, 2017-08-30:
Hy,

is it possible to create a bootcount feature like u-boot does with the current release firmware?

**bootcount**

The bootloader and the kernel have a shared space to access some variables. On boot the bootloader will increment the variable bootcount from 0 to 1 and starts the kernel. If the kernel boots successfully it will decrement the bootcount variable. But if the kernel does not decrement the variable the bootloader knows that the kernel do not boot successfully. So you are able to configure the number of retries until the bootloder will switch to a second kernel (recovery) and boot this alternativ kernel. 

The shared space could be a file called bootcount. So i think it is easy to realize and a very good solution to get the system working if a update fails.

https://www.denx.de/wiki/DULG/UBootBootCountLimit

popcornmix, 2017-08-30:
The bootloader (bootcode.bin/start.elf) never writes to the sdcard and I'd prefer it stayed that way (the sdcard driver is distinct from the linux one and has never been seriously testing for writing)

mllapps, 2017-08-30:
Ok,

do you know any other solutions without the boot partition?

pelwell, 2017-08-30:
If UBoot does what you want, why not use it?

mllapps, 2017-08-30:
Because I don’t want to use a another bootloader and I prepared my system to use the original bootloader for the raspberry pi. 

I think a solution could be interesting for anyone else.

closed 2019-11-05 by JamesH65