raspberrypi/firmware #865: BUG: Bad page state in process

andersthomson, 2017-08-30:
While streaming TV from tvheadend, I was greeted with this bug. This is on foundation provided 4.9.45 kernel. Never seen this before (on my self-compiled kernels)

[93836.014726] BUG: Bad page state in process tvh:mtimer  pfn:02e3c
[93836.025194] page:ba409070 count:0 mapcount:0 mapping:  (null) index:0x1
[93836.036185] flags: 0x0()
[93836.043067] page dumped because: page still charged to cgroup
[93836.053119] page->mem_cgroup:cbe4db11
[93836.060925] Modules linked in: rc_pinnacle_pctv_hd em28xx_rc rc_core si2157 si2168 i2c_mux tda18271 cxd2820r em28xx_dvb dm_mod dvb_core em28xx tveeprom v4l2_common videodev media evdev bcm2835_gpiomem bridge stp llc veth ftdi_sio usbserial uio_pdrv_genirq uio fixed nfsd sch_fq_codel ip_tables x_tables ipv6
[93836.105395] CPU: 3 PID: 521 Comm: tvh:mtimer Not tainted 4.9.45-v7+ #1031
[93836.116465] Hardware name: BCM2835
[93836.124067] [<8010fb3c>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
[93836.140375] [<8010c058>] (show_stack) from [<80456104>] (dump_stack+0xd4/0x118)
[93836.152155] [<80456104>] (dump_stack) from [<802113f8>] (bad_page+0x108/0x16c)
[93836.163890] [<802113f8>] (bad_page) from [<80211580>] (check_new_page_bad+0x8c/0x90)
[93836.180389] [<80211580>] (check_new_page_bad) from [<80214404>] (get_page_from_freelist+0x584/0xb0c)
[93836.198186] [<80214404>] (get_page_from_freelist) from [<80215164>] (__alloc_pages_nodemask+0xf0/0xdd0)
[93836.216211] [<80215164>] (__alloc_pages_nodemask) from [<802424f0>] (handle_mm_fault+0xb44/0xd98)
[93836.233699] [<802424f0>] (handle_mm_fault) from [<8071b334>] (do_page_fault+0x33c/0x3b0)
[93836.250456] [<8071b334>] (do_page_fault) from [<801011e8>] (do_DataAbort+0x48/0xc4)
[93836.266776] [<801011e8>] (do_DataAbort) from [<8071a7f4>] (__dabt_svc+0x54/0x80)
[93836.282850] Exception stack(0xb3eb9dc8 to 0xb3eb9e10)
[93836.292281] 9dc0:                   b902d8b8 00000000 00000000 00000000 5eb83000 00000000
[93836.309235] 9de0: 00000678 b20953d0 b3eb8018 000003d0 baaaa4f4 b3eb9e44 b3eb9e18 b3eb9e18
[93836.326640] 9e00: 804622d4 8046228c 000f0013 ffffffff
[93836.336452] [<8071a7f4>] (__dabt_svc) from [<8046228c>] (__copy_to_user_memcpy+0x68/0x190)
[93836.354097] [<8046228c>] (__copy_to_user_memcpy) from [<8046263c>] (arm_copy_to_user+0x20/0x2c)
[93836.372218] [<8046263c>] (arm_copy_to_user) from [<8046ab14>] (copy_page_to_iter+0xc8/0x364)
[93836.390152] [<8046ab14>] (copy_page_to_iter) from [<8027980c>] (pipe_read+0xc4/0x304)
[93836.407530] [<8027980c>] (pipe_read) from [<8026f7f4>] (__vfs_read+0xd0/0x120)
[93836.419617] [<8026f7f4>] (__vfs_read) from [<80270600>] (vfs_read+0x9c/0x168)
[93836.431534] [<80270600>] (vfs_read) from [<802717d8>] (SyS_read+0x54/0xb0)
[93836.443132] [<802717d8>] (SyS_read) from [<801080c0>] (ret_fast_syscall+0x0/0x1c)
[93836.459814] Disabling lock debugging due to kernel taint
[93836.469790] BUG: Bad page state in process tvh:mtimer  pfn:02e3d
[93836.480394] page:ba409094 count:-1370488635 mapcount:0 mapping:2a50b5b2 index:0x1 compound_mapcount: -1514626482
[93836.499669] flags: 0xb6f16a68(uptodate|lru|active|arch_1|private|writeback|head|mappedtodisk|mlocked)
[93836.517659] page dumped because: page still charged to cgroup
[93836.527838] page->mem_cgroup:2526e571
[93836.535827] bad because of flags: 0x116a68(uptodate|lru|active|arch_1|private|writeback|head|mappedtodisk|mlocked)
[93836.554550] Modules linked in: rc_pinnacle_pctv_hd em28xx_rc rc_core si2157 si2168 i2c_mux tda18271 cxd2820r em28xx_dvb dm_mod dvb_core em28xx tveeprom v4l2_common videodev media evdev bcm2835_gpiomem bridge stp llc veth ftdi_sio usbserial uio_pdrv_genirq uio fixed nfsd sch_fq_codel ip_tables x_tables ipv6
[93836.598306] CPU: 3 PID: 521 Comm: tvh:mtimer Tainted: G    B           4.9.45-v7+ #1031
[93836.614104] Hardware name: BCM2835
[93836.621381] [<8010fb3c>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
[93836.636840] [<8010c058>] (show_stack) from [<80456104>] (dump_stack+0xd4/0x118)
[93836.648105] [<80456104>] (dump_stack) from [<802113f8>] (bad_page+0x108/0x16c)
[93836.659222] [<802113f8>] (bad_page) from [<80211580>] (check_new_page_bad+0x8c/0x90)
[93836.674651] [<80211580>] (check_new_page_bad) from [<80214404>] (get_page_from_freelist+0x584/0xb0c)
[93836.691645] [<80214404>] (get_page_from_freelist) from [<80215164>] (__alloc_pages_nodemask+0xf0/0xdd0)
[93836.709070] [<80215164>] (__alloc_pages_nodemask) from [<802424f0>] (handle_mm_fault+0xb44/0xd98)
[93836.726101] [<802424f0>] (handle_mm_fault) from [<8071b334>] (do_page_fault+0x33c/0x3b0)
[93836.742547] [<8071b334>] (do_page_fault) from [<801011e8>] (do_DataAbort+0x48/0xc4)
[93836.758678] [<801011e8>] (do_DataAbort) from [<8071a7f4>] (__dabt_svc+0x54/0x80)
[93836.774756] Exception stack(0xb3eb9dc8 to 0xb3eb9e10)
[93836.784191] 9dc0:                   b902d8b8 00000000 00000000 00000000 5eb83000 00000000
[93836.801154] 9de0: 00000678 b20953d0 b3eb8018 000003d0 baaaa4f4 b3eb9e44 b3eb9e18 b3eb9e18
[93836.818596] 9e00: 804622d4 8046228c 000f0013 ffffffff
[93836.828401] [<8071a7f4>] (__dabt_svc) from [<8046228c>] (__copy_to_user_memcpy+0x68/0x190)
[93836.846150] [<8046228c>] (__copy_to_user_memcpy) from [<8046263c>] (arm_copy_to_user+0x20/0x2c)
[93836.864365] [<8046263c>] (arm_copy_to_user) from [<8046ab14>] (copy_page_to_iter+0xc8/0x364)
[93836.882393] [<8046ab14>] (copy_page_to_iter) from [<8027980c>] (pipe_read+0xc4/0x304)
[93836.899862] [<8027980c>] (pipe_read) from [<8026f7f4>] (__vfs_read+0xd0/0x120)
[93836.911993] [<8026f7f4>] (__vfs_read) from [<80270600>] (vfs_read+0x9c/0x168)
[93836.923964] [<80270600>] (vfs_read) from [<802717d8>] (SyS_read+0x54/0xb0)
[93836.935609] [<802717d8>] (SyS_read) from [<801080c0>] (ret_fast_syscall+0x0/0x1c)
[93836.952372] BUG: Bad page state in process tvh:mtimer  pfn:02e3e
[93836.963111] page:ba4090b8 count:1387088643 mapcount:0 mapping:a5b8a24d index:0x1
[93836.979898] flags: 0x9f290fb3(locked|error|dirty|lru|slab|owner_priv_1|arch_1|reserved|private|mappedtodisk|unevictable)
[93837.000052] page dumped because: page still charged to cgroup
[93837.010372] page->mem_cgroup:9f8cb31d
[93837.018567] bad because of flags: 0x90fb3(locked|error|dirty|lru|slab|owner_priv_1|arch_1|reserved|private|mappedtodisk|unevictable)
[... 700 lines cut ...]
[   20.080885] usb 1-1.2.4.1: DVB: registering adapter 0 frontend 0 (Sony CXD2820R)...
[   20.083951] em28174 #0: DVB extension successfully initialized
[   20.083966] em28178 #1: Binding DVB extension
[   20.150944] i2c i2c-6: Added multiplexed i2c bus 7
[   20.160072] si2168 6-0064: Silicon Labs Si2168-B40 successfully identified
[   20.160084] si2168 6-0064: firmware version: B 4.0.2
[   20.229623] si2157 7-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   20.229688] DVB: registering new adapter (em28178 #1)
[   20.229714] usb 1-1.2.4.3: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[   20.236811] em28178 #1: DVB extension successfully initialized
[   20.236824] em28xx: Registered (Em28xx dvb Extension) extension
[   20.299787] em28174 #0: Registering input extension
[   20.346572] Registered IR keymap rc-pinnacle-pctv-hd
[   20.347398] input: em28xx IR (em28174 #0) as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.1/rc/rc0/input2
[   20.347847] rc rc0: em28xx IR (em28174 #0) as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.1/rc/rc0
[   20.348477] em28174 #0: Input extension successfully initalized
[   20.348485] em28178 #1: Registering input extension
[   20.349293] Registered IR keymap rc-pinnacle-pctv-hd
[   20.349954] input: em28xx IR (em28178 #1) as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.3/rc/rc1/input3
[   20.350401] rc rc1: em28xx IR (em28178 #1) as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.3/rc/rc1
[   20.351008] em28178 #1: Input extension successfully initalized
[   20.351015] em28xx: Registered (Em28xx Input Extension) extension
[   22.218420] Adding 2097148k swap on /dev/mapper/4TB_VG-swap.  Priority:-1 extents:1 across:2097148k FS
[   24.798521] EXT4-fs (dm-2): mounted filesystem with ordered data mode. Opts: (null)
[   27.057406] EXT4-fs (dm-19): mounted filesystem with ordered data mode. Opts: (null)

pelwell, 2017-08-30:
My usual rule of thumb says that seemingly random crashes in MM code are often due to a power supply fault. Is your Pi2 adequately powered? Are the USB devices externally powered?

Start your use case then after a while run `vcgencmd get_throttled` to check for undervoltage issues.

If the crashes are more systematic then I would begin to suspect one of the drivers.

andersthomson, 2017-08-30:
The PSU has nice rapi logo on it, and identifies itself as 
STONTRONICS
Model DSA-13PFC-05 FCA  5.1 V, 2.5 A

USB wise, I have two self powered hubs connected to it. I've run vcgencmd get_throttled a few times and always get 0x0 back. It's running at 600 MHz, using the powersave controller.

pelwell, 2017-08-30:
So it's not power then - thanks, I had to ask.

Looking more closely at the log you can see that it appears to be corruption of a contiguous set of `struct page`s:
```
[93836.025194] page:ba409070 count:0 mapcount:0 mapping:  (null) index:0x1
[93836.480394] page:ba409094 count:-1370488635 mapcount:0 mapping:2a50b5b2 index:0x1 compound_mapcount: -1514626482
[93836.963111] page:ba4090b8 count:1387088643 mapcount:0 mapping:a5b8a24d index:0x1
[93837.460506] page:ba4090dc count:1864203433 mapcount:0 mapping:e6b80c29 index:0x1
[93837.984584] page:ba409100 count:2044786721 mapcount:0 mapping:a32773b5 index:0x1
[93838.472386] page:ba409124 count:-1079530538 mapcount:0 mapping:d009fdbd index:0x1 compound_mapcount: -456512508
[93838.969906] page:ba409148 count:941024227 mapcount:0 mapping:e4ca2c03 index:0x1
[93839.450101] page:ba40916c count:1460397259 mapcount:0 mapping:78b8b8b8 index:0x1
[93839.932911] page:ba409190 count:-117901064 mapcount:0 mapping:f8f8f8f8 index:0x1 compound_mapcount: 67372037
[93840.460110] page:ba4091b4 count:-2071690236 mapcount:0 mapping:04040404 index:0x1 compound_mapcount: 0
[93840.826750] page:ba4091d8 count:-1 mapcount:0 mapping:ffffffff index:0x1 compound_mapcount: 0
[93841.222381] page:ba4091fc count:0 mapcount:0 mapping:ffffffff index:0x1 compound_mapcount: 1
```
This is interesting because `struct page`s don't move - they are stored in a single large array allocated at start of day - so something has been writing in the wrong place. A hex dump of the corruption would have been useful, but looking at the values above it is a mix of repeated byte patterns and seemingly random values - perhaps video or audio data?

If it happens again then report back, but until then we don't have enough to go on.

andersthomson, 2017-08-30:
Will do. I've run this box for 2+ years and all but the initial few months using own compiled kernels (gentoo). I've never seen this exact error, but other kernel crashes has been there. 

As a tidbit, when I moved to gcc5 (userspace and kernel) I got the perception that I got more segfaults (userspace), _and_ kernel memory stuff such as unaligned accesses in kernelspace and what not. Got me thinking that gcc5@gentoo was not stable for arm. Now I'm back to all gcc4, and _those_ errors are gone (user and kernel space). I'm inclined to move userspace to gcc5 in a few months to see what happens.

I see you use crosstool gcc 4.x. Any reason you're not on gcc5 yet?

pelwell, 2017-08-30:
Another question I should have asked earlier - are you overclocked? Toolchain changes can reveal different silicon-specific timing issues when running the chip at out-of-spec clock speeds.

I'll let @popcornmix answer the gcc5 question.

popcornmix, 2017-08-30:
gcc 4.9 has been used to match the version debian jessie uses.
I'm currently testing gcc 6.4 (which matches debian stretch version) and gcc 7.2 (latest version supported by crosstool-ng).
I'll add one or both to tools repo when I'm happy everything works.
The latest kernel update (4.9.46) was built with gcc 6.4 and appears to be fine.
I'd be very surprised if a stable version of gcc causes panics when building the linux kernel.
As @pelwell says, power supplies and overclocking are far more likely causes of crashes.

andersthomson, 2017-08-31:
Hi,

I'm not overclocked. config.txt says:

dtdebug=on
kernel=kernel7.img

cpupower frequency-info:
analyzing CPU 0:
  driver: BCM2835 CPUFreq
  CPUs which run at the same hardware frequency: 0 1 2 3
  CPUs which need to have their frequency coordinated by software: 0 1 2 3
  maximum transition latency: 355 us.
  hardware limits: 600 MHz - 900 MHz
  available frequency steps: 600 MHz, 900 MHz
  available cpufreq governors: conservative, ondemand, userspace, powersave, performance, schedutil
  current policy: frequency should be within 600 MHz and 900 MHz.
                  The governor "powersave" may decide which speed to use
                  within this range.
  current CPU frequency is 600 MHz (asserted by call to hardware).
  cpufreq stats: 600 MHz:100.00%, 900 MHz:0.00%

If I get more memory related errors I'll look into changing the psu. As for now, the kevent errors are the most common source of lockups.

pelwell, 2017-08-31:
> If I get more memory related errors I'll look into changing the psu.

I would advise against spending money on a new PSU - based on what you've said so far it now seems unlikely to be a power problem. However, transient power glitches may go unnoticed by the firmware, so we can't rule it out.

If you get a similar crash again, one option would be to patch the "Bad page state" error code to print a hex dump of a larger window around the offending `struct page` in the hope of being able to identify the source of the corruption.

andersthomson, 2017-08-31:
Thanks. I'll keep that in mind.

andersthomson, 2017-08-31:
Another weird kernel error which might be related.
As the kernel paging thing is followed by the rcu stall which causes an immediate reboot. I have no chance to check e.g. the throttling before the reboot. Any chance some of that diagnostic can be added to the oops handling (or the reboot-on-oops handling)? That would make it to the serial log...

[58483.819723] Unable to handle kernel paging request at virtual address f5613015
[58483.831713] pgd = b8b54000
[58483.838913] [f5613015] *pgd=00000000
[58483.846863] Internal error: Oops: 805 [#1] SMP ARM
[58483.855897] Modules linked in: rc_pinnacle_pctv_hd em28xx_rc rc_core si2157 si2168 i2c_mux tda18271 cxd2820r em28xx_dvb dm_mod dvb_core em28xx tveeprom v4l2_common ftdi_sio videodev media usbserial evdev bcm2835_gpiomem uio_pdrv_genirq uio fixed bridge stp llc veth sch_fq_codel nfsd ip_tables x_tables ipv6
[58483.900462] CPU: 0 PID: 40 Comm: kswapd0 Not tainted 4.9.45-v7+ #1031
[58483.911123] Hardware name: BCM2835
[58483.918756] task: b9ef9d80 task.stack: b9f34000
[58483.927604] PC is at __es_shrink+0xa8/0x368
[58483.936028] LR is at 0xf5613011
[58483.943236] pc : [<80349854>]    lr : [<f5613011>]    psr: 600e0013
[58483.943236] sp : b9f35d20  ip : b9f35d20  fp : b9f35d84
[58483.962874] r10: 00000000  r9 : b250f180  r8 : 00000000
[58483.972027] r7 : 00000845  r6 : b250f0f4  r5 : b9fe66b4  r4 : b9fe6400
[58483.982441] r3 : b250f3b8  r2 : 00005087  r1 : 00000000  r0 : b9fe66b4
[58483.992760] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[58484.003686] Control: 10c5387d  Table: 38b5406a  DAC: 00000055
[58484.013243] Process kswapd0 (pid: 40, stack limit = 0xb9f34210)
[58484.022842] Stack: (0xb9f35d20 to 0xb9f36000)
[58484.030788] 5d20: b9f35d94 b9f35d30 8071a880 801e3578 c40f4a98 00003530 00000001 00000031
[58484.046589] 5d40: b9fe6740 00000000 c40f4a98 00003530 b9f35e94 0000004d 00000000 00000940
[58484.063154] 5d60: 00000000 00000000 b9fe6694 00000940 b9fe6000 00000080 b9f35db4 b9f35d88
[58484.080163] 5d80: 80349ea4 803497b8 00000001 0000008d 00000080 00000000 b9fe6694 00000080
[58484.097924] 5da0: 80b7cd38 00000000 b9f35e64 b9f35db8 80221c08 80349dcc 80222538 8022242c
[58484.116103] 5dc0: 00000000 0000000d b9f35f08 00000253 b9f35e64 b9f35de0 80225244 802224e4
[58484.134810] 5de0: b9f34000 0000094c 00000000 80b7cd38 0000007f 00000253 0000008d b9c82000
[58484.154051] 5e00: 80c1654c 00000000 00000000 00000000 024000c0 00000000 00000000 00000000
[58484.174012] 5e20: b9f35db8 00000940 024000c0 00000080 00000000 00000000 8026ae80 b9f35f08
[58484.194522] 5e40: 00000253 00000000 00000000 b9f35e94 00000000 80c6ef00 b9f35ecc b9f35e68
[58484.215568] 5e60: 80225678 80221960 0002f5a6 80226508 b9f35e94 80d05578 80c02e80 b9f35f00
[58484.237087] 5e80: 00000000 00000000 00000000 0002f5a6 0000000d 80c6ef00 00000008 00000008
[58484.259010] 5ea0: b9f35ecc 80c6ef00 80d06ddc 80d0557c 00000000 00000000 00000000 80c6ef00
[58484.281433] 5ec0: b9f35f5c b9f35ed0 80226558 80225324 b9f35f44 80b7cd44 80c729dc 80c0312c
[58484.303891] 5ee0: b9ef9d80 b9f35ed0 80c6f5bc b9f35f14 801615a0 00000003 80ca1c24 b9f34000
[58484.326325] 5f00: 00000000 00000000 00002fdd 024000c0 00000000 00000000 00000000 00000008
[58484.348769] 5f20: 00000000 00000007 00000253 00000253 00000000 00000000 b9f065c0 80c6ef00
[58484.371166] 5f40: 802261f8 00000000 00000000 00000000 b9f35fac b9f35f60 8013c938 80226204
[58484.393537] 5f60: 397ff000 00000000 55155555 80c6ef00 00000000 00000000 b9f35f78 b9f35f78
[58484.415895] 5f80: 00000000 00000000 b9f35f88 b9f35f88 b9f065c0 8013c82c 00000000 00000000
[58484.438229] 5fa0: 00000000 b9f35fb0 80108148 8013c838 00000000 00000000 00000000 00000000
[58484.460552] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[58484.482913] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 55555555 55555555
[58484.505307] [<80349854>] (__es_shrink) from [<80349ea4>] (ext4_es_scan+0xe4/0x170)
[58484.526887] [<80349ea4>] (ext4_es_scan) from [<80221c08>] (shrink_slab+0x2b4/0x504)
[58484.548213] [<80221c08>] (shrink_slab) from [<80225678>] (shrink_node+0x360/0x364)
[58484.569090] [<80225678>] (shrink_node) from [<80226558>] (kswapd+0x360/0x804)
[58484.582844] [<80226558>] (kswapd) from [<8013c938>] (kthread+0x10c/0x124)
[58484.596085] [<8013c938>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
[58484.609601] Code: e3580000 e2439f8e e5930004 e593e000 (e58e0004)
[58484.629614] ---[ end trace b41d873321c5b2f4 ]---
 [58509.505208] INFO: rcu_sched self-detected stall on CPU
[58509.543689]  2-...: (1 GPs behind) idle=1cd/140000000000001/0 softirq=403909/403910 fqs=1048
[58509.618253]   (t=2100 jiffies g=385158 c=385157 q=2950)
[58509.656285] Task dump for CPU 2:
[58509.692051] reader_check    R  running task        0   449    423 0x00000002
[58509.731978] [<8010fb3c>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
[58509.806591] [<8010c058>] (show_stack) from [<8014a908>] (sched_show_task+0xb8/0x10c)
[58509.881886] [<8014a908>] (sched_show_task) from [<8014cfe8>] (dump_cpu_task+0x48/0x4c)
[58509.957408] [<8014cfe8>] (dump_cpu_task) from [<80209ec0>] (rcu_dump_cpu_stacks+0x9c/0xd0)
[58510.033699] [<80209ec0>] (rcu_dump_cpu_stacks) from [<80180ea4>] (rcu_check_callbacks+0x800/0x9c0)
[58510.110842] [<80180ea4>] (rcu_check_callbacks) from [<80186698>] (update_process_times+0x40/0x6c)
[58510.188176] [<80186698>] (update_process_times) from [<80198898>] (tick_sched_handle+0x58/0x64)
[... cut ...]
[    0.000000] Linux version 4.9.45-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1031 SMP Fri Aug 25 19:02:16 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] Memory policy: Data cache writealloc

closed 2019-11-05 by JamesH65