raspberrypi/firmware #866: How to add modules and DT overlays

iurly, 2017-08-31:
I'd like to compile some extra modules (namely max310x) and add some device trees overlays.
I managed to do this locally but I'd like to keep using rpi-update, and propagate this to several identical systems.
Shall I fork this repo and push my changes to my fork? Is there a script for it?
Shall I ask for my changes to be merged mainstream?
I honestly don't see what the typical collaboration workflow looks like...
Thanks!

pelwell, 2017-08-31:
The source for modules and overlays lives in the [raspberrypi/linux](https://github.com/raspberrypi/linux) repo. The steps are:
1. Fork the main repo.
2. Clone your fork to your own computer (unless the change is trivial you'll want to use the command line utilities).
3. Make the required changes.
4. Build and test them.
5. Push the changes back to your own repo.
6. Create a Pull Request via the GitHub GUI.

Once the PR is merged the new modules and overlays will appear in the next firmware release.

iurly, 2017-08-31:
OK, that's what I thought then.
What if the pull request is not accepted or takes too long?
Any way I could do this on my own fork of **this** repo (i.e. making my own firmware release somehow)?

pelwell, 2017-08-31:
Reasons why your PR might not be accepted:
a. It changes upstream (not-Pi-specific) code - we want changes like that to be pushed upstream.
b. It doesn't compile.
c. It negatively affects stability or performance.
d. It makes the static kernel significantly larger. This is particularly an issue on early Pis with smaller RAMs.

Provided your PR just enables a few modules in the standard defconfig files (`bcmrpi_defconfig` and `bcm2709_defconfig`) and/or adds some overlays, it is unlikely we would either take very long to approve or flat-out refuse a PR.

It's a *lot* of work to set up and maintain your own firmware distribution, but rpi-update is open source and uses GitHub for distribution so clone it and knock yourself out.

iurly, 2017-08-31:
Bummer. I actually had to tweak the upstream driver to support the lack of an external clock.
Gotta go knock on that door first, then, I'm afraid.
Which means, if I get this right, posting a patch to the subsystem mailing list, wait for it to be picked by the maintainer and then to be pulled upstream.
Any other way for me to package this as a kernel-independent binary and install it manually?

pelwell, 2017-08-31:
I don't think there's any such thing as a kernel-independent kernel module. You could do what @MrEngman does for WiFi dongles and maintain a parallel distribution of just the added/changed modules needed for each kernel version. See [this forum post](https://www.raspberrypi.org/forums/viewtopic.php?t=148389#p985935) for an idea of how it work.

closed 2019-11-05 by JamesH65