raspberrypi/firmware #868: RPi 3 USB boot mode: Not able to boot from USB to SATA HDD adapters.

drencorxeen, 2017-09-02:
I have not been able to get any of my USB 2.0 or USB 3.0 to SATA converters working as a bootable USB device with any HDD I have.  Now these devices do work once Raspbian is fully loaded.  I can mount the volumes on the HDD(s) without problem.  I have tried using the Raspbian Jessie & Stretch and could not get either to boot off of HDD with these adapters.

This is a Thermaltake BlacX Duet Dock.  Uses USB 3.0 to dual SATA Dock. Bought from Best Buy
Bus 001 Device 004: ID 152d:2561 JMicron Technology Corp. / JMicron USA Technology Corp.

This is a USB 3.0 to SATA adapter.  From the packaging it says APRICORN on it and was a device bought from Best Buy.
Bus 001 Device 004: ID 0984:0301 Apricorn

This is a generic USB 2.0 to SATA / IDE adapter I got a few years ago.
Bus 001 Device 006: ID 152d:2338 JMicron Technology Corp. / JMicron USA Technology Corp. JM20337 Hi-Speed USB to SATA & PATA
Combo Bridge

ghollingworth, 2017-09-04:
I'd suggest trying to use the bootcode.zip from this issue (go to the bottom of the issue and look for the last version)

https://github.com/raspberrypi/firmware/issues/854

Then follow the instructions to enable uart output and capture the log whilst booting.  It's quite possible (and is generally quite normal) that the problem is the SATA hard-drive takes more than 5 seconds to boot up and therefore requires the TIMEOUT file added to the SD card (when booting just with bootcode.bin)

drencorxeen, 2017-09-04:
Well the problem is I am not using a microSD card.  I am using the USB BOOT feature. I don't want to use a microSD card.  Also the USB HDD Dock is powered up and fully running before the Raspberry Pi's power is supplied so the HDD should be fully active and ready by that time anyway.

To at least get the diagnostic stuff I stuck the bootcode file and uart on to a microSD card so you can get the log.

Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 0, enabling interface 0
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Found device on port 5
Found highspeed device
Device found: type = Ethernet adapter, addr = 6
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC xx:xx:xx:xx:xx:xx
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC xx:xx:xx:xx:xx:xx
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply

drencorxeen, 2017-09-04:
I also replaced the bootcode.bin file on the FAT32 partition on the HDD and also created a uart file there.  The thing is that the Raspberry Pi 3's firmware doesn't seem to be able to load the bootcode.bin off of the HDD.   I tried this on all 2 of the USB to SATA adapters and also the USB to SATA DOCK..

ghollingworth, 2017-09-04:
I understand you don't want to boot from the SD card, but this will help us diagnose the problem.

From the following:
`
Found device on port 5
Found highspeed device
`
This tells us that there is a device plugged into the Pi on port 5 (port 1 is the internal ethernet, the four USB ports are 2 - 5), I assume this is the USB->SATA interface?  You can check by moving it to a different USB port and it should move...

Since it has found the device but doesn't recognise it as a HDD that suggests that it is not responding to device configuration requests...  This could just be a timing thing, try creating a small file called 'timeout' on the SD card alongside the bootcode.bin (make sure it is not an empty file but the contents do not matter).  Then try again and see if that makes any difference...

Thanks

drencorxeen, 2017-09-04:
Sorry the one device might have been my USB USR 56K Modem.  I removed it and also changed the slot the USB to SATA Dock is plugged into.

Here is the uart output.

Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 0, enabling interface 0
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC b8:27:eb:ec:f4:eb
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply

pelwell, 2017-09-04:
And with the non-empty `timeout` file present?

drencorxeen, 2017-09-04:
I added the timeout file as request.
Here is the output.

Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 1
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 0, enabling interface 0
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC xx:xx:xx:xx:xx:xx
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC xx:xx:xx:xx:xx:xx
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Trying booting from Ethernet device addr 5

drencorxeen, 2017-09-04:
I did a lsusb -v to get more information about the device.
[lsusb20170904_Thermaltake_BlacX_DuetDock.txt](https://github.com/raspberrypi/firmware/files/1275575/lsusb20170904_Thermaltake_BlacX_DuetDock.txt)

drencorxeen, 2017-09-04:
here is the lsusb -v for the Apricorn USB to SATA adapter.
[lsusb20170904apricorn.txt](https://github.com/raspberrypi/firmware/files/1275626/lsusb20170904apricorn.txt)

drencorxeen, 2017-09-04:
Finally here is the lsusb -v for the generic USB 2.0 to SATA/PATA adapter.
[lsusb20170904usbsatapata.txt](https://github.com/raspberrypi/firmware/files/1275628/lsusb20170904usbsatapata.txt)

ghollingworth, 2017-09-05:
It's clearly not responding to the USB reset for more than five seconds...  

(Basically after a USB reset is signalled by the host the device is supposed to apply pull resistors to the USB lines to indicate low or full speed, (high speed is negotiated after this).  This is what happens to trigger the output "Found device on port 1", since your device is not doing this inside the five second time limit it is assuming there are no further devices and going to boot from Ethernet.)

I could increase this time limit but it'll never work with the bootrom anyway (meaning you'll always require an SD card)

Gordon

ghollingworth, 2017-09-05:
The attached bootcode.bin has an extended timeout of 20s so it should hang around for quite a while before dropping out of the enumeration loop (this could mean as much as 40 seconds to try booting, but should at least tell us if that's the problem)

Gordon

[bootcode.zip](https://github.com/raspberrypi/firmware/files/1276736/bootcode.zip)

pelwell, 2017-09-05:
You could try booting to Linux from and SD card without the SATA adaptor attached, then plug it in and see how long it takes the device to be detected. Enter this command on the Pi and hit Return as you plug in the adaptor:
```
while true; do date -Ins; lsusb; done
```
Make sure you have plenty of scrollback, or redirect the command to a file. Wait until you see the device appear, or for disk activity, or for 30 seconds, then kill the command with Ctrl-C and go back through the file. You should then be able to work out the interval.

drencorxeen, 2017-09-06:
@ghollingworth , I downloaded that new bootcode.zip file and copied the bootcode.bin to the microSD card, but it didn't seem like there was any extra delay.  As soon as the debug serial data popped up it started checking the USB ports.

@pelwell , here is the log file from Putty when I did your while loop.  Also I think from when I hit ENTER to the time I plugged the USB cable in was maybe 2 seconds.

[Thermaltake_BlacX_Duet_lsusb_time201709052040.txt](https://github.com/raspberrypi/firmware/files/1279483/Thermaltake_BlacX_Duet_lsusb_time201709052040.txt)

pelwell, 2017-09-06:
Assuming your 2 seconds is accurate, I make that an 8 second delay.

From what I understand, the delay is between enabling USB and starting to boot ("Trying booting from...") . In that time, devices will appear as they announce their presence to the host.

ghollingworth, 2017-09-06:
Yes, the delay will be evident elsewhere...

Can you please post the UART output with the timeout file as well.

Gordon

drencorxeen, 2017-09-06:
@ghollingworth ,
the timeout file is empty.

I did a Google search trying to find out more about this timeout file and only thing that showed up was:
https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/README.md
and in that just says add the timeout file, but doesn't describe what should be in it.

Here is the serial output from the uart capture:
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 1
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC xx:xx:xx:xx:xx:xx
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC xx:xx:xx:xx:xx:xx
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply

ghollingworth, 2017-09-07:
So there was no delay between:

`
Device found: type = Ethernet adapter
'
and
'
Trying booting from Ethernet device addr 5
'

drencorxeen, 2017-09-07:
I downloaded Kitty which is a fork of Putty.  I enabled timestamp feature of it so I could get some timestamp information.
Looks like 6 seconds?
[Thermaltake_BlacX_Duet_bootcode_uart_dump_201709071401.txt](https://github.com/raspberrypi/firmware/files/1285856/Thermaltake_BlacX_Duet_bootcode_uart_dump_201709071401.txt)

Now my next question is a newbie question since I don't know the process of USB initialization, but during the wait for devices to respond to reset shouldn't there be a longer delay in there?

closed 2019-01-17 by ghollingworth