raspberrypi/firmware #869: start.elf loads initramfs at same address as bcm2708-rpi-b-plus.dtb

drudru, 2017-09-06:
Hi,

I am working on getting a mini distro going for the Raspberry Pi Zero.
I'm using a Raspbian build that I did a recent rpi-update with. 
The kernel is 4.9.40+ built on Jul 30, 2017.

I was investigating a separate issue surrounding module loading for device tree items.

After setting config.txt with dtdebug=1 (which is a great feature),

I noticed in the 'vcdbg log msg 2>&1 | less'

```
Loading 'kernel.img' to 0x8000 size 0x42d588
brfs: File read: 4380040 bytes
brfs: File read: /mfs/sd/initrd.gz
Loading 'initrd.gz' to 0x435588 size 0x1825704
No kernel trailer -- assuming DT-capable
brfs: File read: 25319172 bytes
brfs: File read: /mfs/sd/bcm2708-rpi-b-plus.dtb
Loading 'bcm2708-rpi-b-plus.dtb' to 0x435588 size 0x3c60
```

Notice that the initrd.gz and dtb are being loaded to the **same address**.

The interesting thing is that the initrd image appears to be working.

Is this a bug in the start.elf dtdebug log message?

6by9, 2017-09-07:
Quirk more than a bug.

There is a helper function to load files. That is what is logging the "Loading 'file' to <addr> size <size>". It returns an offset pointer that points to the end of the newly loaded file, and that is often used for the next load.

In the case of an initramfs file, the loaded contents is loaded and then memmove'd elsewhere. That is why the pointer for the next file is not moved on.

The same is also true of the device tree , which is why a line after the parsing will list something like ```Device tree loaded to 0x2effa100 (size 0x5e96)``` (my original line was ```Loading 'bcm2708-rpi-2-b.dtb' to 0x489938 size 0x58f8```). You'll never notice that as the dtb is the last thing to be loaded.

pelwell, 2017-09-07:
It may help to understand that the intermediate API used in the firmware doesn't include a method to determine the length of a file (which makes more sense when you know that the source of the file could be a TFTP server), so unless given an explicit load address the firmware will load most things to the next available space after the kernel. In the case of the device tree file this works well because the application of overlays causes the blob to grow and there's lots of headroom after the kernel. Once the DT final size is known it is copied out of the way. The initrd doesn't grow after loading, but the extra cost of the copy is not large, so in this case we have chosen simplicity over absolute performance.

Finally, the file loading diagnostics are a recent addition, so some of the older messages could perhaps be reworded for clarity - "moved" or "placed" instead of "loaded", perhaps.

drudru, 2017-09-09:
Thanks for the comments. This has been very helpful 👍

closed 2017-09-09 by drudru